riscv/riscv-isa-manual #499: More than 16 PMP registers

aswaterman, 2020-04-06:
Some RISC-V users have requested more than 16 PMP registers.  We can fit another 48 in CSR address ranges 0x3A4-0x3AF and 0x3B0-0x3DF.

One hitch is that the current spec says that, if any PMP registers are supported, all of the PMP registers must be readable, even if hardwired to 0.  Obviously this statement needs to be amended if more PMP registers are added, or it would invalidate existing implementations.  We could just amend the spec to say that any number of PMP registers may be implemented, and some of those might be hardwired to 0.  Or require they be expanded in groups of 16, while still allowing some to be hardwired to 0.

allenjbaum, 2020-04-15:
to say that "any number of PMP registers may be implemented, and some of those might be hardwired to 0" isn't backwards compatible, but if no existing firmware touches those upper 48, and no one uses them as custom CSRs (they shouldn't be...) it doesn't matter. Another option is to say if any are implemented above 16, then all 48 are implemented. From a HW cost point of view, once you've defined any CSR as being all zero, the cost of implementing all of the upper 48 as all zero is negligible, just a few gates at most.  In hindsight, it probably would have been easier for both HW and SW to say that they're always implemented but could be hardwired 0.

aswaterman, 2020-04-15:
Right, I guess I was implying that it’s a benign backwards incompatibility. Maybe it’s not. I’m fine with your alternate formulation (if any of the upper 48 are implemented, all are).

joxie, 2020-04-22:
So the idea is to simply extend the CSRs to fit another 48?

pdonahue-ventana, 2020-09-09:
@aswaterman: This can be closed, right?

closed 2020-09-09 by aswaterman