riscv/riscv-isa-manual #500: New releases don't have PDFs

pdonahue-ventana, 2020-04-06:
Releases up to and including [draft-20200229-27b40fb](https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20200229-27b40fb) always had PDFs for the unprivileged and privileged specs.  [draft-20200229-f251913](https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20200229-f251913) only included a PDF for the unprivileged spec.  More recent releases don't include PDFs at all.

There were changes to the content of the spec(s) so having pre-built PDFs is useful.  Installing LaTeX and building the spec is non-trivial so asking all consumers to do that seems like too much to ask.

https://github.com/riscv/riscv-isa-manual/releases

allenjbaum, 2020-04-06:
Agreed; much better if this bit of work is done once instead of multiple
times.
Also makes it a bit more likely that everyone is looking at the same
version with minimal friction

neelgala, 2020-04-06:
This probably happened cause I had to change the travis file to use dpl-v2 to get the releases working again. During this switch I probably ignore other deprecated settings.

The solution now is to set ``cleanup: false`` in the yaml file. I don't have write access to it but have indicated the same solution to andrew as well.

Based on the travis build log and [this](https://blog.travis-ci.com/2019-08-27-deployment-tooling-dpl-v2-preview-release) blog.. it seems the above should solve it.

David-Horner, 2020-04-06:
When I geneate them with installed packages the rendering is substantial
different. Making comparisons of pdf difficult, even (especially?) with pdf
compare tools. I'd volunteer otherwise to generate and upload them.

On Mon, Apr 6, 2020, 13:18 Allen Baum, <notifications@github.com> wrote:

> Agreed; much better if this bit of work is done once instead of multiple
> times.
> Also makes it a bit more likely that everyone is looking at the same
> version with minimal friction
>
>

David-Horner, 2020-04-06:
Thank you so much.

aswaterman, 2020-04-06:
Thanks to @neelgala's travis wizardry, this is now fixed.

closed 2020-04-06 by aswaterman

pdonahue-ventana, 2020-04-06:
That was fast!  Thanks.