riscv/riscv-isa-manual #510: Reset state of misa

pdonahue-ventana, 2020-04-14:
Section 3.1.1 of the privileged spec says:
> At reset, the Extensions field should contain the maximal set of supported extensions, and I should be selected over E if both are available.

Note the use of "should" here which conveys a recommendation.

Section 3.3 of the same document says:
> The misa register is reset to enable the maximal set of supported extensions and widest MXLEN,
as described in Section 3.1.1.

This reads more like a requirement than a recommendation.

Based on #342, I assume that 3.3 was originally intended to reflect whatever 3.1.1 said so section 3.3 should be changed.

I started to make PR to change 3.3 to just "The misa register is reset as described in Section 3.1.1." Then I considered that resetting misa seems to be entirely optional because 3.1.1 allows misa to have random legal values on reset.  An implementation may therefore choose to not reset misa at all (as long as it doesn't reset to something illegal).  In that case, the sentence in 3.3 can probably be removed altogether since resetting misa is optional just like resetting any other register.

aswaterman, 2020-04-14:
No, it's just that the misa register reset constraint was added without updating the reset section. We should fix the reset section.

aswaterman, 2020-04-14:
Poor choice of words - the intent was to make it mandatory, so we should change "should" to "shall".

commit baea0255 (2020-04-14): Avoid "should" when describing a mandate

Resolves #510
closed 2020-04-14 by aswaterman