riscv/riscv-isa-manual #512: Clarification on misaligned loads/stores in RV64I, RV128I, and FDQ

knightsifive, 2020-04-17:
The RV32I spec suggests that portable code should avoid misaligned integer loads/stores:

> Loads and stores where the effective address is not naturally aligned to the referenced datatype (i.e., on a four-byte boundary for 32-bit accesses, and a two-byte boundary for 16-bit accesses) have behavior dependent on the EEI.
> [...]
> When an EEI does not guarantee misaligned loads and stores are handled invisibly, the EEI must define if exceptions caused by address misalignment result in a contained trap (allowing software running inside the execution environment to handle the trap) or a fatal trap (terminating execution).

The RV64I section says

> This chapter presents only the differences with RV32I, so should be read in conjunction with the earlier chapter.

It goes on to introduce several new memory operations, but doesn't explicitly state that the same alignment discussion applies to them too. Should I assume it does?

Similarly, the RV128I section says

> The variant is a straightforward extrapolation of the existing RV32I and RV64I designs.

and it introduce several new memory operations, but again doesn't explicitly state that the same alignment discussion applies to them too. Should I again assume it does?

The F-, D-, and Q- extensions only say,

> FLW and FSW are only guaranteed to execute atomically if the effective address is naturally aligned.

> FLD and FSD are only guaranteed to execute atomically if the effective address is naturally aligned and XLEN≥64.

> FLQ and FSQ are only guaranteed to execute atomically if the effective address is naturally aligned and XLEN=128.

How can I expect misaligned accesses to be handled in these cases?

aswaterman, 2020-04-17:
The same remarks about alignment do apply to the RV64 load/store instructions.  IMO, "This chapter presents only the differences with RV32I" is sufficient, because the RV32 text about misaligned loads and stores is already written pretty generically (i.e. the text isn't limited to the LH[U]/LW/SH/SW instructions).

I added a back-reference from the F extension chapter to the RV32I misaligned load/store section, which I think suffices to clarify your question there.  I wrote it in such a way that I don't think we don't need to add similar text to the D/Q chapters.  470a973251cd719786d990920e1a34858ba9e99c

knightsifive, 2020-04-17:
> "pretty generically"

> Loads and stores where the effective address is not naturally aligned to the referenced datatype (i.e., on a four-byte boundary for 32-bit accesses, and a two-byte boundary for 16-bit accesses) [...]

Maybe replace "i.e." by "e.g."?

aswaterman, 2020-04-17:
Good point.

aswaterman, 2020-04-17:
I went with "i.e., the effective address is not divisible by the size of the access in bytes".

closed 2020-04-17 by aswaterman