riscv/riscv-isa-manual #514: No definition of 'channel' in I/O Ordering

Dolu1990, 2020-04-22:
Hi,

In the `RVWMO Explanatory Material`, in the `I/O Ordering` section, there is the notion of `channel` used multiple times.

The issue is that there is no definition of what a `channel`  exactly is.

Personnaly i might have an idea, but i might be wrong / miss some details.

regards
Charles

Dolu1990, 2020-04-29:
Found some channel definitions in the previlegied spec. We might add a pointer ?

kersten1, 2024-03-29:
@Dolu1990 I realize this has been languishing for a while, but are you talking about linking to the section in the machine.adoc section? https://github.com/riscv/riscv-isa-manual/blob/37298e3248b8ddbaf21df417640a5970c1457281/src/machine.adoc#memory-ordering-pmas

Can you confirm location and I'll add a link?

We are also looking at getting a glossary going, but wanted to actually complete this issue.

Dolu1990, 2024-03-29:
Hi ^^

Ahhh it has been a long time, i think i was refering to https://github.com/riscv/riscv-isa-manual/blob/37298e3248b8ddbaf21df417640a5970c1457281/src/mm-eplan.adoc#io-ordering

which talk about channel 0 and 1 without realy explaining what they were.

Yes, a reference to https://github.com/riscv/riscv-isa-manual/blob/37298e3248b8ddbaf21df417640a5970c1457281/src/machine.adoc#memory-ordering-pmas would probably be good, as it explain what they are.

Thanks :)

kersten1, 2024-03-29:
@Dolu1990 Thanks! I'll have a link in ASAP. Glossary will take a bit longer :)