riscv/riscv-isa-manual #516: Which should be prioritized, non-compressed instruction or compressed instruction?

takahirox, 2020-05-02:
**Question**

Imagine that CPU fetches 32-bit instruction data from program including both compressed and non-compressed instructions and the data satisfies the both

* the 32-bit data is valid as non-compressed instruction
* the first 16-bit data is valid as compressed instruction

In this case, which should the fetched data be recognized as non-compressed instruction or as compressed instruction? If I'm right, it isn't clarified in the manual. (Please let me know if I've missed.)

I disassembled [Linux RISC-V port](https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html) kernel binary with riscv-gnu-toolchain and saw that it seems non-compressed instruction is prioritized. Is it correct?

**Suggestion**

If it's correct, I'd like to clearly word in the manual that

> If fetched 32-bit data is valid as non-compressed instruction and also its first 16-bit data is valid as compressed instruction, the data should be recognized as 32-bit non-compressed instruction.

(Feel free to correct my English because English is not my first language.)

What do you think? If it sounds good, I'd like to make PR.

Shadlock0133, 2020-05-02:
Instruction can't be both non-compressed and compressed, they have different prefixes. Normal instruction will have xxxxxx11 in LSB, compressed - xxxxxx00, xxxxxx01 or xxxxxx10. It's explained more in length in section 1.5 "Base Instruction-Length Encoding" of spec.

takahirox, 2020-05-02:
For example, if fetched 32-bit data is

000x_xxxx_xxxx_xx00_x011_xxxx_x000_0011

Then the first 16-bit represents as ADDI4SPN while the full 32-bit represents LD.

takahirox, 2020-05-02:
Ah, sorry. I think I misunderstood.

closed 2020-05-02 by takahirox