riscv/riscv-isa-manual #519: top-most grain of 34-physical-bit RV32 systems can't be addressed by PMP?

rbohn, 2020-05-07:
In the case of an RV32 system that supports 34-bit physical addresses, is it not possible to target the upper-most grain? Since the CSR can only hold physical bits 33:2, it's impossible to specify an address greater than the top-most grain for the purposes of the TOR "top" comparison?

With a less-then-34-bit physical system, one could implement one bit more than the physical address width to allow the top grain to be selected, but with 34-bit-physical we're out of address bits.

A fix might be to have all-zeros encode this just-greater-than the maximum address value and then use the implicit zero below entry 0 to cover the bottom of the range. Or state that, for the purposes of TOR, the address meaning is actual 0 if used as the prior entry base value, or if used at the top it is this just greater than the maximum grain address.

Apologies if this is covered somewhere in the spec already, I couldn't spot it.

aswaterman, 2020-05-07:
You can either forbid access to the top few bytes (not a significant loss
in a system with 16 GiB or address space) or use a NAPOT PMP. I don’t think
either workaround is so bad that an ISA change is warranted.

rbohn, 2020-05-07:
It might be worth a mention in the spec so software developers are aware. It's not real uncommon for systems to have things sitting at the top of the address range.

aswaterman, 2020-05-07:
Agreed.

zetalog, 2025-01-16:
> You can either forbid access to the top few bytes (not a significant loss in a system with 16 GiB or address space

Why not just specify an address with bit 34 is 1. Or if it can not be expressed in this way, just use 0.

Because an address of y should:

lower (pmpaddr[n-1]) <= y < pmpaddr[n]

1. There is an implication is pmpaddr[n] > pmpaddr[n-1].
2. y < pmpaddr[n] can be converted to y <= (pmpaddr[n] - 1).

I couldn't see significant difference between NAPOT and TOR, both should express the range in its own style.

aswaterman, 2025-01-17:
It's true that defining an implicit final PMP with address value 2^34 would suffice, but we can't change a ratified spec (especially given that there's no real problem here).

allenjbaum, 2025-01-17:
Um, we have precedent for changing ratified specs (or rather creating a new
spec with a change that isn't backward compatible if it is judged that it
will have unmeasurable SW impact.
In this case, it merely enables an extra world, but may also save a PMP
entry.
I will admit I haven't thought through if there is an actual
incompatibility, vs. simply enabling a configuration that wasn't possible
before - but we shouldn't dismiss this out of hand.
A more useful change would be to treat an allzero TOR as one with the most
significant+1 PA bit set (And PMPADDR-1 contents as 0 if it is encoded as
0.)
. Again, I haven't explored whether there are any incompatibilities

zetalog, 2025-01-20:
IMO, there is nothing wrong in the specification around TOR. We just need a clarification to correct those misunderstandings. Especially there is a real bug introduced by spike (which is a golden model of riscv specification) guys, confusing riscv developers.
[image]

aswaterman, 2025-01-21:
The range expression is de jure, not de facto.  And there's no bug in Spike.

zetalog, 2025-01-21:
Let me find some examples:

1. Linux memblock:
```
/**
 * memblock_add - add new memblock region
 * @base: base address of the new region
 * @size: size of the new region
 *
 * Add new memblock region [@base, @base + @size) to the "memory"
 * type. See memblock_add_range() description for mode details
 *
 * Return:
 * 0 on success, -errno on failure.
 */
int __init_memblock memblock_add(phys_addr_t base, phys_addr_t size)
{
        phys_addr_t end = base + size - 1;

        memblock_dbg("%s: [%pa-%pa] %pS\n", __func__,
                     &base, &end, (void *)_RET_IP_);

        return memblock_add_range(&memblock.memory, base, size, MAX_NUMNODES, 0);
}
```
end is deemed as base + size - 1 which wraps over 0.

2. device tree:
```
 The soc node specifies a ranges property of
 <0x0 0xe0000000 0x00100000>;
```
 **This property value specifies that for a 1024 KB range of address space, a child node addressed at physical
 0x0 maps to a parent address of physical 0xe0000000. With this mapping, the serial device node can be
 addressed by aload orstore at address 0xe0004600, an offset of 0x4600 (specified in reg) plus the 0xe0000000
 mapping specified in ranges.**
This is handled in base + len style and can also wraps over 0.

And more...

So far people couldn't see problems in real applications.
Probably spike could just change from addr < top_addr() to addr <= top_addr() - 1, or change from addr - tor_base_paddr() < top_addr() - tor_base_paddr(), then we could have this TOR usage model satisfied as NAPOT.

aswaterman, 2025-01-21:
@zetalog My point is that you keep describing it as a change to Spike, but it is really an ISA change you are requesting.  Spike correctly implements exactly what the ISA spec mandates.  Spike is not relevant.

zetalog, 2025-01-21:
Ah, I see. Let me describe my opinion in details and see if we can reach an agreement.

This is what ISA defined for TOR:
```
If PMP entry i’s A field is
set to TOR, the entry matches any address y such that pmpaddri−1 ≤ y < pmpaddri (irrespective of
the value of pmpcfgi−1). If PMP entry 0’s A field is set to TOR, zero is used for the lower bound,
and so it matches any address y < pmpaddr0.
If pmpaddri−1 ≥ pmpaddri and pmpcfgi.A=TOR, then PMP entry i matches no addresses.
```
Here we can see the formular used to determine if y is in the range: "pmpaddri−1 ≤ y < pmpaddri".
My opinion is this formular is **correct**. However people implementing it should treat it as an "address range" formular, which normally is handled by software and hardware in this way:
"pmpaddri−1 ≤ y ≤ pmpaddri - 1"
Or
"y - pmpaddri−1 < pmpaddri - pmpaddri−1"

Thus IMO, this is an implementation level ambiguity, not an issue of the ISA itself.

aswaterman, 2025-01-21:
I understand what you're asking for, but it is a backwards-incompatible change.  "y < upper" and "y <= upper-1" are not equivalent, since they differ in the specific case that the upper bound is 0.  Changing the spec to the latter would have two negative effects: it would render current (correct) implementations invalid, and it would open up a security hole for software that (correctly) assumes that an upper bound of 0 means "no addresses are valid".  Defining both implementations to be legal would fix the first problem, but would retain the second problem.

At the end of the day, this is a feature request--allowing RV32 to access 16 GiB of address space instead of 16 GiB minus epsilon.  This feature has some value, but not very much.  The cost of the incompatible change is significant.  I can't stop you from continuing to ask for it, but I won't support it.

zetalog, 2025-01-22:
IMO, there is a way of being equivalent:
pmpaddri−1 ≤ y < pmpaddri - 1
Then:
pmpaddri−1 - pmpaddri−1 ≤ y - pmpaddri−1 < pmpaddri - pmpaddri−1
Then:
0 ≤ (y - pmpaddri−1) < (pmpaddri - pmpaddri−1)
This finally could lead us to the same result.
That's why we treat range definitions' lower and upper in different way as upper is always greater than lower.

Could this be persuasive?

aswaterman, 2025-01-24:
> For "no addresses are valid case“，I suppose you mean pmpaddri−1 should be equal to pmpaddri.
> IMO, shall we imply that:
> **pmpaddri−1 < pmpaddri**
> Since:
> **pmpaddri−1 ≤ y < pmpaddri**

Not quite.  There is no requirement that `pmpaddr[i−1] < pmpaddr[i]`.  `pmpaddr[i−1] ≤ y < pmpaddr[i]` is not a statement of fact; it is a query.  Software is entitled to set `pmpaddr[i]` to 0, for example, to guarantee that query always returns false.  This is a property that can't be changed without changing the ISA.  There's no option for implementation-dependent behavior in this context.

zetalog, 2025-03-11:
Do you mean unless the following line is removed from the ISA, nothing can be changed then?
```
If pmpaddr[i−1] ≥ pmpaddr[i] and pmpcfg[i].A=TOR, then PMP entry i matches no addresses.
```
By my reading, this wording is to convince readers the implication "**pmpaddr[i−1] < pmpaddr[i] when pmpcfg[i].A = 1 (TOR)**" to make our previous inference possible rather than forcing hardware developers to introduce a control path "**pmpaddr[i-1] >= pmpaddr[i] && pmpcfg.A = 1 (TOR)**" into the silicon.
IMO, most of the system software/hardware developers will have the same feeling upon having a first glance of this wording.
If there is such a control path developed in the hardware, I would think it rather a hardware bug due to the different understanding of the wording. Software should prefer to use a control path value to achieve an invalid address entry, say **pmpcfg.A = 0 (Off)** rather than using this logic. Thus there is no real case affected by the different understanding other than the unconfigurable 1 byte (or several bytes due to the G) of TOR.

allenjbaum, 2025-03-11:
TOR mode has a bottom of a range, and the Top Of Range in 2 consecutive
PMPADDR registers.
The PA that is being checks must fall between those two ranges, (it has to
be >= than bottom, and < the top,
which is impossible if the top is less than the bottom. There are indeed
other ways to accomplish that,
and setting the mode to off  would do that (and IMHO would be preferred if
the next region is also a non-contiguous TOR region,
as it enables the off region to be a bottom of range regardless of
whether it is above or below the previous TOR.)

That doesn't change the fact that SW is expecting this behavior, and we
don't want to make incompatible changs..