riscv/riscv-isa-manual #520: Definition of WARL

pdonahue-ventana, 2020-05-08:
I think that the definition of WARL is a bit confusing and incomplete.

My initial concern was that the definition of "read" implicitly includes reads by hardware (which means that, practically speaking, the fixup happens on the csrw and not on a subsequent csrr).  Sure, that may be obvious to some but I think it would be nice to make that explicit.

Upon further consideration, I even doubt my conclusion above.  I think that this is incomplete:
> Implementations can return any legal value on the read of a WARL field when the last write was of
an illegal value

How about when the last write was of a legal value but that value subsequently became illegal?  For instance:
1. write misa.H=1 (a legal value)
2. write mie[12]=1 (a legal value for the SGEIE bit which is a WARL field)
3. write misa.H=0 (which causes mie[12]=1 to no longer be legal)
4. read mie

Or this similar scenario where the last write was of an illegal value but by the time the read is performed it is legal:
1. write misa.H=0 (a legal value)
2. write mie[12]=1 (an illegal value for the SGEIE bit which is a WARL field)
3. write misa.H=1 (which causes mie[12]=1 to become legal)
4. read mie

Is WARL generally intended to mean that the write performed in step 2 is the only place where legality is forced?  Or is hardware supposed to **also** fix up the mie value returned in step 4 (which affects the return value in the first scenario)?  Or is hardware **only** supposed to fix up the mie value in step 4 (which affects the return value in the second scenario)?  Or perhaps, as someone from another RISC-V company suggested to me, is step 3 in the first scenario supposed to follow its own WARL behavior and refuse to clear misa.H because mie[12] is set?

This is just an example scenario.  I'm sure there are more so a general clarification would be helpful.

Thanks.

pdonahue-ventana, 2020-05-14:
Assuming S and U modes are supported, mstatus.MPP is WARL and there are only 3 legal encodings.  WARL says "Implementations can return any legal value on the read of a WARL field when the last write was of an illegal value..."  The reset section says that this field is UNSPECIFIED after reset.

What if there was no previous write of a field such as MPP?  Must the first read of MPP after reset (either via an explicit CSR read instruction or as part of performing an MRET) return a legal value?  Is that legal value required to be deterministic based on the architectural state of the hart?  If so, the state of the hart at reset (when there is a requirement for misa.S=1) or at the time of the read (when, for instance, software may have set misa.S=0 which further limits the legal values of MPP)?

aswaterman, 2020-05-14:
Good question about reset. I think WARL registers should always honor the
“RL” part, so that following reset, they aren’t allowed to contain
arbitrary garbage. Perhaps reset could be defined to be a “write”; then the
existing WARL definition would handle this.

gfavor, 2020-05-14:
This does mean, though, that all WARL fields with any illegal encodings,
across all CSR's, will need to be reset by the hardware hart reset signal.
Whereas the current specification of most hart state after reset as being
UNSPECIFIED, otherwise allows for a very limited fanout of the reset signal
to just a few architectural state flops.

Also, this doesn't address the question of whether encoding legality is
based on the architectural state at the time of writing the CSR or at the
time of reading the CSR (or both)?

Greg

gfavor, 2020-05-14:
An alternative solution would be to say that hart state that is UNSPECIFIED
after reset truly is UNSPECIFIED, i.e. a read of a WARL field could return
any value.  It is not until a WARL field has been implicitly or explicitly
written that it is no longer UNSPECIFIED and the standard WARL
definition now holds.

Greg

aswaterman, 2020-05-14:
Limiting reset signal fanout is definitely a goal. I’m not sure we’re
talking about that many flops, though—probably a substantially smaller
number than the number of microarchitectural state bits that need to get
reset in most designs. (Would be good to quantify that claim, of course.)

Re: the question about legality at the time of the write vs. the read: I
don’t think we want to silently put the machine into an illegal state, so I
think WARLness should be enforced on writes.

pdonahue-ventana, 2020-05-14:
> Re: the question about legality at the time of the write vs. the read: I don’t think we want to silently put the machine into an illegal state, so I think WARLness should be enforced on writes.

But enforcing it only on writes may put it into an illegal state.  Enforcing it on reads (both explicit reads from CSR instructions and implicit reads by hardware) seems to be the only way to prevent illegal state.  For instance, my example of writing S mode to MPP and then disabling S mode via misa.S.  The only way to fixup MPP to be legal on later reads is to take misa.S into account at the time of the read (either CSRR or MRET).

gfavor, 2020-05-14:
To argue a little more for the alternative of WARL fields being UNSPECIFIED
after reset and before first being written:

Even if one resets WARL fields to legal values, all other CSR state will be
in legal but rather weird and arbitrary states.  Invariably boot software
needs to eventually initialize much of this stuff to suitable starting
values, at which point what does it matter that a small subset of CSR state
held illegal values if read before being properly initialized?

Early on (after reset) all these UNSPECIFIED values, whether legal or
illegal, don't affect anything (otherwise they would need to be initialized
to suitable specified values by reset).  And boot software needs to set
them to suitable starting values before transitioning to a mode in which
they can start affecting things.

Greg

aswaterman, 2020-05-14:
@gfavor Yeah, that's a defensible point of view.  I'm still weakly in favor of being strict about this, but I'd like to hear some other folks chime in.

@pdonahue-ventana You're right, of course.  It also matters whether the HW's consumption of the value in those registers counts as a "read".  There are other ways to address your specific example, e.g., we could define that writable misa.S doesn't impact which values of mstatus.MPP are legal.  (I'm not advocating we do that; just pointing it out.)

allenjbaum, 2020-05-14:
Suppose misa.S is RW, misa.S =1,  mstatus.MPP=01 (S).  Now misa.S is
cleared: does status.MPP become... zero? Remain unchanged?
Then, set misa.S back to 1: what value is in mstatus.MPP? 00? The previous
value of 01?
If the answer is 01, then it's clear that both HW and SW reads will see
only the legal value, so it's the CSR state output that is affected by some
other field, not the input.
If the answer is 00, then it's clear that the write of one CSR must affect
the state of another CSR (in this and other cases), so it's the input that
is affected, not the output
At the same time, one common method for mapping illegal->legals value is to
simply leave the existing legal value unchanged.
This becomes more difficult, because now the implementation has to detect
illegal combinations on both the input (to inhibit the write) and the
output (in case misa.S is RW).

CSR fields whose legal values changes depending on the value of other
architectural state (generally other CSRs) are ugly to define and test; the
above is just one example.
It would be useful to compile a definitive list of all the dependencies.
The formal model will need that (or has it and perhaps can be extracted to
guide implementations).

The compliance WG has come up with a WARL description syntax that does
allow different illegal->legal mappings depending on other state.
I don't think it is enough to describe all the possibilities above, however
- and that's for just a two bit field and a single dependency.
The particular mappings described are the only ones that compliance will
test, and implementations that use non-standard mappings will fail
compliance tests (someday, when we get that far with tests). The formal
model will only handle (need to) those as well, instead of dealing with
total arbitrary mappings.

I hope we don't have fields whose legality also depends on the values of
other fields that are being concurrently written. That was an issue for a
forthcoming extension but had a simple fix made in the most current draft
spec.

There are 3 types of fields defined., since WIRI fields have been
eliminated in the latest .
WARL fields, as above.
WPRI fields are limited to xstatus CSRs and xIE. Illegal values are allowed
to be written, yet the field must be hardwired to zero for future
compatibility.
    (from a compliance point of view, they are treated as WARL-zero until a
spec that uses them is defined)
WLRL fields are limited to xCAUSE and unused misa bits. Illegal values may
be written SW or HW, but are not used by HW,  are pure status
   (from a compliance point of view, they are simply RW)
   I'm unclear why misa was defined this way instead of making the field
WPRI.

With my compliance hat on (I say this a lot) I'd rather see a far more
restrictive version of WARL fields.

gfavor, 2020-06-14:
After further thought I suggest the following resolution to this issue - which is an expanded or clarified version of the existing "always read legal" requirement.  This addresses the weird or "interesting" cases mentioned in earlier posts, and also avoids creating any backward compatibility issues for existing implementations.

_All explicit software reads and implicit hardware uses of a WARL bit or field must always see a currently legal value and the same value must be seen by both explicit and implicit reads/uses.  An implementation can choose to legalize a value being written to a WARL bit/field - based on what are legal values at that moment - either fully, partially, or not at all.  This allowance is just for implementation convenience.  (Consequently, a value that was illegal when written and is later read when it would be legal, is not guaranteed to be the value read back since it may have been legalized at the time of the write to a different value.  Further, that legalized value - if now illegal at the time of the read - may be re-legalized to yet another value.)_

Note that this rule also covers post-reset scenarios.  An implementation can choose to initialize WARL state or not.  All that matters is ensuring that all explicit and implicit reads of the state are legal.

allenjbaum, 2020-06-14:
I am guessing this means that the illegal->legal mapping can happen on the
input, or the output., or both.
For fields whose legal values change depending on some other state, it must
have mapping on the output, but may also do it on the input as well. That
behavior is observable if you write an illegal value, change external state
to make that illegal value now legal, and see if the field reverts to the
original illegal value or not.
Fields that don't have that conditional behavior or whose should never see
a difference (outside of scan logic).

I don't understand the "partially" option - why?

Putting my compliance hat on:
 I want to constrain WARL behaviors as tightly as possible; they are
incredibly difficult to test, and the formal model must be updated for
every possible behavior, of which there are basically an infinite number.
Maintaining the formal model will be difficult as it is currently stands;
in fact, there is no owner to for it. I don't want to do anything to make
that task more difficult.

Implementations will not pass compliance tests if illegal->legal mappings
are not constrained to specific documented  (in the compliance framework
docs) mappings, (and I'd like to reduce those cases).
Even with that, for some CSRs (e.g. triggers, whose formats - not just
legal field values -  change depending on another field in the CSRs), I
don't know how to test WARL behavior at all, and I don't know how the
formal model can model them. There aren't a lot of cases outside those
so far..

I suspect the easiest, most common mapping is to inhibit writing the field
if an illegal value is detected on the input, and I'd like to encourage
that (for non-conditional cases). That avoids needing any mapping logic at
all, just illegal value detection. Otherwise my preference is to map the
output but not map the input for conditional fields.  That should simplify
the logic for the conditional cases (they only need one set of mapping
logic, not two), For non-conditional cases that don't write inhibit (I
would like to discourage not inhibiting) it's just moving the logic from
one side of the register to the other; the design effort is no worse, and
that logic is pretty static, so timing shouldn't be much of an issue
either.(yes, I know of the corner cases).

I understand the desire for implementation freedom, but this is a case
where I assert that constraining the behavior does not measurably affect
design effort, does not decrease value to the customer, and does make the
job of passing compliance tests easier.

gfavor, 2020-06-14:
On Sat, Jun 13, 2020 at 11:51 PM Allen Baum <notifications@github.com>
wrote:

> I am guessing this means that the illegal->legal mapping can happen on the
> input, or the output., or both.
> For fields whose legal values change depending on some other state, it must
> have mapping on [...]

Yes.  It is just these "weird" cases - where legality is arch state
dependent - that cause all the trouble.

> I don't understand the "partially" option - why?
>

I was simply allowing complete flexibility to an implementation.  It can do
whatever it wants - just as long as the apparent-to-hardware-and-software
"always read legal" requirement is satisfied.

But yes, for many WARL bits and fields I would imagine most implementations
would *fully* legalize write values if they do any legalization at the time
of write.  There may, though, be some WARL fields where an implementation
would prefer to map write values to "supported" values - and then leave it
to read legalization to do the arch state-dependent mapping to whatever may
be the currently legal values.

> Putting my compliance hat on:
> I want to constrain WARL behaviors as tightly as possible; they are
> incredibly difficult to test, and the formal model must be updated for
> every possible behavior, of [...]

I appreciate and can agree with the sentiments.  But someone has to come up
with a precise architectural definition that works across all the "weird"
cases and narrows down the allowed behaviors, and this has to be accepted
by people, and it probably (?) needs to not make any
existing implementations non-compliant.

If someone can propose such and get it accepted, that would be great.
What's most important is removing the current ambiguity in the spec and
providing some form of specification for how the "weird" situations are to
be handled.  I'm not wedded to my proposal - it is just an expedient
attempt at trying to get a resolution to occur by not trying to force a
narrow requirement.  At the expense for compliance testing, as you noted,
by allowing implementation flexibility.

One alternative, I imagine, could be to require full legalization on write
(based on the current arch state at that moment) as well as on read (based
on the current arch state at that moment).  Which some may find
objectionable implementation-wise for some WARL fields?

I would encourage proposing a concrete and *specific* better alternative,
with argument for why it should not be objectionable, so that people can be
persuaded that it is the best/right resolution to this issue.

Greg

>

allenjbaum, 2020-06-14:
Understood, and I do know where you're coming from (and vice versa I think).
 I will try to write a more constrained definition, and see if the
community objects.
That will expose any implemention that matches your definition, but not the
my TBD constrained proposal.
I can't imagine anyone reasonable has implemented something that goes
beyond it, at least.

gfavor, 2020-07-04:
To come back around to try and get a resolution on this general issue, I'd like to propose a tighter version of what I previously proposed (for the sake of compliance testing) - which essentially constrains the allowed implementation behaviors when handling illegal WARL values.

- All explicit software reads **and** implicit hardware uses of a WARL bit or field must always see a currently legal value and the same value must be seen by both explicit and implicit reads/uses.  This requirement for "read legalization" is always in effect.

- For WARL address fields in various CSR's, a very small set of behaviors for representing invalid address values would be supported by compliance tests.  Allen has started down the path of sorting this out.

- For non-address WARL fields, two legalization options would be supported by compliance tests:

1) No write legalization other than dropping bits of a field that correspond to unimplemented bits (e.g. for a 5-bit field that only has three flops to store the low three bits of the field, the high two bits are dropped on write).  Any further legalization that is needed within the stored bits is done via "read legalization".  The stored flop values remain unaffected by any read legalization.

2) Full write legalization.   For fields whose legal values are not state-dependent on other CSR bits, this also ensures "read legalization".  For fields whose legal values are state-dependent on other CSR bits, explicit legalization on reads still applies as needed (without modification of the stored flop values).

- After hart reset, WARL bits and fields that are defined by the Privileged spec to be UNSPECIFIED, are stored with undefined values.  But read legalization is always in effect.  (Note: This does allow for an implementation that chooses to initialize such bits/fields to specific legal values, but does not require it.)

Greg

allenjbaum, 2020-07-04:
Please look at this documentation for the proposed Compliance
illegal->legal mappings.

https://riscv-config.readthedocs.io/en/latest/yaml-specs.html#warl-field-definition

#1 above corresponds (I think) to the mask/match version.

The most common mapping is likely to be "inhibit write", i.e. just keep the
existing  legal value. There are existing WARL fields that don't seem to
conform to either of the schemes you list above, e.g. SATP.mode which has a
discontiguous set of legal values.

Note that it is possible to assign different mappings to subfields of a
field, e.g. MTVEC could have the lower bits defined to be read_only zero
(to force cache or page boundary alignment), and the upper bits be an
address range (not necessarily a power of two).

None of this addresses whether the state of a field is maintained if it is
conditional on some other CSR state, i.e. where changing that other CSR
state transitions the field from legal to illegal and then the other CSR
state reverts back, making the original field state legal again.

Having said that: these definitions are all the possible ways I think any
reasonable implementation would define their mapping - and it is almost
certainly waaay overkill.
I welcome any suggestions to reduce it further.

jscheid-ventana, 2021-08-10:
Converging on this remains valuable.  What's the next step?

allenjbaum, 2021-08-10:
Which part?
Defining what happens if changing some other CSR affects the legality of
this CSR, i.e.
does it simply affect the output value, but not the stored value, so
reverting the other CSR also reverts back to it original (now legal again)
value?
vs. actually changing the store value, so reverting the other CSR won't
revert to the original value - but might itself now be illegal and need
remapping?
Or suggestions for reducing the scope of illegal->legal mappings?

jscheid-ventana, 2021-08-11:
The spirit of the initial question didn't include the compliance constraint model issues.  I mostly interested in that initial aspect.

I think the net is that implementations shall legalize based on current architectural state on explicit and implicit reads.

Is seems that is it also allowed for implementations to legalize on explicit and implicit writes, but not required.  

The latter may be observable through the order in which other state changes between write and subsequent read.

Is it allowed or prohibited to observe state changes based on other control state being written?

For example, 

1. write A=V when V is a legal value
2. write control0 to further restrict legal values of A to not include V
3. restore previous value of control0
4. read A and observe ~V.

allenjbaum, 2021-08-12:
We've talked about that in terms of 
 - input mapping: convert the value being written into a legal value.
 - output mapping: convert the output state into a legal value.
 - implementation choice if input or output mapped (or both - one for explicit writes, one for implicit writes)
 Input mapping requires forcing a legal value to be actually written even when not addressed
 Both have timing challenges (getting the mapped value and forcing teh write for input mapping,
 output delay in the output mapping case.)

jscheid-ventana, 2021-08-12:
In my model above, there's also "reconfiguration mapping" as a possibility to require, allow, or prohibit.

An example architecture, not that this is a proposal at this time:
- Require output mapping
- Allow observable input mapping (i.e., the type that is observable beyond output mapping)
- Prohibit observable reconfiguration mapping (i.e., the type that is observable beyond output and input mapping)

allenjbaum, 2021-08-12:
Is reconfiguration mapping what we were describing where a write to a
different CSR makes a currently legal mapping illegal (or vice versa)?
I thought that was referred to as an implicit write - is that just giving a
name to that concept?

And - can you give an example of something beyond input and output mapping?
My imagination is clearly limit there.

jscheid-ventana, 2021-08-12:
A more concrete example than what is in https://github.com/riscv/riscv-isa-manual/issues/520#issuecomment-897169786.  This specific case may not be technically correct (sepc may need to hold all possible valid values regardless of current mode), but I hope it at least conveys the spirit.

1. write satp to Sv48 mode
1. write sepc with a nice large legal/valid "48b" value, which is also illegal/invalid for Sv39
1. write satp to Sv39 mode
1. do anything that has no implicit or explicit writes to sepc
1. write satp to Sv48 mode
1. read sepc

If the read of sepc can be something other than what was written, that's an "allowed" reconfiguration mapping.  Perhaps in your mind the write to satp would have an "allowed" implicit write mapping of sepc.  But since these "cross" implicit writes aren't otherwise a thing in the ISA, I'm calling this a separate reconfiguration mapping notion.

scottj97, 2021-08-12:
Here's a real example in Spike: [Fix disabling hypervisor via misa](https://github.com/riscv/riscv-isa-sim/pull/658)

The arch spec requires that when misa.H=0, everything works as if hypervisor is not implemented. This requires either output mapping (masking certain CSRs when misa.H=0) or implicit writes at the time misa.H is written to 0.

I implemented the latter in that PR.

The difference is visible when the software later re-enables misa.H and examines the bits that were either implicitly written or output masked. The spec says these bits are undefined at such time, so either implementation is permissible.

allenjbaum, 2021-08-13:
OK, if the spec says that these bits are undefined (for this case, at
least) I think it would be fair to say that would be the case for all other
reconfigurations.
So that's an disallowed reconfiguration as defined by Joshua, except it
isn't.
So you can input map or output map or map&rewrite (which is how I would
characterize the disallowed reconfiguration
If CSRs are implemented so they just recirculate every clock, then both
input and output mapping are the same.
If they're implemented with clock gating, output mapping is easier and
saves power which is why I wold favor it.
But from a spec point of view, either works. From a practical standpoint,
it probably works as well - its a rare enough case,
and explicitly specing that it could be either forces code to rewrite -
which probably results in more work, but fewer bugs.

closed 2025-12-09 by aswaterman