riscv/riscv-isa-manual #521: Atomic patching of instructions

Amanieu, 2020-05-11:
Code patching at runtime has many uses, for example a debugger patching code to insert a breakpoint instruction or a JIT engine patching a branch/jump instruction to point to a different piece of code. Sometimes this patching must be done atomically because another hart may be concurrently executing the code that is being patch.

The ISA manual does not specify what rules must be followed to ensure that a concurrently executing hart does not observe a partially written instruction. This is particularly a concern with the C extension since an implementation may choose to read a 32-bit instruction using two 16-bit memory accesses.

Essentially, I feel that the ISA manual should answer these two questions:
- Under what circumstances is patching a 16-bit instruction atomic with regards to concurrent instruction fetches?
- Under what circumstances is patching a 32-bit instruction atomic with regards to concurrent instruction fetches?

jscheid-ventana, 2021-07-30:
I think this was addressed here [https://github.com/riscv/riscv-isa-manual/issues/650#issuecomment-852737970], stating that it's up to the platform.

Amanieu, 2021-07-30:
Good enough for me!

closed 2021-07-30 by Amanieu