riscv/riscv-isa-manual #527: FADD.S/FSUB.S underflow

tariqkurd-repo, 2020-06-04:
Hi,

The manual says FADD.S and FSUB.S can set NV, OF, UF, NX.
We cannot find any cases where UF (underflow is set). We have complete FPU verification and used a formal proof to show that it cannot happen.
Can you please confirm if UF is possible, and if it is possible tell us how to set it?

Regards

Tariq Kurd

Rolfy360, 2020-06-04:
https://stackoverflow.com/questions/19053681/causing-underflow-in-ieee-754-floating-point-format-using-subtraction

tariqkurd-repo, 2020-06-04:
from that link: "When default exception handling is in effect, a subtraction that produces a tiny (in the subnormal interval1) non-zero result conceptually causes an underflow exception, but there is no observable effect"

I agree, we can't observe it, so I think the ISA manual shouldn't state that it is possible to observe it.

Rolfy360, 2020-06-05:
Hi Tariq Kurd:
May I ask which formal tool do you use to do the FPU verification?

aswaterman, 2020-06-05:
Yes, this is all true.

closed 2020-06-05 by aswaterman

tariqkurd-repo, 2020-06-05:
Hi Rolfy360 we use Jasper from Cadence

Rolfy360, 2020-06-05:
> Hi Rolfy360 we use Jasper from Cadence

Thank you for the reply, we use the same one. :)

commit a37867a2 (2020-06-05): FADD/FSUB can't raise UF, hence, no PPO dependence

Resolves #527