riscv/riscv-isa-manual #532: Should we also encode the spec versions into architecture string?

Nelson1225, 2020-06-29:
Greetings,

I'm working on the spec version checking in GNU binutils recently.  Ideally, all instructions and CSR should be generated and controlled by the spec versions, if there are some conflicts between different spec versions.  I know we will do our best to avoid conflicts between versions in the future, but in case some inevitable conflicts, I think do some checks for spec versions in the assembler and linker should be helpful.

For now on the FSF binutils, we only check the spec versions for the privileged CSR.  The privileged spec version is chosen by the assembler option `-mpriv-sepc=[1.9.1|1.10|1.11]`.  But there are some unprivileged CSR should be controlled by other specific specs rather than the privileged spec.  For example, the debug/float/vector CSR should be controlled by the debug/float/vector spec.  I expect to support more specs checking in the future, so it might seem cumbersome to add multiple options to choose different spec versions.  Besides, since we already record the privileged spec versions into the elf attributes section `.riscv.attributes` [1], we might need to add new elf spec attributes to record each spec.

I'm wondering if we can encode the spec versions into the architecture string, then we don't need to add multiple assembler options and elf attributes for each specs.  But the architecture string will become more difficult to be analyzed, and is not user-friendly enough, though it was already...

For example,
`-march=rv32ifv_ziprivspec1p11_zidebugspecxxx_zfspecxxx_zvspec1p0`

Thanks
Nelson

[1]  https://sourceware.org/binutils/docs-2.34/as/RISC_002dV_002dATTRIBUTE.html#RISC_002dV_002dATTRIBUTE

jim-wilson, 2020-06-29:
The unprivileged spec already provides a way to specify the f and v version numbers e.g. -march=rv32if1p0v1p0 or -march=rv32i_f1p0_v1p0 if you want it a little more readable.  See chapter 27 of the unprivilege spec.  These should already work with binutils and gcc, though may not be well tested, and probably don't do anything useful yet.  The problem is with the priv spec and debug spec as we don't have letters for those.  So it is two extra options for priv spec and debug spec, or define extension letters for them so we can add them to -march.

Nelson1225, 2020-06-29:
> The unprivileged spec already provides a way to specify the f and v version numbers e.g. -march=rv32if1p0v1p0 or -march=rv32i_f1p0_v1p0 if you want it a little more readable.

Oh yes, I was a little bit confused when creating this issue...  I also correct the example above.
Thanks for pointing this :)

pdonahue-ventana, 2020-06-29:
> The problem is with the priv spec and debug spec as we don't have letters for those.

Specifically, there's no letter or name for the M mode portion of the privileged spec.  There are letters and names for the S and H and N portions of that spec.  For instance, you can currently specify S1p12_H0p6p1_N1p1.  Do there need to be names like Zipriv (for the base privileged architecture without the S/H/N extensions) and Zidebug?

(I notice that the latest draft H extension 0.6.1 has two dots and therefore doesn't comply with the major+minor version numbering scheme of section 27.4 of the unprivileged spec, though that will presumably be cleaned up before ratification.)

aswaterman, 2025-12-09:
There's a standard way to do this now (e.g. `Sm1p13` and `Ss1p13`).

closed 2025-12-09 by aswaterman