riscv/riscv-isa-manual #533: Clarify version for G extension

kito-cheng, 2020-07-02:
Does it possible to clarify version for G extension, current ISA spec version scheme no longer using major/minor version number, so the version for G would become very confusing, so I think does it possible to forbid version appear with G at spec?

I saw the usage of G with version was all removed at https://github.com/riscv/riscv-isa-manual/commit/a7826272c6bcfdcd74acf05b38bb28e46594190a , but I would like have an more explicitly wording on this.

jim-wilson, 2020-07-02:
There is a closely related problem with implied extensions.  D implies F, and F implies Zicsr, so if someone specifies d2p2 what if anything does that imply about the f and Zicsr versions?  The D and F extensions seem to be in lock step, so making d2p2 mean f2p2_d2p2 might be OK.  But F and Zicsr are not in lock step, so f2p2 can't imply zicsr2p2 as there is no 2p2 version of zicsr.  We could forbid a version spec when an extension implies another.  Or we could say that a version spec is OK when an extension implies another, but the implied extension will have an unknown version number.  Or we could set up a table so that D versionX implies F versionY implies zicsr versionZ.  Or maybe there is another solution.

aswaterman, 2025-12-09:
Since 2018, `G` has meant the ratified versions of `IMAFD_Zicsr_Zifencei`.