riscv/riscv-isa-manual #534: Clarify that mstatus.FS is still visible when writable misa.F bit is cleared

kasanovic, 2020-07-03:
Clearing misa.F does not cause mstatus/sstatus.FS to disappear.  The state bits are still visible.  Needs to be added to spec.

allenjbaum, 2020-07-03:
Interesting - do you know if the SAIL model  models that?
And... why?

This goes back to an earlier discussion about WARL bits : if the condition
for legality of a WARL field is dependent on the state of some other CSR
bit, and that state bit changes to make it illegal - and then legal again -
must the old state be retained?
Is this true for all WARL fields that are conditional, or is it
 - implementation defined, or is it
 - case by case or is it
 - implementation defined case by case?

In this example the "appearance" of the bits  is dependent on
      (RW(misa.F)) | misa.F=1)  && ~Zfinx, rather than
                             (misa.F=1)  && ~Zfinx. (where ~appearance
means RdOnly 0)

This is a general issue that the formal model would much rather handle
uniformly...

I'm also trying to figure out if this simplifies or complicates the config
YAML description.

pdonahue-ventana, 2020-07-03:
The manual says that FS is WARL.  It seems inconsistent with the spirit of WARL to allow it to be non-zero when misa.F=0.

Because misa.F is controllable only by M mode and FS is visible to S mode via sstatus.FS, is there a virtualization problem when M mode tries to present the SEE as a non-F implementation?

Does this new requirement provide value to the overall architecture or is it to accommodate an existing implementation that behaved in this way?

gfavor, 2020-07-03:
This "FS" case seems analogous to the situation with the H extension.
Before the H extension comes along, certain CSR bits and fields are defined
as WPRI, WARL, or 0's.  Then the H extension gives functional meanings to
these bits.  But when misa.H=0, all those bits are supposed to appear as if
the H extension was not implemented (i.e. as 0's and not as whatever last
values they held when misa.H=1).

Treating FS differently when misa.F=0 would be inconsistent, but obviously
the arch spec could choose to officially and clearly make this exception
and define FS accordingly.

Greg

kasanovic, 2020-07-03:
I think the cleanest way to think of mstatus.FS bits when misa.F is 0
is as an extension (Zffs??) that supports FP emulation.  This
extension is implied with F extension, so is always present when
misa.F is 1.

This new extension does not (currently) have a standard way of being
turned off if supported.

Krste

| This "FS" case seems analogous to the situation with the H extension.
| Before the H extension comes along, certain CSR bits and fields are defined
| as WPRI, WARL, or 0's. Then the H extension gives functional meanings to
| these bits. But when misa.H=0, all those bits are supposed to appear as if
| the H extension was not implemented (i.e. as 0's and not as whatever last
| values they held when misa.H=1).

| Treating FS differently when misa.F=0 would be inconsistent, but obviously
| the arch spec could choose to officially and clearly make this exception
| and define FS accordingly.

| Greg

| On Fri, Jul 3, 2020 at 12:07 PM Paul Donahue <notifications@github.com>
| wrote:

|| The manual says that FS is WARL. It seems inconsistent with the spirit of
|| WARL to allow it to be non-zero when misa.F=0.
|| 
|| Because misa.F is controllable only by M mode and FS is visible to S mode
|| via sstatus.FS, is there a virtualization problem when M mode tries to
|| present the SEE as a non-F implementation?
|| 
|| Does this new requirement provide value to the overall architecture or is
|| it to accommodate an existing implementation that behaved in this way?
|| 
|| —
||
|| <https://github.com/riscv/riscv-isa-manual/issues/534#issuecomment-653653178>,
|| .
|| 

| —

allenjbaum, 2020-07-03:
That is part of my question.
Is the state
 - retained (so it will read as zero, but when MISA.F is set again the previous state reappears?
  (And what happens if it is written? Does it overwrite the hidden state or just ignore the write) 
 -or retained and visible (same question as above - what happens with writes)

-Allen

aswaterman, 2020-07-04:
FS is orthogonal to the F extension. Hence, misa.F doesn’t govern whether it’s readable or writable.

There is already precedent for FS existing even when the F extension isn’t implemented. The spec mandates that FS be implemented if S-mode is implemented, irrespective of the F extension.

Inversely, there is also already a proposal for FS to be hardwired to zero even when the F extension is implemented (Zfinx).

I suppose I agree with Krste that FS could be formalized as a separate extension.

pdonahue-ventana, 2020-07-04:
Wouldn't emulating FP (whether an extension or part of the base architecture) also require instructions such as FADD.S to trap?

If misa.F=0, accesses to fcsr/frm/fflags must trap per section 2.1 of the privileged spec but the behavior of reserved instructions is very explicitly UNSPECIFIED in section 2.2 of the unprivileged architecture.  Nowhere is that latter statement amended by the privileged architecture.

I considered that perhaps FADD.S is an unimplemented "standard" instruction rather than a section 2.2 "reserved" instruction but two things argue that it is still not required to trap.  One is that the privileged architecture seems to reenforce the notion that instructions in unimplemented extensions don't have to trap:
> SRET must be provided if supervisor mode is supported, and should raise an illegal instruction exception otherwise.

The use of the word "should" must be taken as a suggestion, especially when it is obvious that this word was carefully chosen over the word "must" that appears earlier in the same sentence.

The other, more compelling argument is that the general philosophy of RISC-V is that anything that is unstated is unconstrained and nowhere is the behavior of unimplemented standard or custom instruction encodings constrained.

allenjbaum, 2020-07-04:
It is a obvious that, if FP is not in HW and it’s emulation is required, then all FP opcodes must trap. 

There was a long discussion about how a profile must separately specify whether FP  is required, and whether FP HW is required. If HW is not required, then it implies that 
- Mmode code must never use/rely on FP ops and 
- an implementation would be considered compliant to that profile if either MISA.F is 1, or all FP ops trapped properly.

Note that compliance has no current way to make that differentiation, the formal model has not been updated to allow traps of specific undefined opcodes, and profile descriptions are few and far between and also don’t make that differentiation.

-Allen

Rolfy360, 2020-07-10:
'It is a obvious that, if FP is not in HW and it’s emulation is required, then all FP opcodes must trap."
It is also related to vector-extension FP instructions.
If there is an RV64IMACV core, any vector FP instructions should trap no matter what FS is

allenjbaum, 2020-07-10:
Ah: if the vector-scalar opcodes are not optional in the base V-extension
spec, and ZFinx is not imnplemented (which would then require the presence
of FP-regs) then that would be the case. If neither of those conditions
hold, or if it is a requirement than the F or D extensions are required,
then the above core type isn't even legal.

jscheid-ventana, 2021-08-10:
> The spec mandates that FS be implemented if S-mode is implemented, irrespective of the F extension.

Is this a sufficient resolution, or is the scope here that FS should have a way to exist in non-S-mode-supporting implementations?  If the latter, then the proposal here is not that a "clarification" is required, but that an extension is needed?

allenjbaum, 2021-08-10:
I'm not sure what it means for FS (as subfield of mstatus) to be
implemented without F extension.
Does that mean it won't trap if accessed (the usual meaning of
"implemented" for an entire CSR)?
  (that's a rhetorical question)
Does it mean that it must be RW, and not RdOnly0?
Is it expected that any other extension might cause the value of mstatus.FS
to change without F-extension being implemented?

aswaterman, 2021-08-10:
@allenjbaum In this context, "implemented" means "not hardwired to 0".

commit a37624ef (2021-08-11): State that misa.F does not affect mstatus.FS

Closes #534
closed 2021-08-11 by aswaterman