riscv/riscv-isa-manual #535: Relationship ISA manual with Sail ISA description

martinberger, 2020-07-06:
Hello, I'm wondering if there is a way of relating the manuals to the version of the RISC-V Sail interpreter they correspond to. In theory the manual and the Sail interpreter describe the same ISA.

allenjbaum, 2020-07-06:
In theory (only) SAIL descriptions? pseudocode?  are supposed to be merged
into the manual - that was a goal, at least.
That will be a long term project, and it is not staffed right now.

martinberger, 2020-07-06:
Thanks. Is there a version of the Sail interpreter (or indeed any executable specification) that is closest to what's described in the manual?

allenjbaum, 2020-07-06:
From an email last month:

The current state is that:

- The specification is here:  https://github.com/rems-project/sail-riscv
- It generates C and OCaml emulators, theorem-prover definitions for Coq,
HOL4, and Isabelle, SMT definitions, and LaTeX documentation.
- It covers RV32/RV64IMAFDC and MUS,Sv32,39,48 (floating-point is only in
the C emulator, using softfloat).  It doesn't cover RV128, separate M/U/S
control of endianness or register width, or the hypervisor extension.
- It's integrated into our operational (rmem) and axiomatic
(isla-axiomatic) concurrency-model tools.
- We have draft versions (by Prashanth Mundkur) of the RISC-V ISA manual
that incorporate the Sail-generated LaTeX:

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-spec-sail-draft.pdf

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-privileged-sail-draft.pdf

  Volume 1 (unprivileged) inlines the Sail definition with each group of
instructions.
  Volume 2 (privileged) is in a more preliminary state: it includes some of
the Sail definitions, but not always adjacent to the corresponding prose.

Some of these and other things are new: since the Formal ISA TG concluded
its selection process, we've:

- added FD
- integrated with our isla-axiomatic concurrency-model tool
- added support (in Sail) for specification branch coverage measurement; and
- added parameterisation for one version of the YAML
configuration discussed in the compliance group.

closed 2020-07-07 by martinberger