riscv/riscv-isa-manual #536: Clarify for mip.MSIP is read-only in mip

guanhaohz, 2020-07-07:
In description of Machine-Level CSRs/Machine Interrupt Registers (mip and mie):

Bits mip.MSIP and mie.MSIE are the interrupt-pending and interrupt-enable bits for machine level software interrupts. **MSIP is read-only in mip, and is written by accesses to memory-mapped control registers,** which are used by remote harts to provide machine-level interprocessor interrupts.

How to  understand "MSIP is read-only in mip, and is written by accesses to memory-mapped control registers".

allenjbaum, 2020-07-07:
It means that a CSRRW op will cause changes to MSIP.
The only way to cause changes is to perform a STx op to an (implementation
defined) address which will cause MSIP to be modified.

Note that not only is there no universal address (each implementation must
define their own), there is also no definition of precisely what writing to
that address - or set of addresses  - will do, e.g.
 - there may be separate addresses to cause MSIP bits to be set or cleared,
or
 - there could be separate addresses for each individual bit in MSIP

These address(es) are global, in the sense that any hart can modify MSIP on
a hart by writing to that address, including the hart that contains that
CSR.
Each hart has  their own unique global address(es) that will enable
modifying the MSIP CSR for that specific hart.

guanhaohz, 2020-07-07:
@allenjbaum Thank you for your answer
MSIP is read-only in mip
SSIP is writable in mip

closed 2020-07-07 by guanhaohz