riscv/riscv-isa-manual #538: Can cached translations be visible when translation is off?

palmer-dabbelt, 2020-07-10:
The `sfence.vma` text describes translations as being cacheable and the `satp` text describes 0 as translations being off.  From that I'd expect that turning off translation in `satp` prevents cached translations from being visible, as translation is off (as opposed to having an identity translation).  As a result, we don't need a `sfence.vma` after the CSR write that turns off translation in `satp`.

I don't really have any issue with that behavior, but it's a bit subtle and I thought it would go the other way before reading the text.  If that's the desired behavior, we should put another note in the sfence.vma describing the situation.

aswaterman, 2020-07-10:
Since we write "SFENCE.VMA is used to synchronize updates to in-memory memory-management data structures with current execution", and satp is not an in-memory memory-management data structure, aren't we already on solid footing?

palmer-dabbelt, 2020-07-10:
I was considering this to fall into the "reusing an ASID" category, which normally requires a fence.

alercah, 2020-07-31:
It doesn't seem inconsistent with ASIDs to me: changing the ASID in `satp` doesn't require a fence to load the correct table; reusing it to point at a new page table does, because you need to flush any previously stored tables associated with that ASID.

aswaterman, 2020-07-31:
Decision made in side discussions is to clarify current intended behavior.

closed 2020-07-31 by aswaterman

alercah, 2020-08-01:
Would it maybe make sense to also mention that changes to `satp` that update ASID take effect immediately, so that it's clearer that ASIDs can be freely moved in and out of `satp` if the page table is the same, as opposed to reusing an ASID for a new, different page table?

aswaterman, 2020-08-01:
@alercah Yeah, I suppose we say something close to the inverse (SFENCE is necessary if an ASID is recycled), but we don't explicitly remark that SFENCE is unnecessary when merely switching ASIDs.  I'll update accordingly.

commit 866326d9 (2020-07-31): Clarify that satp.MODE transitions to/from Bare don't require SFENCE.VMA

Resolves #538