riscv/riscv-isa-manual #541: Add appendix that lists and briefly describes all instructions

NilsGraf, 2020-07-17:
Would it make sense to add an appendix to the RISC-V spec that lists all instructions alphabetically and describes them succinctly? This would be similar to the Appendix A of book "The RISC-V Reader."  

The official [RISC-V website](https://riscv.org/risc-v-books/) links to the [free Spanish](http://riscvbook.com/spanish/guia-practica-de-risc-v-1.0.5.pdf), Chinese, and Portuguese versions of this book. 

I think such an appendix would be very useful, especially the pseudo-code for each instruction such as `x[rd] = x[rs1] + x[rs2]` for the ADD instruction.  Thanks!

kasanovic, 2020-07-17:
Would be a great thing to generate from the formal model.
Any volunteers?

allenjbaum, 2020-07-17:
I think the formal modeling group started that effort- but the priv mode hasn’t been finished. I can dig up a link to it later.

-Allen

allenjbaum, 2020-07-17:
Here is a pointer to an annotated ISA manual

We have draft versions (by Prashanth Mundkur) of the RISC-V ISA manual that
  incorporate the Sail-generated LaTeX:

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-spec-sail-draft.pdf

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-privileged-sail-draft.pdf

  Volume 1 (unprivileged) inlines the Sail definition with each group of
instructions.
  Volume 2 (privileged) is in a more preliminary state: it includes some of
the Sail definitions,
  but not always adjacent to the corresponding prose.

gfavor, 2020-07-19:
Btw, what were the Sail models verified against (to flush out any bugs in their representation of the arch specs)?  And do both of those things model the entire range of behaviors and implementation options that are allowed in the arch specs (especially wrt the many options in the Privileged spec)?

allenjbaum, 2020-07-19:
Originally there were 5 formal models, and the sail model was selected as
the golden one.
All of them gave the same results on tests, though I don't know how
comprehensive those tests were.
Currently, no model implements the entire range of behaviors, especially
regarding the WARL illegal->legal mappings. And as we uncover more features
that turn out to be optional, the less like it is that the formal model
models them in every combination.
We do need a plan for formal model maintenance and updates - we need
resources to do that, and it is being looked into.

If someone fails compliance, and argues that their
implementation correctly handles whatever case failed, then we have to
decide if we should rev the spec to agree, or add wording to the spec to
make it clear why the spec handles this case correctly. (and theirs doesn't)

NilsGraf, 2020-07-21:
I would like to volunteer. What are the next steps to add an appendix that lists all RV32I instructions alphabetically and describes them succinctly?  Thanks

allenjbaum, 2020-07-22:
There is an annotated version of the unprivileged spec here, with the sail
formal model semantics code for each op.

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-spec-sail-draft.pdf

https://github.com/rems-project/riscv-isa-manual/blob/sail/release/riscv-privileged-sail-draft.pdf
(priv spec doesn't have the code next to the description, I've been told)

The instruction specifications use Sail nomenclature,  are inline (not in
an appendix), and almost certainly not sorted the way you want... but it is
a place to start looking at what is possible. The ISA manual is currently
in Latex format
My understanding is you want to add an appendix, you can
 - download the source
 - create an appendix file, populated the way you want it to look.
 - modify some other bit of latex so this new appendix gets appended
 - submit a pull request  and wait for fireworks.

A better way might be to contact the sail team, find out how they extracted
the text from the model, and use the same technique to do it with sorted
appendix instead of manually inserting the text (e.g. write a script that
exacts the code and puts into a latex table. I would also submit the script
if you do that.)

jscheid-ventana, 2021-08-10:
Close as an inactive enhancement initiative?

allenjbaum, 2021-08-10:
I think the intent is still to fully implement this; people are manually
inserting the code into extension descriptions, but we want it to be
automatic.

allenjbaum, 2021-08-10:
oops - I was responding to the sail annotated version, not the appendix,
which started the thread, to be clear.

ethindp, 2021-08-10:
This is something that I've been working on whenever I'm able (which
really isn't much) using MdBook. My idea is to contain a list of all
instructions sorted by extension, with each instruction containing the
instruction encoding (including opcodes), a description of the
instruction, any exceptions/errors/etc. that it can generate, and
(possibly) pseudocode, in the style of the Intel SDMs (vol. 2). I've
documented all of the "core" ones (following the opcodes in the
riscv-opcode repository) but I hesitated at adding
RV32/64k/RV32/64p/RV32/64V because those aren't fully standardized,
and I'm unsure whether I should include draft extensions.
I haven't (yet) posted the source on GH (primarily because I've had so
little time to work on it that I've only got the TOC down) but I'd be
happy to open a new repository and then submit it here if you guys
want to help add to it. Something like this -- as an open initiative
-- would be wonderful to see.

aswaterman, 2021-08-11:
This is an active initiative (with an admittedly low activity factor) and for the sake of honest accounting I think it should remain open.

jscheid-ventana, 2021-08-11:
Good to hear.  Should this be assigned to @ethindp ?

aswaterman, 2021-08-11:
Not yet, I think.  The backstory behind my "low-activity-factor" remark is that we've started taking the suggested approach for new specs, but for the original unpriv spec, this augmentation is effectively blocked behind an ongoing RVI project to port this document to an asciidoc-based flow.  We're shunning substantive additions to the LaTeX spec in the meantime to reduce the porting burden.  I suggest we revisit this topic once that effort nears completion.

ethindp, 2021-08-12:
I currently don't have the time to maintain my little project. I'll
upload it to GH though if anyone wants to help, but its currently
manually written. (And like I said, only the TOC exists right now). I
do cover RV32/64B but I wonder if I should remove that... It seems
like a (relatively) stable extension at the moment.
I'll have more time, I think, once the semester starts. Unless I get
bogged down in homework.

aswaterman, 2021-08-12:
@ethindp in my experience, spending your time doing your homework is a worthwhile prospect :)  We'll figure this one out eventually.

ethindp, 2021-08-12:
Lol yeah it certainly is, but I find working on riscv and other things like
that a lot more interesting than doing homework. But, as you said, it is a
worthwhile prospect. Regardless of how boring it happens to be.

>
> @ethindp in my experience, spending your time doing your homework is a
worthwhile prospect :) We'll figure this one out eventually.

NilsGraf, 2022-12-15:
Alternatively, would it make sense to add an updated version of [this RISC-V cheat-sheet](https://inst.eecs.berkeley.edu/~cs61c/fa18/img/riscvcard.pdf), which briefly describes the instructions of the I, M, and A extensions (and is only 2 pages long)?   

On a similar note, I recently implemented [many I, M, and F instructions in Python here](https://github.com/OpenMachine-ai/tinyfive), the first ~150 lines are copied below. 
```python
#-------------------------------------------------------------------------------
# base instructions (RV32I) 

# arithmetic
def ADD (s,rd,rs1,rs2): s.x[rd] = s.x[rs1] + s.x[rs2]; _pc(s)
def ADDI(s,rd,rs1,imm): s.x[rd] = s.x[rs1] + imm;      _pc(s)
def SUB (s,rd,rs1,rs2): s.x[rd] = s.x[rs1] - s.x[rs2]; _pc(s)

# bitwise logical
def XOR (s,rd,rs1,rs2): s.x[rd] = s.x[rs1] ^ s.x[rs2]; _pc(s)
def XORI(s,rd,rs1,imm): s.x[rd] = s.x[rs1] ^ imm;      _pc(s)
def OR  (s,rd,rs1,rs2): s.x[rd] = s.x[rs1] | s.x[rs2]; _pc(s)
def ORI (s,rd,rs1,imm): s.x[rd] = s.x[rs1] | imm;      _pc(s)
def AND (s,rd,rs1,rs2): s.x[rd] = s.x[rs1] & s.x[rs2]; _pc(s)
def ANDI(s,rd,rs1,imm): s.x[rd] = s.x[rs1] & imm;      _pc(s)

# shift (note: 0x1f ensures that only the 5 LSBs are used as shift-amount)
def SLL (s,rd,rs1,rs2): s.x[rd] = s.x[rs1]     << (s.x[rs2] & 0x1f); _pc(s)
def SRA (s,rd,rs1,rs2): s.x[rd] = s.x[rs1]     >> (s.x[rs2] & 0x1f); _pc(s)
def SRL (s,rd,rs1,rs2): s.x[rd] = _u(s.x[rs1]) >> (s.x[rs2] & 0x1f); _pc(s)
def SLLI(s,rd,rs1,imm): s.x[rd] = s.x[rs1]     << imm;               _pc(s)
def SRAI(s,rd,rs1,imm): s.x[rd] = s.x[rs1]     >> imm;               _pc(s)
def SRLI(s,rd,rs1,imm): s.x[rd] = _u(s.x[rs1]) >> imm;               _pc(s)

# set to 1 if less than
def SLT  (s,rd,rs1,rs2): s.x[rd] = 1 if s.x[rs1]     < s.x[rs2]     else 0; _pc(s)
def SLTI (s,rd,rs1,imm): s.x[rd] = 1 if s.x[rs1]     < imm          else 0; _pc(s)
def SLTU (s,rd,rs1,rs2): s.x[rd] = 1 if _u(s.x[rs1]) < _u(s.x[rs2]) else 0; _pc(s)
def SLTIU(s,rd,rs1,imm): s.x[rd] = 1 if _u(s.x[rs1]) < _u(imm)      else 0; _pc(s)

# branch
def BEQ (s,rs1,rs2,imm): s.pc += imm if s.x[rs1] == s.x[rs2]         else 4
def BNE (s,rs1,rs2,imm): s.pc += imm if s.x[rs1] != s.x[rs2]         else 4
def BLT (s,rs1,rs2,imm): s.pc += imm if s.x[rs1] <  s.x[rs2]         else 4
def BGE (s,rs1,rs2,imm): s.pc += imm if s.x[rs1] >= s.x[rs2]         else 4
def BLTU(s,rs1,rs2,imm): s.pc += imm if _u(s.x[rs1]) <  _u(s.x[rs2]) else 4
def BGEU(s,rs1,rs2,imm): s.pc += imm if _u(s.x[rs1]) >= _u(s.x[rs2]) else 4

# jump
def JAL (s,rd,imm): s.x[rd] = s.pc + 4; s.pc += imm
def JALR(s,rd,rs1,imm): t = s.pc + 4; s.pc = (s.x[rs1] + imm) & ~1; s.x[rd] = t

# load immediate
def LUI  (s,rd,imm): s.x[rd] = imm << 12;          _pc(s)
def AUIPC(s,rd,imm): s.x[rd] = s.pc + (imm << 12); _pc(s)

# load, note the different argument order, example: 'lb rd, offset(rs1)'
def LB (s,rd,imm,rs1): s.x[rd] =  _i8(s.mem[s.x[rs1] + imm]);  _pc(s)
def LBU(s,rd,imm,rs1): s.x[rd] =      s.mem[s.x[rs1] + imm];   _pc(s)
def LH (s,rd,imm,rs1): s.x[rd] = (_i8(s.mem[s.x[rs1] + imm+1]) << 8) + \
                                      s.mem[s.x[rs1] + imm];   _pc(s)
def LHU(s,rd,imm,rs1): s.x[rd] =     (s.mem[s.x[rs1] + imm+1]  << 8) + \
                                      s.mem[s.x[rs1] + imm];   _pc(s)
def LW (s,rd,imm,rs1): s.x[rd] = (_i8(s.mem[s.x[rs1] + imm+3]) << 24) + \
                                     (s.mem[s.x[rs1] + imm+2]  << 16) + \
                                     (s.mem[s.x[rs1] + imm+1]  << 8)  + \
                                      s.mem[s.x[rs1] + imm];   _pc(s)

# store, note the different argument order, example: 'sb rs2, offset(rs1)'
def SB(s,rs2,imm,rs1): s.mem[s.x[rs1] + imm]   =  s.x[rs2] & 0xff; _pc(s)
def SH(s,rs2,imm,rs1): s.mem[s.x[rs1] + imm]   =  s.x[rs2] & 0xff; _pc(s); \
                       s.mem[s.x[rs1] + imm+1] = (s.x[rs2] >> 8)  & 0xff
def SW(s,rs2,imm,rs1): s.mem[s.x[rs1] + imm]   =  s.x[rs2] & 0xff; _pc(s); \
                       s.mem[s.x[rs1] + imm+1] = (s.x[rs2] >> 8)  & 0xff; \
                       s.mem[s.x[rs1] + imm+2] = (s.x[rs2] >> 16) & 0xff; \
                       s.mem[s.x[rs1] + imm+3] = (s.x[rs2] >> 24) & 0xff
[... 49 lines cut ...]

def _fsgn(s,rs1,msb): return b2f((msb & 0x80000000) | (f2b(rs1) & 0x7fffffff))
def FSGNJ_S (s,rd,rs1,rs2): s.f[rd] = _fsgn(s, rs1,  f2b(rs2));           _pc(s)
def FSGNJN_S(s,rd,rs1,rs2): s.f[rd] = _fsgn(s, rs1, ~f2b(rs2));           _pc(s)
def FSGNJX_S(s,rd,rs1,rs2): s.f[rd] = _fsgn(s, rs1, f2b(rs2) ^ f2b(rs1)); _pc(s)

def FCVT_S_W (s,rd,rs1): s.f[rd] = np.float32(   s.x[rs1]);   _pc(s)
def FCVT_S_WU(s,rd,rs1): s.f[rd] = np.float32(_u(s.x[rs1]));  _pc(s)
def FCVT_W_S (s,rd,rs1): s.x[rd] = np.int32  (   s.f[rs1]);   _pc(s)
def FCVT_WU_S(s,rd,rs1): s.x[rd] = np.uint32 (   s.f[rs1]);   _pc(s)
def FMV_W_X  (s,rd,rs1): s.f[rd] = b2f(s.x[rs1]);             _pc(s)
def FMV_X_W  (s,rd,rs1): s.x[rd] = f2b(rs1);                  _pc(s)
def FCLASS_S (s,rd,rs1):
  if   np.isneginf(s.f[rs1])    : s.x[rd] = 1       # negative infinity
  elif np.isposinf(s.f[rs1])    : s.x[rd] = 1 << 7  # positive infinity
  elif f2b(rs1) == 0x80000000   : s.x[rd] = 1 << 3  # -0
  elif f2b(rs1) == 0            : s.x[rd] = 1 << 4  # +0
  elif f2b(rs1) == 0x7f800001   : s.x[rd] = 1 << 8  # signaling NaN
  elif f2b(rs1) == 0x7fc00000   : s.x[rd] = 1 << 9  # quiet NaN
  elif (f2b(rs1) >> 23) == 0    : s.x[rd] = 1 << 5  # positive subnormal
  elif (f2b(rs1) >> 23) == 0x100: s.x[rd] = 1 << 2  # negative subnormal
  elif s.f[rs1] < 0.0           : s.x[rd] = 1 << 1  # negative normal
  else                          : s.x[rd] = 1 << 6  # positive normal
  _pc(s)
```

allenjbaum, 2022-12-15:
There is a plan to insert the Sail code for a function into the psec
document, and that might be enough for your needs>

NilsGraf, 2022-12-18:
Awesome, thank you!

I’m looking forward to it

NilsGraf, 2025-12-10:
Would it make sense to just copy appendix A from your book and be done with it?

http://riscvbook.com/spanish/guia-practica-de-risc-v-1.0.5.pdf