riscv/riscv-isa-manual #542: Wording Clarification on PMP Action

billhuffman, 2020-07-17:
The description of PMP action says: "Attempting to execute a store, store-conditional (regardless of success), or AMO
instruction whose effective address lies within a PMP region without write permissions raisesstore access-fault exception."

This might give the idea that the store-conditional could succeed in this case.  Of course it can't.  But it might avoid confusion if "regardless of success" were changed to "regardless of potential success"

aswaterman, 2020-07-17:
Or maybe just delete “regardless of success”?

David-Horner, 2020-07-17:
That's my preference.
An SC succeeds whenever the lock on the cacheline is retained, even if the
write will fail, so the old wording is ok though misleading and the
proposed wording is not specific enough to make a distinction about lock
success.

billhuffman, 2020-07-17:
Removing is fine with me too.

aswaterman, 2020-07-17:
OK, done.  Thanks for bringing this up.

commit 259025b4 (2020-07-17): Remove redundant phrase from access-/page-fault text

Resolves #542
closed 2020-07-17 by aswaterman