riscv/riscv-isa-manual #543: Machine mode only and user CSRs

jm4rtin, 2020-07-17:
Perhaps I missed it but does the privileged specification explicitly state that an implementation must implement user (or supervisor) mode in order for the CSRs in the user (or supervisor) space (bits 9:8 encoded as 00 for user CSRs) to be implemented? Put another way, if an implementation only has machine mode, should access to *any* user (or supervisor) CSR throw an illegal instruction exception?

Secondly, if there is no such hard rule, should the read-only shadows of the mcycle, minstret, mhpmcountr*n*, mcycleh, minstreth and mhpmcounter*n* CSRs: cycle, instret, hpmcounter*n*, cycleh, instreth and hpmcounter*n* respectively be accessible in a machine mode only implementation?

In section 3.1.12 the specification states:
> When the CY, TM, IR, or HPMn bit in the mcounteren register is clear, attempts to read the
cycle, time, instret, or hpmcountern register while executing in S-mode or U-mode will cause
an illegal instruction exception.

In this case the read or write to the read-only shadow CSRs has to be in machine mode (because there is no user mode) and so this does not apply; therefore, if the read-only shadow CSRs were implemented they should respond the same way as the machine mode version.

The question is, if an implementation is machine mode only (no user or supervisor mode), do the read-only shadow CSRs for the HPM exist or do they throw an illegal instruction exception? I can't find anything in the specification to clearly indicate the proper behavior.

aswaterman, 2020-07-17:
If you think of those CSRs as unprivileged CSRs, rather than U-mode CSRs, it becomes a bit clearer that such CSRs can exist even without U-mode.

The counters are actually an orthogonal extension described in the unprivileged ISA manual.  Since they're defined in that spec, it's implicit that they can exist irrespective of the privileged architecture configuration.

My recommendation would be that if you're providing mcycle, you should also provide cycle.  The cost is nearly nil, and it's easier for software.  But you can have a conforming implementation either way.

allenjbaum, 2020-07-18:
Another nice corner case for compliance.

Is the reasoning simply if they appear in the unprivileged spec they may (but NOT must) be implemented?

I’d be much happier if it were just one or the other since instead of implementation specific.
Now we have to configure the formal model to match the implementation behavior for each of the CSRs.
 I wonder what the formal model does for this? I’m sure it only does one or the other of the behaviors.

-Allen

aswaterman, 2020-07-18:
It’s not that they appear in the unpriv spec; it’s that the counters are a
separate extension (which should perhaps be named - Zicount?)

allenjbaum, 2020-07-19:
Ah. chapter 10 is a separate section in the spec for counters - but nowhere
does it indicate, or even hint,  that it is actually a separate extension
(and therefore isn't required to be implemented), as opposed to a base ISA
feature. Am I looking at an outdated version?.
It also says that the H versions that RV32I must implement to get to the
high half of the 64b cycle/time/instret counters are not required for RV64I
- which implies they are optional for RV64i, as opposed to being
prohibited. Why not simply prohibit them?

aswaterman, 2020-07-19:
The high parts should not be implemented in RV64I

allenjbaum, 2020-07-20:
There is still that word there - *should* not, or *must* not?
"Should" not doesn't change the meaning I am objecting to.

Also: If I'm reading unpriv spec chapter 9 correctly, Zicsr is specifically
is there for timer/counter CSR ops.
The CSR instructions themselves (CSRR[W|SC][ I] ) are defined nowhere else.
Since Mmode requires CSR ops, and Mmode is required, that seems to imply
that zicsr is always required, and thus so are the counter/timers.

I know better, of course, but if this is not the case, then this chapter
should be split into a section that describe only counter/timer CSRs, and
another with the description of the instructions.

If zicsr only refers to counter/timer CSRs, then tell me if my
interpretation below is flawed:
The D and F extensions and require CSRs, so
 - an implementation that has either D and/or F extension(s), but not
Zicsr is not compliant.
   if implemented, fflags, frm and fcsr are implemented

 - an implementation that has Umode and N-extension but not Zicsr is
compliant, because it has nothing to do with counter/timers
   if N-extension is implemented, ustatus, uie and utvec are implemented.
   If N-extension and  S-mode are implemented, then s[e/i]deleg are
implemented

This leaves only the umode counter/timer CSRs, which appear to be part of
the draft "Counters" extension (e.g. zicnt) and could
 - always be implemented if Umode is implemented, (in which case zicsr and
zicnt must be implemented if Umode is implemented), or
 - always be optional in which case
    -- Umode without zicnt could be legal or not
    -- Mmode without zicnt could be legal or not
    -- we need to have this officially ratified just to define the above 2
behaviors.

aswaterman, 2020-07-20:
Can't really say "must not" for this sort of thing, because of the possibility of non-conforming extensions.

The unprivileged spec explicitly admits the possibility of alternative privileged architectures, so "Zicsr is always required" isn't strictly speaking true.  If you restrict the goal of compliance testing to systems that implement the current RISC-V privileged architecture, then it is true that omitting Zicsr is noncompliant.

aswaterman, 2020-07-20:
BTW, RDTIMEH/RDCYCLEH/RDINSTRETH are described as "RV32I-only", which I believe addresses your concern.

allenjbaum, 2020-07-20:
My pendantic hat is one very tightly here: IF that is the case, why bother
to use the word "must" anywhere in the priv spec?
That word alone is used 73 times. Require is used at least 41 times, though
many of them don't mean "must" in context.
IF those registers are described as RV32i only, then why does the spec
merely say these " instructions are not required in RV64I" instead of
"these are RV32i only".
Those two statements are not equivalent.

The priv spec should be a description of the priv spec, not a possibly
non-conforming priv spec.

In other news: I said that the CSR ops and the timer/counters should be
separated. As I scroll further down, I see that, yea, they are. Oops.
MY confusion was because the spec says

"The counters and timers are no longer considered mandatory parts of the
standard base ISAs,
and so the CSR instructions required to access them have been moved out of
the base ISA chapter into this separate chapter"

That was what confused me. The same reasoning could be applied to FP CSRs,
but that wasn't sufficient to move them to a separate extension.

Are the "CSR instructions required to access them" specifically the ones
that address the counter/timers (and by extension, FP CSRs)?
Or are they all CSR ops?

If Mmode is required, and Mmode requires zicsr then  (like all the other
base ops) the implication is that zicsr is always required.
So the obvious question is why it needs to be a separate extension instead
of being part of the base spec, which is also always required?.
The statement above does not answer that question.

aswaterman, 2020-07-20:
> If Mmode is required, and Mmode requires zicsr then (like all the other base ops) the implication is that zicsr is always required. So the obvious question is why it needs to be a separate extension [...]

Already answered above: The unprivileged spec explicitly admits the possibility of alternative privileged architectures, so "Zicsr is always required" isn't strictly speaking true.

aswaterman, 2020-07-20:
> That was what confused me. The same reasoning could be applied to FP CSRs, but that wasn't sufficient to move them to a separate extension.

The first paragraph of the F extension chapter says that F requires Zicsr.

aswaterman, 2020-07-20:
> IF those registers are described as RV32i only, then why does the spec merely say these " instructions are not required in RV64I" instead of "these are RV32i only".

It _does_ say they are RV32I-only:
- In the priv spec: https://github.com/riscv/riscv-isa-manual/blob/40583f4fe0f53a2173577b5d035fcfadf8023941/src/priv-csrs.tex#L177
- In the unpriv spec: https://github.com/riscv/riscv-isa-manual/blob/40583f4fe0f53a2173577b5d035fcfadf8023941/src/counters.tex#L115 https://github.com/riscv/riscv-isa-manual/blob/40583f4fe0f53a2173577b5d035fcfadf8023941/src/counters.tex#L135 https://github.com/riscv/riscv-isa-manual/blob/40583f4fe0f53a2173577b5d035fcfadf8023941/src/counters.tex#L54

jm4rtin, 2020-07-20:
So if I implement just the standard stuff (rv32iZicsr) with only machine mode (I may have omitted some other extension or two that was important - sorry if I did) and wish to leave the privileged HPM CSRs tied to 0 does that mean I am implementing "Zicount" with none of the optional features? If I am implementing "Zicount" does that mean that both the privileged and unprivileged CSRs should exist? If I don't implement "Zicount" then neither privileged and unprivileged CSRs should exist? Or is there still the possibility of the privileged but not unprivileged CSRs existing in a "only standard" configuration?

I get that idea that certain extensions should be able to stand on their own and co-exist with other non-standard extensions (or variants of the standard extensions); however, if you're using just the standard extensions it would be good to have some clarity and a definitive determination on if these CSRs are implemented in some fashion or not instead of a "recommendation".

Silabs-ArjanB, 2020-07-27:
Above question of @jm4rtin has not been answered yet. My interpretation of above discussion is as follows (could somebody please confirm or correct these statements?):

- Zicnt (which for now is the name of the specification contained in Chapter 10 of ‘Unprivileged ISA’ spec) is an **optional** extension that deals **only** with CSRs 0XC00-0xC1f (and 0xC80-0xC9F for RV32).
- The ‘Machine Counter/Timers’  (i.e. CSRs 0xB00, 0xB02-0xB1F (and 0xB80, 0xB82-0xB9F for RV32I) are a **mandatory** part of every architecture that is ‘compliant to the current RISC-V privileged architecture’ (**independent** of whether Zicnt is included or not)

Assuming above is a correct interpretation, then I have a follow up question on what would be allowed for a compliant RV32**E** implementation. The mhmcounter[3..31](h) are mandatory but are clearly specified as WARL and are allowed to be implemented as ‘read as 0 always’. mcycle(h) and minstret(h) (i.e. 128 flip-flops and two associated 64-bit counters) are not explicitely defined as WARL and are therefore always required to be fully implemented and will therefore represent a significant part of the area of a compliant RV32E implementation; is that interpretation correct?

jscheid-ventana, 2021-08-10:
Is this stuck on "should" vs "must"?  Aren't non-conforming extensions definitionally violating or "overriding" some requirement, therefore it's appropriate to have extensions simply state their requirements clearly?

allenjbaum, 2021-08-10:
I interpret this as you stated: mcycle and minstret are required to be
compatible, assuming RV32i priv mode requirements carry over to RV32E.
If you don't want to be priv mode compatible, you can drop ziCSR and with
it mcycle, minstret, interrupts, traps, etc. etc. - or keep everything
except mcycle and minstret.

kersten1, 2024-07-03:
I'm closing this one in favor of issue #560 https://github.com/riscv/riscv-isa-manual/issues/560

Please reopen if the discussion needs to continue.