riscv/riscv-isa-manual #549: CSRRS/CSRRC definition

pdonahue-ventana, 2020-07-27:
Zicsr says this of CSRRS:
> Any bit that is high in rs1 will cause the corresponding bit to be set in the CSR, if that CSR bit is writable. Other bits in the CSR are unaffected (though CSRs might have side effects when written).

and this of CSRRC:
> Any bit that is high in rs1 will cause the corresponding bit to be cleared in the CSR, if that CSR bit is writable. Other bits in the CSR are unaffected.

The lack of symmetry with respect to side effects is strange since I assume that CSRRC can also have side effects.

More importantly, I assume that the statement that "other bits in the CSR are unaffected" is not intended.  For instance, consider these situations (which I assume are all indirect effects rather than side effects):
- if only M and U modes are supported then the privileged architecture suggests that it is valid to have only a single flop for mstatus.MPP (in which case setting/clearing one of the MPP bits will affect the other bit)
- if misa.D=misa.F=1 and then CSRRC clears only misa.F then the privileged architecture seems to require that misa.D must be cleared
- other WARL sanitization cases

Perhaps "other bits in the CSR are unaffected except via indirect effects or side effects"?

aswaterman, 2020-07-27:
The definition is supposed to be symmetric. The parenthetical is merely there for clarification and isn’t necessary for completeness. It probably makes sense to either remove it from CSRRS or add it to CSRRC.

pdonahue-ventana, 2020-07-27:
That makes sense.  And the prohibition on CSRRS/CSRRC changing bits other than those in the mask?  The requirements of the privileged architecture with respect to WARL are at odds with the requirements of the unprivileged architecture in this section.

aswaterman, 2020-07-27:
I think the text should say that "other bits are not explicitly written".  It's of course true that they could change because of the bits that are written.

pdonahue-ventana, 2020-07-27:
Sounds good to me.  Thanks.

aswaterman, 2020-07-27:
Fixed, thanks for bringing this up.

commit 8088ba47 (2020-07-27): Clarify effect on unwritten bits for CSRRS/CSRRC

They aren't _explicitly_ written, but that doesn't mean they are
unaffected, since they could change as a result of the bits that
are written.

Resolves #549
closed 2020-07-27 by aswaterman