riscv/riscv-isa-manual #550: Hypervisor: HFENCE.GVMA to TLBs that combine VS and HG translation?

ingallsj, 2020-07-30:
The current Hypervisor Extension v0.6.1 is silent about whether HFENCE.GVMA "applies to caching structures that combine VS and HG translation table entries" (to paraphrase).  Specifically, whether HFENCE.GVMA should also invalidate TLBs that cache the full translation from input VA to final PA.  Currently that silence implies that it **does**, which I suspect is **not** the desired architectural intent.

For comparison: see the section "Invalidation of TLB entries from stage 2 translations" in the ARMv8 ARM, which explicitly states that Stage-2-only `TLBI IPA*` do **not** apply to TLBs that combine Stage-1-and-2 translation table entries.

ingallsj, 2020-07-30:
@jhauser-us

jhauser-us, 2020-07-30:
Your first assessment was correct:  For implementations that cache the full translation from input VA to final PA and keep no other record of G-stage translations, an HFENCE.GVMA must invalidate all cached translations for a given virtual machine.  Avoiding the extra invalidations requires extra hardware, although not necessarily as much as you suggest.  Hypervisor authors would ideally be made aware that an HFENCE.GVMA instruction will, on many machines, result in a complete flush of all address translation caches for the specified virtual machine (or for *all* virtual machines, if none is specified), whether or not an address is specfied in operand *rs1*.

When the G-stage address translation is changed, I would be interested to hear what is the alternative to flushing the translation caches.  Allow the virtual machine to continue to use the old (now dangerously incorrect) G-stage mapping?

gfavor, 2020-07-31:
Note a couple of things about ARMv8's TLBI instructions:

The TLBI IPA* instructions say they are applicable only to TLB caching
structures holding stage 2 only entries.  For designs that have such stage
2 only caches, these instructions are obviously what you want
the hypervisor to use.  But for designs that cache combined stage1+2
translations, these instructions are useless and the hypervisor needs to
instead use the TLBI VMALLS12 instruction (the equivalent of HFENCE.GVMA).

For a hypervisor to support both types of implementations, it needs to know
which type of implementation it is running on (so that it can use IPA*
operations when relevant), or it assumes the worst and always uses the
VMALLS12 operation.

Greg

gfavor, 2020-07-31:
P.S. I would hazard to say that most implementations would cache combined
stage1+2 translations (versus doing two-step address translation through a
cascaded series of stage 1 and stage 2 TLBs).

But more notably, I misspoke in my last email.  Unlike TLBI VMALLS12 which
takes a VA argument, HFENCE.GVMA takes the equivalent of an IPA argument.
So they aren't really equivalent.

Also note that ARMv8's descriptions for invalidating "TLB entries from
stage 2 translations" actually involve performing both IPA-based and
VA-based TLBI's.  In essence ARMv8 approaches invalidating stage 2
translations in a more structured way (that doesn't force TLBI VMALLS12 to
always be used).

For example, to invalidate all cached copies of the stage 2 translation of
an IPA for the current VMID, one can do:

TLBI IPAS2E1        // invalidate stage 2 translations of an IPA
DSB
TLBI VMALLE1      // invalidate all stage 1 and 1+2 translations for a VMID

One point of this is convenience for (typical) implementations that have
both stage 1+2 translation caches, and stage 2 only "page walk" translation
caches for accelerating two-dimensional tablewalks.  Each type of cache is
normally looked up by a VA or IPA respectively, so having a TLB invalidate
that operates in a similar way for each of these caches is convenient.

This approach also provides flexibility so that, for example, to invalidate
all cached copies of the stage 2 translation of an IPA, but only those
stage 1 and 1+2 translations used to translate a given VA, then one can
instead do:

TLBI IPAS2E1       // invalidate stage 2 translations of an IPA
DSB
TLBI VAE1            // invalidate stage 1 and 1+2 translations of a VA
(and specific ASID)

This avoids flushing all the stage 1+2 translations for a VMID.

Lastly, it seems like a RISC-V equivalent of the ARMv8 approach would be to
have HFENCE.GVMA behave like the TLBI IPA* instructions and only apply to
GS-stage only translations.  Then a RISC-V hypervisor would use an
appropriate HFENCE.VVMA + HFENCE.GVMA combination for the situation at hand
- like what ARMv8 does.

Greg

jhauser-us, 2020-08-06:
> For example, to invalidate all cached copies of the stage 2 translation of an IPA for the current VMID, one can do:
>
> TLBI IPAS2E1        // invalidate stage 2 translations of an IPA
> DSB
> TLBI VMALLE1      // invalidate all stage 1 and 1+2 translations for a VMID

It would not be unreasonable for an implementation of HFENCE.GVMA to do exactly what's above, in one instruction.

> [...] one can instead do:
>
> TLBI IPAS2E1       // invalidate stage 2 translations of an IPA
> DSB
> TLBI VAE1            // invalidate stage 1 and 1+2 translations of a VA
> (and specific ASID)
>
> This avoids flushing all the stage 1+2 translations for a VMID.

When a hypervisor changes a valid guest page to now be invalid, or it changes the physical address of a valid guest page, I don't see how this combination of flushes can be accomplished, because the hypervisor doesn't normally have knowledge of the guest virtual addresses that must be invalidated.  When you say for TLBI VAE1, "invalidate stage 1 and 1+2 translations of a VA", what VA?  Where does the hypervisor get this virtual address?  There may in fact be any number of guest virtual addresses that must be invalidated; it's not necessarily just one.

The only situation I can conceive where the latter sequence makes sense is when the hypervisor changes what was an invalid guest page to now be valid.  If the hypervisor is sure that this guest page is not in the address translation caches as being previously valid (e.g. the guest page was never valid for this virtual machine until now), then it suffices for the hypervisor to invalidate the one guest virtual address that may have precipitated the change.  (The only reason we even need to invalidate would be in case the guest virtual page was cached as being *invalid*, and we need to remove this indication from the address translation cache.  As before, there could be any number of guest virtual addresses that need invalidating, but here we can deal with the others one-by-one in response to guest-page faults.)

I'll confer with some of the usual suspects about whether we should support optimizing this specific case.

gfavor, 2020-08-06:
On Thu, Aug 6, 2020 at 1:13 PM John Hauser <notifications@github.com> wrote:

> [...] one can instead do:
>
> TLBI IPAS2E1 // invalidate stage 2 translations of an IPA
> DSB
> TLBI VAE1 // invalidate stage 1 and 1+2 translations of a VA
> (and specific ASID)
>
> This avoids flushing all [...]
That makes sense - and equally questions what the hell was ARM thinking.  I
expect that there were valid reasons running through their minds - that
aren't immediately obvious here.  Maybe for certain type 1 or type 2
hypervisors?

The only situation I can conceive where the latter sequence makes sense is
> when the hypervisor changes what was an invalid guest page to now be valid.
> If the hypervisor is sure that this guest page is not in the address
> translation caches as being previously valid (e.g. [...]
This seems to become a question about speeding up the early to middle
life of a VM as it populates its guest memory.  If, for example, a VM spins
up to using say 32 GB of 4KB guest pages, then that would be 8M guest pages
that change from invalid to valid and each would require only an
hfence.vvma (presuming all translations for the VM's VMID were invalidated
when the VM was created), instead of a heavier weight hfence.gvma (heavier
on many, albeit not all implementations).  And this optimization doesn't
require any change to the current hfence instructions.

This is also an example of where caching of "invalid" translations is
undesirable performance-wise.  Ideally the hypervisor would know whether it
is running on such an implementation - and hence could skip doing any
invalidation in this type of situation.

Greg

jhauser-us, 2020-08-07:
We hope and expect that hypervisors will prefer to allocate in units of megapages, perhaps even gigapages.  So, instead of 8 million guest-page faults to allocate 32 GB in units of 4-kB pages, it would instead be 16 thousand faults to allocate in units of 2-MB megapages.

> This is also an example of where caching of "invalid" translations is undesirable performance-wise.

When a hypervisor needs to emulate a device, the pages containing the fake device will be marked *invalid* to generate guest-page faults.  If the invalidness of those pages isn't cached, the cost of device emulation will be increased, as every memory access to the device by the virtual machine must perform a page-table walk to rediscover the need to trap.  I don't know, but my guess is that these invalid guest-page faults could be more significant for performance than any losses incurred elsewhere from caching the invalidness of guest pages.

jscheid-ventana, 2021-08-10:
Anything to be done here?

aswaterman, 2021-08-11:
We're essentially committed to our current scheme at this point, and re-reading this thread hasn't persuaded me to the contrary.

closed 2021-08-11 by aswaterman