riscv/riscv-isa-manual #552: Extension state save/load missing?

alercah, 2020-07-31:
Section 3.1.6.6 claims that:

> Standard privileged instructions to initialize, save, and restore extension state are provided to
insulate privileged code from details of the added extension state by treating the state as an
opaque object

This seems like an important and useful feature, but also this statement appears, as far as I can tell, to be false. Is this a missing feature?

aswaterman, 2020-07-31:
The comment is meant to apply custom extensions, and so necessarily the instructions must themselves be custom instructions part of those extensions.  So, maybe the bug is that we say "are provided" when we should say "may be provided".

alercah, 2020-08-01:
Ah, that does make sense. However, that seems more covered by this:

> To improve
performance, the user-mode extension can define additional instructions to allow user-mode software
to return the unit to an initial state or even to turn off the unit.

My read of the sentence in my original comment was that there was intended to be a single global state mechanism that privileged code can use to store all user state, in order to be extension-agnostic, rather than needing to be aware of how to handle state for each extension it wishes to expose to userspace. But I can see definite issues with that.

sorear, 2020-08-01:
since S-mode does not make the decisions about what custom extensions are exposed to userspace, there's a big potential for covert storage channels here

aswaterman, 2020-08-01:
I think S-mode needs to be able to make decisions about what custom extensions with additional architectural state are exposed to userspace.  (It's hard to architect that in advance.)

aswaterman, 2020-08-01:
@alercah A single mechanism seems desirable, but it would probably have to be a set of SBI calls rather than a set of instructions, especially since multiple custom extensions with state could be implemented.

jscheid-ventana, 2021-08-10:
Is this somewhat addressed by smstateen?  https://lists.riscv.org/g/tech-privileged/message/572

alercah, 2021-08-10:
Looks like yes, excellent!

closed 2021-08-11 by aswaterman

alercah, 2022-07-24:
I think this should be reopened; I had my attention drawn back to this, and ran into this paragraph:

> Supporting substantial extensions is one of the primary goals of RISC-V, and hence we define a
standard interface to allow unchanged privileged-mode code, particularly a supervisor-level OS, to
support arbitrary user-mode state extensions.

This statement appears to be simply false in the current spec, and the smstateen proposal as I understand it will not solve this design goal, as that proposal would still require the supervisor or hypervisor to add explicit support for each extension. The only way to meet this design goal of arbitrary user-mode extensions without supervisor support (or arbitrary VS/VU extensions without hypervisor support) is a general facility for saving and loading context from registers that the super/hypervisor cannot refer to because it does not know what they are.

sorear, 2022-11-03:
The section you're quoting, in its original version, said that it was talking about SBI calls, not necessarily hardware instructions, as such the place for reopening this is on the SBI repository.  The world switch extension I've been working on would address some of this but not as its primary purpose.