riscv/riscv-isa-manual #553: Hypervisor guest interrupts are confusing

alercah, 2020-07-31:
It took me a long time to understand hypervisor & guest interrupts. It might be useful to have an overview someplace that explains the intended usage, particularly of guest external interrupts, which as far as I can tell the idea is that the hypervisor will normally configure the platform to send external interrupts to a particular guest, set `VGEIN` to correspond to the currently-running guest, and clear the corresponding bit in `hgeie`, to allow those interrupts to be handled directly by the currently-running guest. Is this correct?

It also took me a long time to understand this sentence:

> The active bits of hip and hie cannot be placed in HS-level’s sip and sie because doing so
would make it impossible for software to emulate the hypervisor extension on platforms that do
not implement it in hardware.

I think now that this is referring to M-level emulation, where writes to the hardwired zeroes in `hip` fields would fail to cause traps, and the M-level would not be able to implement them. Is this correct? If so, should it possibly be explained a bit more?

jscheid-ventana, 2021-08-10:
Were you looking for something like the content in "Interrupts for Virtual Machines (VS
Level)" in https://github.com/riscv/riscv-aia/releases/download/0.2-draft.24/riscv-interrupts-024.pdf?

alercah, 2021-08-10:
It's been so long I don't remember, and I haven't had reason to come back to this. Probably?

aswaterman, 2021-08-11:
Good enough for me.

closed 2021-08-11 by aswaterman