riscv/riscv-isa-manual #554: Hypervisor trap instruction translation is ill-defined on RV32E

alercah, 2020-08-01:
On RV32E, the upper registers values `x16`-`x31` do not exist, freeing up one bit in each instruction encoding per register argument. This means, in particular, that standard loads and stores, both of which store the base address in *rs1*, never have bit 19 (the upper bit of *rs1* set).

The process for instruction transformation in `mtinst` and `htinst` set out in 5.6.3 of the privileged specification transforms a standard load or store into an instruction with an address offset in bits 19-15, relying on the fact that this transformation always produces a valid standard instruction as set out above in the explanation of the meaning of the value. This represents the difference between the faulting address and actual base address. If, somehow, the high bit of this offset were set, the resulting encoding would *not* be a standard instruction, and therefore the encoding would incorrectly represent a custom instruction.

In practice this is incredibly unlikely to be ever encountered as it requires:

- An embedded system.
- With hypervisor support.
- And support for unaligned loads or stores of at least 17 bytes, say vector extensions with unaligned operation support.

Although on that last point, an unaligned large vector load of 128 bytes could potentially trigger a page fault that overflows Addr. Offset, and that's possibly a bigger conern.

Perhaps only slightly less unrealistically, a custom extension on an embedded, hypervisor-enabled system could validly encode its custom value to one of these values instead, which could be misinterpreted by code not written to ensure bit 19 is 0 in embedded code as a transformed standard instruction.

I think the easiest solution is to explicitly define that, in the weird RV32EH architecture, for the purposes of `htinst` and `mtinst` encoding, the high bit of a register can be 1 and it will still be considered a valid instruction.

jscheid-ventana, 2021-08-10:
@jhauser-us

jhauser-us, 2021-08-17:
Addressed by #713.

aswaterman, 2021-08-17:
Closed by #713

closed 2021-08-17 by aswaterman