riscv/riscv-isa-manual #555: Check for Z* extensions?

stnolting, 2020-08-04:
I wonder if there is an "official" way to check for `Z*` CPU extensions (like `Zicsr` or `Zifencei`)?!

Executing a certain instruction (like `fence.i` from the `Zifencei` extensions) and checking for an illegal instruction exception works fine, but is there an "official" way to check for these extensions? Or is there something planned? Like a Z-specific version of the `misa` CSR that shows all implemented `Z*` extensions?

sorear, 2020-08-04:
Any opcode other than all-zero or all-ones can be repurposed for a nonstandard extension, so you cannot assume unimplemented anything will result in an illegal instruction exception.

stnolting, 2020-08-04:
If I have an architecture where `misa.x` is zero (so no non-standard extensions), I *can* assume that all non-implemented instruction trigger an illegal instruction exception, right?

Anyway, if I cannot assume that unimplemented instruction/extensions/whatever trigger an illegal instruction exception, doesn't this require the need for a standardized method of checking available `Z*` extensions? For example via an CSR, that lists all officially specified `Z*` extensions?

allenjbaum, 2020-08-04:
If a bit is set, you can assume that it exists.
 I seem to recall that MISA is allowed to be all zero- in which case you can assume nothing. I have been told that the X bit may not always be set even if an implementation has custom extensions.

And ... you cannot depend on illegal op traps for much of anything, even if there are no custom extensions. They might alias to an existing op, they might do nothing.

There is a task group that is starting that is supposed to define a method for interrogating what is implemented at some level beyond MISA, but which details (and whether it is even mandatory) is TBD.

-Allen

pdonahue-ventana, 2020-08-11:
> If I have an architecture where `misa.x` is zero (so no non-standard extensions), I _can_ assume that all non-implemented instruction trigger an illegal instruction exception, right?

Section 2.2 of the architecture says: "The behavior upon decoding a reserved instruction is UNSPECIFIED."

jscheid-ventana, 2021-08-10:
https://github.com/riscv/configuration-structure

and

Smdisc [https://lists.riscv.org/g/tech-config/message/479]

are attempting to resolve this.

stnolting, 2021-08-11:
@jscheid-ventana Thanks for the information!
I already saw recent development that goes in this direction in #697

lbmeng, 2021-08-14:
Current latest privilege spec is missing an important part on probing those Z* standard extensions which will makes software crazy. Glad we will have a solution.

aswaterman, 2021-08-16:
It's not exactly "missing".  It's currently being debated within RVI.

stnolting, 2024-03-29:
I think this has been answered, e.g. https://github.com/riscv/riscv-isa-manual/issues/414#issuecomment-859706272

closed 2024-03-29 by stnolting

nihui, 2024-03-30:
Hello everyone, if you want to implement detection on an old version of the kernel, the ruapu project is a good reference. It handles sigill and completes the detection of the extended instruction set.

https://github.com/nihui/ruapu

sorear, 2024-03-30:
That cannot possibly work without the `Sstrict` extension which guarantees that you will get a SIGILL for unimplemented standard instructions, which means it doesn't solve this problem in any way. The base ISA without `Sstrict` makes **no guarantees whatsoever** about behavior when executing an opcode sequence corresponding to an unimplemented standard instruction since a non-standard instruction may reuse the free opcode sequence. Re-read chapter 30 "Extending RISC-V" of the specification.