riscv/riscv-isa-manual #556: Temporal locality HINTs

aswaterman, 2020-08-07:
It's sometimes desirable to express that a load or store has poor temporal locality, so that the microarchitecture can optimize its use of the memory hierarchy accordingly.  Temporal locality hints are primarily of interest to the vector extension, but this mini-proposal defines them in such a way that they can apply to scalar memory accesses, too.  Remote AMOs for contended synchronization variables are one obvious application of the scalar form.

Limited encoding space makes it impractical to add a locality hint field to the various loads and stores.  If we went that route, we'd either need to use 48-bit instructions or greatly restrict the addressing modes (e.g., register-indirect only, and, for vectors, unit-stride only).  So the proposal is to immediately precede the load or store with a HINT instruction, which essentially acts as a fusable prefix.

For now, I'll call it `ntlh`, for "non-temporal locality hint".  It takes no arguments.

We should provide a 16-bit variant to reduce impact on code size: perhaps `c.nop 1`.  I'm not sure whether a 32-bit form is really necessary.  If it is, we probably don't want to use the natural expansion of `c.nop 1`, which is `addi x0, x0, 1`, because we'd like to preserve those I-type HINTs for things that use the whole 12-bit immediate.  An alternative would be to choose `c.mv x0, x3` and its natural expansion `add x0, x0, x3`.  (x3/gp is almost never written so spurious interlocks won't be a problem.)

gfavor, 2020-08-07:
This is a great idea, especially given its ready applicability to all load/store instructions present and future.  And this is a well established idea, i.e. of having non-temporal load/store instructions in an ISA.

To me a 16-bit hint variant would be sufficient.  Low-end embedded designs will often be using the C extension.  Equally Linux class processors will be using the C extension since RV64GC is the architectural baseline.

tariqkurd-repo, 2020-08-07:
does this mean every non-temporal load/store will need a prefix? if so it really should be 16-bit, and maybe in the future we can fuse the two into a single 48-bit instruction.

aswaterman, 2020-08-07:
@tariqkurd-repo Yeah, there's no debate as to whether we should supply a 16-bit variant; the only question is whether a 32-bit prefix should be provided in addition so that the feature can be used without the C extension.  I tend to agree with @gfavor that we can skip the 32-bit prefix.

David-Horner, 2020-08-07:
I too would lean toward miserly allocation of hints.
An implementation that requires a 32 bit variant has numerous options.
- It can use 32 bit custom space.
    32bit only machines are envisioned for custom experimentation after all. 
- It can use any 32 bit hint custom or reserved.
   This would be specific to the implementation after all.
- It could make an exception for a pair of ntlh to be considered the 32 bit variant.
    This is somewhat at odds with a design using 32bit only code to simplify the decoder, however the alignment matters are still eliminated.

The big advantage to 16bit only is simplifying the software stack support. Only a single construct is needed.

The big disadvantage to 16bit only is the complication of the current software stack for C extension support which supports this as a separate compression pass.
I suggest the first stage provides a pair of ntlh hints, which the compression pass reduces to a single one if it wants to. (the pair of hints would still work). I would expect the inclusion of the hint would be a compile option that the ILEN32 only target would avoid.
There could be a clean up pass to convert the pair to a standard 32bit nop for ILEN32.

allenjbaum, 2020-08-07:
"32bit only machines are envisioned for custom experimentation after all."
This is news to me.

David-Horner, 2020-08-07:
On Fri, Aug 7, 2020, 14:27 Allen Baum, <notifications@github.com> wrote:

> "32bit only machines are envisioned for custom experimentation after all."
> This is news to me.
>

Somewhat tounge in cheek, but....
The early idea was that research especially in universities and undergrad
courses would use the 32bit without C extension to facilitate chip design
unencumbered by instruction halfword alignment and parcels.
Of course other applications are possible, however the early design
envisioned C and reserved the opcode space for it, intentionally
restricting immediate size among other aspects to allow for it. The
evolution of lui and auipc in particular is indicative of the prior
commitment to the C extension and the accommodations made to support it in
practice.

>

gfavor, 2020-08-07:
On Fri, Aug 7, 2020 at 4:48 AM David-Horner <notifications@github.com>
wrote:

> The big disadvantage to 16bit only is the complication of the current
> software stack for C extension support which supports this as a separate
> compression pass.
>
Is the current software stack not able to understand instructions that
happen to be 16-bit in length (before the compression pass)?  This
instruction in any case is not going to be generated by a compiler, but
presumably is introduced into the code via assembly code or use of an
intrinsic.  The compiler itself doesn't care about the instruction length
and, with or without use of the compression pass, the assembler and linker
should happily handle this 16-bit instruction mixed in with all the other
32-bit instructions.  (Or am I missing something?)

Although is that maybe the wrong way to view the matter.  It seems like the
toolchain needs to view the "hint; ld/st" instruction pair as a single
pseudo-instruction (for lack of a better term) that later (at assembly/link
time) is expanded into a 48-bit "instruction" or into a pair of 16b+32b
instructions (the same end result in any case) - so that the hint and ld/st
parts of this thing are kept together.  Or alternatively the "hint; ld/st"
instruction pair are understood by the compiler as a potentially
fuseable pair of instructions that are kept together through all the
compiler's optimization passes.

Put differently, imagine there is no C extension and we're simply adding
this one 16-bit instruction to the ISA.  What in the software stack would
have a problem with supporting that?

Greg

>

David-Horner, 2020-08-07:
Re: [riscv/riscv-isa-manual] Temporal locality HINTs (#556)

>
> > The big disadvantage to 16bit only is the complication of the current
> > software stack for C extension support which supports this as a separate
> > compression pass.
> >
> Is the current software stack not able to understand instructions that
> happen to be 16-bit in length (before the compression pass)?
correct.
> This
> instruction in any case is not going to be generated by a compiler, but
> presumably is introduced into the code via assembly code or use of an
> intrinsic. 
These too currently only know ILEN=32.
> The compiler itself doesn't care about the instruction length
The simplification of 4bytes per instruction, aligned on 32 bit 
boundaries is significant
> and, with or without use of the compression pass, the assembler and linker
> should happily handle this 16-bit instruction mixed in with all the other
> 32-bit instructions.
Perhaps they would be overjoyed, but currently the blissful state of C 
awareness is not (significantly) coded.
> (Or am I missing something?)

Because the C extension has been designed so well that the instructions 
correspond to a single base RVI instruction,
  there has been no need for the front end to be C aware.
This allows rapid implementation of the toolchain with substantial 
effectiveness of the Compressed code via replacement in a near the end pass.
Up until now the pressure to do better has not been sufficient to make 
the early components C aware, although there has been much desire to do so.
Alex can speak to this much better than I.

> Although is that maybe the wrong way to view the matter. It seems like the
> toolchain needs to view the "hint; ld/st" instruction pair as a single
> pseudo-instruction (for lack of a better term) that [...]
Yes. there is a need to keep the prefix and its partner in order and 
together.
    Multiple approaches are possible to do this.
> Or alternatively the "hint; ld/st"
> instruction pair are understood by the compiler as a potentially
> fuseable pair of instructions that are kept together through all the
> compiler's optimization passes.
It is entirely possible for the compiler/assembler to refer to a 
psuedo-instruction that is this instruction-pair.
 From my cursory reading of LLVM components this is not uncommon.
>
> Put differently, imagine there is no C extension and we're simply adding
> this one 16-bit instruction to the ISA. What in the software stack would
> have a problem with supporting that?
The support is not currently there, and as I understand it,
it would be a big internal change for both LLVM and gcc (and thus for 
other home growers).

gfavor, 2020-08-07:
On Fri, Aug 7, 2020 at 12:54 PM David-Horner <notifications@github.com>
wrote:

> It is entirely possible for the compiler/assembler to refer to a
> psuedo-instruction that is this instruction-pair.
> From my cursory reading of LLVM components this is not uncommon.
>

Thanks for the explanation.  Is the preceding suggesting that a
pseudo-instruction approach would avoid all the complications that you
mentioned wrt the current software stack, i.e. that it happens late enough
in the overall flow?

If so, then that would enable us to only define a 16-bit 'ntlh'
instruction.  Otherwise it sounds like it will be very desirable (to say
the least) to have a 32-bit version of the instruction as well.

Greg

sorear, 2020-08-07:
> The support is not currently there, and as I understand it, it would be a big internal change for both LLVM and gcc (and thus for
other home growers).

I am extremely skeptical of this argument since both llvm-mc and gas are multitarget assemblers which support many variable-width instruction sets, e.g. x86, Thumb, s390x, microMIPS, POWER (VLE but also prefixed instructions are handled as single 8-byte instructions by LLVM), etc.  Can someone actually familiar with the code confirm or deny?

David-Horner, 2020-08-07:
> Thanks for the explanation. Is the preceding suggesting that a
> pseudo-instruction approach would avoid all the complications that you
> mentioned wrt the current software stack, i.e. that it happens [...]
No. Even then current assumption is all 32 bit opcodes.

>
> If so, then that would enable us to only define a 16-bit 'ntlh'
> instruction. Otherwise it sounds like it will be very desirable (to say
> the least) to have a 32-bit version of the instruction as well.
Except we can simulate the 32bit opcode by providing the encoding of two
successive idential 16bit codes.

David-Horner, 2020-08-07:
I am confident Alex Bradbury can.
I have been following his LLVM Weekly updates. I am particularly interested
in the C aware efforts and I would be surprised if I missed any significant
advances. But it is certainly possible.

gfavor, 2020-08-07:
On Fri, Aug 7, 2020 at 1:14 PM David-Horner <notifications@github.com>
wrote:

> Except we can simulate the 32bit opcode by providing the encoding of two
> successive idential 16bit codes.

If I understand correctly, then there would be a 32-bit ntlh "instruction"
(encoded as you described) that the toolchain understands all the way down
to the compression pass - which then eliminates one of the 16-bit ntlh's
and leaves the final 48-bit instruction pair in place.  Along the way the
compiler would need to understand to keep these two 32-bit "instructions"
together.  And if the compression pass was bypassed for whatever reason,
the 64-bit end result would still execute fine (i.e. the first ntlh would
effectively be ignored).

Yes?

Greg

David-Horner, 2020-08-07:
On 2020-08-07 4:29 p.m., gfavor wrote:
> On Fri, Aug 7, 2020 at 1:14 PM David-Horner <notifications@github.com>
> wrote:
>
> (to say
>
> If I understand correctly, then there would be a 32-bit ntlh "instruction"
> (encoded as you described) that the [...]
correct.
>
> Greg
>
> —
>
While we wait on confirmation this strongly implies what I believe to be 
true in LLVM at least:

  [RISCV] Compress instructions based on function features

Jan 24, 2020.

When running under LTO, it is common to not specify the architecture
spec, which is used for setting up the target machine, and instead rely
on features specified in each function to generate the correct
instructions.

This works for the code generator, but the RISC-V backend uses the
AsmPrinter to do instruction compression, which does not see these
features but instead uses a MCSubtargetInfo object to see whether
compression is enabled. Since this is configured based on the
TargetMachine at startup, it will result in compressed instructions not
being emitted when it has not been given the 'c' TargetFeature, but the
function has it.

This changes the RISCVAsmPrinter to re-initialize the STI feature set
based on the current MachineFunction, such that compressed instructions
are now correctly emitted regardless of the method used to enable them.

jim-wilson, 2020-08-07:
On Fri, Aug 7, 2020 at 1:12 PM sorear <notifications@github.com> wrote:
> The support is not currently there, and as I understand it, it would be a big internal change for both LLVM and gcc (and thus for
> other home growers).
>
> I am extremely skeptical of this argument [...]

I don't know about llvm-as, but for GNU as it just maps a string to an
encoding.  There is no need for a 16-bit instruction to have a 32-bit
equivalent.  The only issue would be if the operands don't fit, then
we can't back off and use a 32-bit instruction instead, but have to
emit an error.  But we already do exactly the same thing if you use a
compressed opcode like c.add.  If you use the opcode add, then we emit
a compressed instruction if possible, and a 32-bit instruction
otherwise.  If you use c.add, then you get an error if the compressed
instruction is not possible.  It is the user's responsibility to get
it right if they use c.add.

Jim

sorear, 2020-08-07:
I just added support for a 16-bit instruction with no 32-bit counterpart to LLVM, with full support in the assembler (llvm-mc), disassembler (llvm-objdump), and inline asm in clang.  It took me 11 minutes, including build time and the time I spent figuring out the arguments to llvm-mc which are apparently very different from gas.  I have never modified the LLVM assembler before.

```
diff --git a/llvm/lib/Target/RISCV/RISCVInstrInfoC.td b/llvm/lib/Target/RISCV/RISCVInstrInfoC.td
index f687678..5c382b6 100644
--- a/llvm/lib/Target/RISCV/RISCVInstrInfoC.td
+++ b/llvm/lib/Target/RISCV/RISCVInstrInfoC.td
@@ -576,6 +576,12 @@ def C_UNIMP : RVInst16<(outs), (ins), "c.unimp", "", [], InstFormatOther>,
   let Inst{15-0} = 0;
 }
 
+let hasSideEffects = 1, mayLoad = 0, mayStore = 0 in
+def C_ASDFTEST : RVInst16<(outs), (ins), "c.asdftest", "", [], InstFormatOther>,
+              Sched<[]> {
+  let Inst{15-0} = 4;
+}
+
 } // Predicates = [HasStdExtC]
 
 //===----------------------------------------------------------------------===//
```

aswaterman, 2020-08-07:
Once upon a time, it was true that GAS didn’t have explicitly compressed instructions, but as Jim points out, it’s not at all a limitation anymore. LLVM can also definitely be made OK with this.

The decision to include a 32-bit version of the hint really only needs to be made on the basis of supporting the hint in systems without RVC.

sorear, 2020-08-07:
I think we should have a general rule that when supporting extensions that nominally have 16-bit or 48-bit instructions, if C is not also enabled they must be padded to a 4-byte boundary with a c.nop.  This came up years ago in the context of the L extension, although I forget which forum.

David-Horner, 2020-08-07:
@sorear Excellent.

Do you know if LLVM pads a nop to ensure 32bit alignment for the other RISCV instructions?
Is it before or after c.asdftest? 
For c.ntlh after would be problematic, unless the nop were removed during the C compression pass.

The ability to place such a 16bit instruction is of considerable benefit to proposed C enhancement for multiple register stores load/pops. The list discussion about custom instructions to reduce code size.

sorear, 2020-08-07:
No, it just leaves the instruction stream unaligned. That would have to be added, but it's a logically separate feature and only needed to support a narrow class of cores.

kasanovic, 2020-08-08:
All 16-bit instructions should have a 32-bit equivalent to support ILEN=32.  If it can fit/be justified as a 16-bit encoding, it is easy to fit/justify the 32-bit encoding. This avoids having to later design an ILEN=32 equivalent (and having to repeat the above conversation on every 16b instruction addition).  The toolchain issues are not the issue, rather there are multiple reasons to not abandon/fragment ILEN=32 designs.

sorear, 2020-08-09:
If a hinted load or store traps, where does xepc point?

Does it point at the hint, forcing modification to all code which emulates loads and stores for alignment, virtual MMIO, or other reasons?  If the H extension is also implemented, what value is stored in xtinst?

Or does it point at the load/store proper, causing the load/store to be treated as temporal when restarting after a page fault?  If data-modifying cache maintenance operations are also implemented, can this have a different effect observable beyond timing?

David-Horner, 2020-08-09:
On 2020-08-09 3:25 a.m., sorear wrote:
>
> If a hinted load or store traps, where does xepc point?
>
> Does it point at the hint,
>
no.
>
> forcing modification to all code which emulates loads and stores for 
> alignment, virtual MMIO, or other reasons?
>
it is desirable to avoid this, and fortunately pointing at the "hinted" 
instruction can avoid all manner of complication.
>
> If the H extension is also implemented, what value is stored in xtinst?
>
It should be the same as without the hint.
>
> Or does it point at the load/store proper,
>
yes.
>
> causing
>
(it need not cause, the Uarch is the determinant of what happens)
>
> the load/store to be treated as temporal when restarting after a page 
> fault?
>
maybe.

It is allowed for the implementation to check the hint-prefix on return 
from the interrupt.

It is also allowed for an implementation to ignore the hint at any time,
   or to provide equivalent hint behaviour even when the hint 
instruction is not present.
For example on return from an exception,
   or when the Uarch has determined a pattern of behaviour that 
conflicts with the explicit hint or
      suggest the hint behaviour is warrented even in the absence of an 
explicit hint.
>
> If data-modifying cache maintenance operations are also implemented,
>
executed during the exception handling? (I talk to two readings of "are 
... implemented" below)
>
> can this have a different effect observable beyond timing?
>
The hint could change timing and cache behaviour sufficiently for 
observable effects on concurrent processes' access to the same 
data/cacheline/page.
This is precisely the intent of the hint.

Depending upon the "data-modifying cache maintenance operations" 
executed in an exception handler, anything is possible.
It is incumbent upon the exception handler to avoid any activity that 
disrupts the architecturally visible "normal" execution of the 
interrupted process.

Some disruption on an exception will occur, but the behaviour that a 
hint invokes cannot be beyond what is allowed for an implementation in 
the absence of an explicit hint.

Stated another way, an implementation may wrongly allow certain 
behaviour but a hint is not justification for such.

aswaterman, 2020-08-09:
Chill. The HINT is just a hint. It’s architecturally equivalent to executing a nop. Timing channels could’ve been effected through other means.

David-Horner, 2020-08-09:
@aswaterman re chill.
I see @sorear questions as valid and useful for clarification as the intent of this hint is as an instruction prefix.
It thus conjours the x86 prefix byte managary with its unintended consequences and complications.
It is good  to fully understand the implications.
Without raising plausible and challenging alternatives clarity and deep understanding rarely occurs.
Scrutiny of the intended affect is neccessary to ensure a hint is just a hint.

gfavor, 2020-08-09:
I'll take a stab at arguing for the acceptability of the "compromise"
approach, i.e. trapping just the hinted instruction.

If hinted instructions are trapped commonly, then performance is going to
really, really suck due to the trap cost.  At which point ignoring the hint
on return is not going to materially/qualitatively change the matter.

Conversely, if such traps are rare enough to be acceptable
performance-wise, then the impact of the compromise should also be
acceptable.

Greg

sorear, 2020-08-09:
None of my questions were directed at David-Horner and I still think they are important questions for the privileged architecture and CMO people.  We specified caches to be 100% software transparent in the originally frozen specs for a reason…

allenjbaum, 2020-08-10:
A question regarding the hint: Does the hint apply to the next Load op, or to the next sequential instruction?

If it is really important, then the handler could return to xEPC-2 or -4 instead of xEPC

-Allen

sorear, 2020-08-10:
There is no way for a handler to distinguish between a load/store that happens to be prefixed and a load/store that is the first instruction in a function and follows rodata that has the same bit pattern as a prefix, or perhaps a U-type instruction the last 16 bits of which happen to match the prefix.  We hit this already with trying to use prefixes to distinguish semihosting ebreaks from hosted debugging ebreaks.

aswaterman, 2020-08-10:
Even if it were possible, it wouldn't be worth making the ISR go out of the way to return to the HINT instead of the load/store.  One additional cache miss following an exception is not a big deal.  The HINT is useful even if it only works in the common case.

The code-generation recommendation will be to place the HINT immediately before the load or store, so that it can be fused.  Under those constraints, it wouldn't matter much if it were defined to affect the next instruction only or the eventual next memory-access instruction.  In fact, different machines could legitimately employ either strategy, since there's no architecturally visible effect.

sorear, 2020-08-10:
They're hints now but if https://github.com/riscv/riscv-CMOs is allowed to complete it's all going to become architecturally visible.

aswaterman, 2020-08-10:
The HINT will only be defined to affect cacheable idempotent memory, in which case the presence of CMOs doesn't really architecturally expose the HINT: anything the HINT did could've been done by the processor's microarchitectural speculation mechanisms, without any dependence on the instruction stream.

David-Horner, 2020-08-12:
> An interesting note: The 267264 custom designated 32bit hints are
> about 10th the reserved, whereas the 111 custom RVC comprise
> approximately 40%.

> All 16-bit instructions should have a 32-bit equivalent to support 
> ILEN=32.
I believe this requirement is unnecessarily restrictive; ILEN=32 systems 
may not needed this support.

Mandates simplify considerations, but limit possibilities.

> If it can fit/be justified as a 16-bit encoding, it is easy to 
> fit/justify the 32-bit encoding.
This is obviously currently true as there are many more 32-bit HINT 
formulations than RVC HINTs .

Reserved for future use in RV32, the numbers are 2629662 and 283 
respectively.

So fitting into a 32-bit encoding is not a challenge, however, 
justifying a specific code is another matter.

Selecting any 32-bit code ignores the pressure/considerations unique to 
each set of hints. As we can see this is non-trivial:

  * By far the largest are LUI and AUIPC with 2^20 hints each. Each has
    an immediate encoding equal to their number of hints.  The available
    use for such an encoding should leverage the extensive range that
    such an instruction has, a range that is impossible to map into RVC
    reserved hints.
    Similarly the ADDI, ANDI, ORI and XORI based hints, with (approx) 2
    ^ 17 each, are unlikely to have mappings from RVC hints. Perhaps
    special cases of them might map into a RVC hint, but still unlikely.
  * the 31 FENCE hints are expected to be exclusively for WMO use.
    Possible, but unlikely candidates for RVC hints.
  * ADDW, SUBW, SLLW, SRLW and SRAW are RV64 only and are similarly
    expected to be exclusively for Word specific hints (whatever that
    might be)
  * That leaves the more generic RV32 and RV64 shared ADD, SUB, AND, OR,
    XOR, SLL, SRL and SRA hints, 2 ^ 13 in total.

Even within this latter large group, some are more desirable for use 
than others.

  * ADD X0,X0,X0 is already used as an internal alternate expansion of
    C.NOP.
  * It makes sense then, all other matters being equal to allocate from
    ADD for RVC equivalent hint one offs.
  * As Andrew shared, not all register usage, even in hints are equal.
    X0 is least involving (as would X9- X15 on RV32E).
  * as a result for RV32 the remaining 7 hints in the form X0,X0,X0 are
    least impactful.
  * Of the "normal" registers, Andrew identifies X3/gp as low impacting
    as by convention if is setup once for a thread.
  * Each of these 8 HINTs opcodes has a next optimal 3 register sets (
    X0,X3 : X3,X0 ; and X3,X3 ), meaning only 24 further very good variants.

Thus the issue must examine whether C.NTLH warrants one of the remaining 
32bit hints: from the least involving 7, any of the 24 very good, one of 
the many less optimal; or none at all.

> This avoids having to later design an ILEN=32 equivalent
This presumes that an ILEN=32 equivalent will be needed, and that no 
preferred alternative to a hint exists if a ILEN=32 expression is needed.

I this case I propose that the alternative that Andrew suggested, a 
64bit load/store variant, may be better for ILEN=32.

A judicious selection of encoding would provide for minimal bit 
checking, allow the first word as a prefix, in this case as a hint.

Such encapsulating 32bit ops within 64bit opcodes has been proposed 
previously (as additional pages of opcodes, which could just mean hinted) .

[... 50 lines cut ...]
> The toolchain issues are not the issue,

I believe toolchain issues continue to be relevant to this concern.

I mention LTO above, and as compiler because more RVC aware (and not 
just 16 opcode tollerant).

> rather there are multiple reasons to not abandon/fragment ILEN=32 designs.

ILEN=32 ubiquity is not one of the compelling arguments. Indeed, RVC is 
and is expected to remain dominant in the industry.

Applications that are ILEN=32 are niche and most likely have custom 
extensions (which could include a custom 32bit equivalence to a 16bit hint)

Further, implementations that are ILEN=32 appear to target trivial 
systems, thus are less likely to have a cache and thus need for a 
temporal use hint.

In principle I agree ILEN=32 should not be abandoned nor fragmented, but 
that does not mean that 32bit equivalence for 16bit ops is needed to 
accomodate that objective.

We may be at a point where more creative actions will provide better 
support than token allocations.

David-Horner, 2020-08-15:
>
>
>     Even if it were possible, it wouldn't be worth making the ISR go 
> out of the way to return to the HINT instead of the load/store. One 
> additional cache miss following an exception is not a big deal. The 
> HINT is useful even if it only works in the common case.
agreed.
>
>     The code-generation recommendation will be to place the HINT 
> immediately before the load or store,

These targetted hints anticipate three characteristics:
     That the default behaviour is opposite to (or significantly 
different from) what the hint requests.
     That they will propogate to the intended instruction
     That they will be bounded;
               definitely limitted in their scope to the next 
targettable instruction.
               optionally limitted to some other characteristic, such as 
number of executed instructions.

If any of these characteristics are uncertain, then the motivation of 
programmers to use them diminishes.

     Programmers will be adverse to using such a hint if they cannot 
rely on avoiding
         - frequent misses (that the intended target does not recieve 
the hint)
         - frequent collateral damage (that unintended targets recieve 
the hint)

     A hint that is not bounded negates the first benefit.
              (unless such hints were to toggle requested state, which 
may become untenable).

  Also, so its effects can be limitted.

To place the HINT immediately before the load or store is workable as 
long as there is only one hint applicable to the next instruction.

An ordering problem arises If two or more  hints of the "applies (only) 
to the next instruction of this prescibed class" are intended to apply 
to the "next" instruction (which matches these prescribed classes). If 
any of them are not implemented in this hart, then those should preceed 
the ones that are implemented.

But, in general, software is unaware of the implemented hints on a hart.
Further,  in general, a hart is unaware of the existance of hints; it 
executes all currently defined hints as either completely with no 
architecturally observable results or as an explicit nop. As an extreme 
example consider MUL X0,X0,X0; it is a potentially future specification 
defined hint.

To reliably apply hints of this targetted nature, It appears that Uarchs 
will need to propogate the hint "directive" over at least instructions 
that affect no architecturally observable result.

> so that it can be fused.
As I see it this is a secondary consideration.
Optimizing the processing doesn't require fusing.
The hint doesn't affect architectural state, so it has no hazzards.
As with other noops it can be elided/eliminated from execution path.,
What may be required is a delay queue to activate the directive at the 
correct poimnt in the pipeline.
Then, for a read the "release/evict" cache line operation.
The write queue may need a "release/evict" tag until the write occurs.

_The hint as a read-ahead directive.

_  Temporal locality is often linked to a pattern of successive reads 
that are predicatable.

   If the hint for the next read were in program order in advance of the 
read, it could act as a hint to the hart that read ahead is allowed from 
this point without concern for side-channel attacks. If this were 
sufficiently in advance it could provide useful overlap of execution 
with the prefetch. In general, most Uarch designs would only recognize 
this after the first occasion of that hint with the affected read/write. 
But could be retained (in similar manner as branch predictipon is 
currently).

_Encoding multiple hont directives._

This read-ahead/prefetch could be made explicit.

In addition there are other characteristics for the read that could 
reasonably be hinted.
Specifically :
      sequential cache-line access
      sequential page access
      next access lower in memory (vs the presumption of increasing 
memory addresses)

These 5 charaacteristics (when including alllow prefetch, and temporal 
locality hint) could be encoded in the 5 bits of a C.ADD like instruction.
Alternatively, common combinations can be encoded into fewer bits.
The consolidation of multiple related hints into a single hint 
instructionwould save both space and decoding/intent tracking complexity.

aswaterman, 2020-08-15:
We should offer the clear recommendation that these HINTs immediately precede the corresponding load or store. 
 Beyond that, let's stop over-thinking this one.

jscheid-ventana, 2021-08-10:
Further discussion:
https://lists.riscv.org/g/tech-unprivileged/message/166

kersten1, 2024-04-05:
Closing this one as I see that Zihintntl is in the Unpriv doc!

closed 2024-04-05 by kersten1