riscv/riscv-isa-manual #557: Fixing "N" sharp edges

sorear, 2020-08-08:
"N" remains an attractive nuisance for supervisor-capable cores (for instance it is on the NOEL-V roadmap) due to the virtualization and storage-channel issues in the current spec.  While I remain skeptical of the value of "N" and support satp.MODE=Bare as an alternative, I now believe that it is possible to repair the actively harmful portions of "N" with minimally consequential changes:

1. Fixing the virtualization issue:

   > `uret` may be executed at any privilege level; regardless of privilege level, it sets `pc` to `uepc` and `ustatus.uie` to `ustatus.upie`.  Unlike other _x_ ret instructions, `uret` never modifies the current privilege level.
   >
   > _Since `uret` does not read or modify any privileged state, including the privilege level, it has no effect on virtualization._

   (Currently uret is an untrappable instruction which drops to U-mode, so code written for S-mode can detect classical virtualization by executing uret, taking a trap, then checking whether the trap appeared to originate from S or U mode.)

2. Fixing the storage channel:

   > If the N extension is implemented in an implementation that also provides supervisor mode, it must be possible to disable the N extension by setting `misa.N` to 0.  While N is disabled, any attempted access to `uepc`, `ustatus`, `uscratch`, `utvec`, `utval`, `ucause`, `sideleg`, `sedeleg`, `uip`, `uie`, or attempted execution of the `uret` instruction, must raise an illegal instruction exception.  Platform initialization code must set `misa.N` to 0 prior to the first entry to supervisor code, and `misa.N` must remain 0 until a specific request to enable it is received from the supervisor via a SBI call or other platform-defined mechanism.
   >
   > _Disabling access to U-mode trap CSRs prevents storage channels between user threads with a supervisor that is not aware of the N extension.  Disabling access to the S-mode `sideleg` and `sedeleg` prevents a storage channel between virtual machines managed by a hypervisor unaware of the N extension._

   (This is essentially redundant with what the Unix platform spec says about "user-visible state", however the latter is extremely terse and frequently not read by hardware designers.  This is also largely redundant with what the functionality of setting NS=Off would be if NS were fully specified, but hardware designers do not seem to be waiting for us to specify NS.)

sorear, 2020-08-10:
Since I received a private email requesting clarification on this: "NS" refers to the hypothetical future feature described in https://github.com/riscv/riscv-isa-manual/blob/d9ea1ca369d6f1f8e61644b0bb87416a4f2857d8/src/n.tex#L232-L235 .

aswaterman, 2025-12-09:
It is likely that some user-level interrupt facility will eventually be defined, but it probably won't take the shape of what was previously described as `N`; and it definitely will not be named `N`, but will instead have an `Su` prefix.

> I don't know what, if anything, should be done about the broader issue.

Profiles, which were defined after the original post was made, address this issue to some extent.