riscv/riscv-isa-manual #558: pmp legal implmentation

neelgala, 2020-08-10:
I was adding support for pmpaddr csrs to riscv-config and had a question:

Is it a legal choice to implement pmp1cfg but not pmpaddr1 ? if so what is the benefit ?

sorear, 2020-08-10:
The most conservative reading of the spec is that an implementation is allowed to pick a number of PMP comparators (0-64) and a PMP address granularity (4 bytes - physaddr size, power of 2 only) and these two parameters determine everything else.  I do not understand "implement pmp1cfg but not pmpaddr1"; if any PMPs are implemented, all of them must be readable, but PMP registers above the implemented limit can be hardwired to zero.

Note that the recommendations for software probing only probe the two parameters above.  Allowing PMP implementations to vary in ways not captured by those parameters (for instance allowing holes in the PMP space, or allowing regions of specific exact sizes but not smaller or larger sizes) would cause the recommended probing procedure to fail, and should be forbidden.

neelgala, 2020-08-11:
I overlooked the commentary on sw discovery. Thanks for pointing it out.

closed 2020-08-11 by neelgala

allenjbaum, 2020-08-11:
Probably legal- PMPADDR is WARL and not implemented means the are zero.
I’d have to look at the exact definition, but in TOR mode, could that mean the entire address space?

But yes, legal but useless. It’s probably not the only bit of architecture like that.

-Allen