riscv/riscv-isa-manual #561: mcounteren in M mode only systems

tzwaenn, 2020-08-12:
mcounteren is kind of similar to the m?deleg - it is an M mode register about access of less privileged modes. For the m?deleg registers, the spec clearly states they should not exist if there is no mode to delegate to. No such provision is made for mcounteren, although in am M mode only system, it servers no pupose as no less privileged mode exists.
Shouldn't mcounteren (like the m?deleg) be specified as not existing in M mode only systems?

aswaterman, 2020-08-12:
Yes - no reason to have mcounteren without U-mode.

allenjbaum, 2020-08-12:
So, to be clear, not implemented means it will trap as opposed not being
physically implemented with flops and being RdOnly 0.
  (unlike opcodes where not-implemented opcode are not required to trap,
correct? -which is the subject of another question)

OK, so here is a gap: Is your answer "correct"?
I.e. does the Sail formal model agree with your answer? Does Spike agree?
QEMU?
If they do, great.
If they don't - either your answer needs to change or those models do. (and
if they disagree with each other, one at least needs to be updated)
We need a mechanism to make sure questions like this can easily be
confirmed with the models that are out there, especially those that
compliance will depend on.

allenjbaum, 2020-08-12:
Correction: not existing (rather than not implemented)
further question: can there be cases where trapping on a non-existent CSR
is optional?
E.g. either behavior of trapping or returning zero is legal for the same
hart configuration?

aswaterman, 2020-08-12:
There is never any "required to trap" because non-conforming extensions can always hijack that encoding space.  You are welcome to say that such implementations are outside the scope of compliance-testing, though.

I've clarified this point in the spec.

Yes, Spike does the right thing.

closed 2020-08-12 by aswaterman

allenjbaum, 2020-08-12:
Ah: CSR accesses are required to trap only if they violated priv level or
R/W restrictions based on the CSR address map - I misremembered.
But.. what is the "right thing" in this case?
IF there is no non-conforming extension, it traps?
OR, does returning a zero make it non-conforming?
And I will be quite happy not to test this in compliance thank you.

aswaterman, 2020-08-12:
My preference would be that we limit the scope of compliance testing to implementations that raise precise illegal-instruction exceptions when a standard feature isn't implemented (instructions as well as CSRs).

allenjbaum, 2020-08-13:
I love it when you limit the scope of compliance, I'll have you know.

"standard feature" in this context probably means something quite specific,
but I'm not sure where that is defined.
I would interpret this to mean that I should only test CSRs that may?must?
exist if an implemented, ratified feature/extension is implemented
(regardless if enabled/disabled, I think)
But if the presence of a CSR is required to determine if a feature is
implemented - then this is broken; it must trap and we must test for it.
I'm not sure how to resolve this.

We need a list of CSRs and the conditions under which they may (or must) be
implemented (or are optional but could be zero, I guess)

aswaterman, 2020-08-13:
To rephrase, I was suggesting that we should limit the scope of compliance testing to implementations that don't repurpose standardized encodings.  If a standard encoding doesn't raise a precise illegal-instruction exception, then it must do what the spec says it's supposed to do.

Curating a list of CSRs and the conditions under which they may/must/mustn't be implemented/etc. is a related, but distinct, concern.

aswaterman, 2020-08-13:
Also, disclaimer: this is just my opinion; I am sure there are other people who will argue to the contrary, that we should have a wider scope and support compliance testing of implementations that repurpose some of the standard encodings.  That's justifiable, too, but I think we should first become capable of walking before we attempt to run...

tzwaenn, 2020-08-13:
Good point about the Sail spec - I'll raise an issue there as existence of registers does not seem to be handled too well there.

allenjbaum, 2020-08-14:
I don't have a strong opinion; I just need to unambiguously know what the
rules are.

So: if no CSR has been repurposed for custom use, then we have a list of
CSRs which are allocated to various extensions and features.
CSR            feature            feature
repurposed implemented   enabled
  1                0                      x              how does SW
determine if it exists? (compliance will be told externally)
  1                1                      x              not allowed, don't
care
  0                1                      0              all feature CSRs
should trap with illegal op trap
  0                1                      1              should not trap
unless wrong priv level or write a RO CSR

   Z-extensions have no MISA bit to clear to disable them - but that's
their problem.

We may need that curated list eventually to ensure that the SAIL model
works correctly, and we know what to test and what not to test. Even better
if the Sail spec can read a file and to do configure itself automatically
(it would still need to merge that with the list of features that claim to
be implemented

commit 1eaebccc (2020-08-12): mcounteren only exists if U-mode exists

Resolves #561