riscv/riscv-isa-manual #573: Do we have a standard format or document for assembly expression?

Rolfy360, 2020-08-27:
For example, instruction ADD
in the manual: "ADD performs the addition of rs1 and rs2."
is it 
ADD rd, rs1, rs2 
or 
ADD rd, rs2, rs1
or 
ADD rs1, rs2, rd

Another case, for floating point instruction with rounding mode,
FADD.S rd, rs1, rs2 ## DYN
FADD.S rd, rs1, rs2, rne ## nearest-even
this expression is from binutils, not in the manual.

Lots of assembly code in manual seems like there IS a format, if you know MIPS first.
In the vector extension, all instructions are introduced with assembly example.

Also, I have the book "The RISC-V Reader".
The appendix A is very helpful, provide very specific information.
ADD rd, rs1, rs2
x[rd] = x[rs1] + x[rs2]

AMOADD.W rd, rs2, (rs1)
x[rd] = AMO32(M[x[rs1]] + x[rs2])

BLE rs1, rs2. offset
if (rs1 <=s rs2) pc += sext(offset) ## correct one should be x[rs1] <=s x[rs2]

sorear, 2020-08-27:
It was decided a while ago that assembly syntax was out of scope for the ISA manual (even though the ISA manual uses assembly examples in several places), see https://github.com/riscv/riscv-asm-manual for the (very incomplete) documentation for the mips-derived syntax used by binutils and llvm.  See also #540.

knightsifive, 2020-08-27:
My understanding is that this is the goal of https://github.com/riscv/riscv-asm-manual , although it appears there's a lot of work to be done on that project. Your contributions would be appreciated!

kersten1, 2024-04-05:
Closing as the other project exists.

closed 2024-04-05 by kersten1