riscv/riscv-isa-manual #575: Clarify some rules for implicit extensions?

Nelson1225, 2020-09-02:
Table 27.1 summarizes the standardized extension names and their implicit extensions,
| Extensions  | Implicit extensions |
| ------------- | ------------- |
| F  | Zicsr  |
| D  | F  |
| G  | IMAD_Zifencei  |
| Q  | D  |
| Zam  | A  |

However, I have some problems need your helps to clarify.

1. How to set the versions for the implicit extensions?
Are the implicit extensions' version set by the default ISA spec?  For example, `-march=rv32f`, and the arch string will be expanded to `rv32f2p2_zicsr2p0` according to the newest ISA spec.

2. Can we set the extensions into arch string, but they are set implicitly by other extensions?

**Example 1**, Is the `-march=rv32f_zicsr3p0` valid?
If this is valid, then the final arch string should be `rv32f2p2_zicsr3p0`.

**Example 2**, Is the `-march=rv32g_zifencei3p0` valid?
If this is valid, then the final arch string is `rv32i2p0_m2p0_a2p0_f2p0_d2p0_zifencei3p0`

**Example 3**, Is the `-march=rv32g_zicsr3p0` valid?
If this is valid, then the final arch string is `rv32i2p0_m2p0_a2p0_f2p0_d2p0_zicsr2p0_zifencei2p0`

**Example 4**, Is the `-march=rv32g_f3p0_d3p0` valid?
The current toolchain doesn't allow this rule...  But if the above examples are valid, then this should be valid, too?  If this is valid, then where should we set the `f` and `d`?  before `g` or after `g`?

3. Should the zicsr is added implicitly before i2p0, and shouldn't be added implicitly after i2p1?

4. The standard extension usually use one character to represent.  Should the `WMO` use `w` instead?

Thanks
Nelson

aswaterman, 2020-09-02:
> 1. How to set the versions for the implicit extensions?

I don't know.  I think the spec needs to clarify this one.  Perhaps each extension with an implicit dependence might need its own rule--e.g., `Zam0p1` might explicitly imply `A2p1`.  `D` might imply exactly the same version of `F`, so e.g. `D2p2` implies `F2p2`.

> 2. Can we set the extensions into arch string, but they are set implicitly by other extensions?

I think this is answered by the statement in the spec that "The table also defines the canonical order in which extension names must appear".  F precedes D, so `rv32ifd` is legal.  G doesn't precede F, so `rv32gf` is illegal.  (G is also a special case, like I and E.  If I, E, or G is in the string, it must be the first letter.)

> 3. Should the zicsr is added implicitly before i2p0, and shouldn't be added implicitly after i2p1?

I think that's right.

> 4. The standard extension usually use one character to represent.  Should the `WMO` use `w` instead?

RVWMO doesn't appear in the ISA string.  RVWMO is the default.  If the processor uses RVTSO, then `Ztso` appears in the string.

Nelson1225, 2020-09-02:
> > 1. Can we set the extensions into arch string, but they are set implicitly by other extensions?
> 
> I think this is answered by the statement in the spec that "The table also defines the canonical order in which extension names must appear". F precedes D, so `rv32ifd` is legal. G doesn't precede F, so `rv32gf` is illegal. (G is also a special case, like I and E. If I, E, or G is in the string, it must be the first letter.)

Thanks for the detailed reply :)

One more question, G is a special case that means the extensions `imafd_zifencei` are exactly added rather than implicitly added.  We have clarified the `rv32gf` is illegal, and the `rv32ifd` is legal, so I think the `rv32g_zifencei` is illegal, too.  Is this right?

aswaterman, 2020-09-02:
I have a feeling it will reduce confusion if the "G" rule only applies to MAFD, i.e., I think we should accept `rv32g_zifencei`, even though it is redundant.  This isn't based on anything technical, just my intuition.

kito-cheng, 2020-09-02:
> rv32g_zifencei
I think it should accept if `G` has no exception rule, otherwise that mean `rv32ifd` is invalid since `d` implied `f`.

but apply same rule, that mean we should accept `rv32gf`?

aswaterman, 2020-09-02:
@kito-cheng G was previously defined to mean IMAFD, rather than IMAFD_Zifencei.  I guess I foresee more problems with Zifencei than with MAFD... it's not very scientific.

kito-cheng, 2020-11-13:
@aswaterman I sent a patch set to support that on GCC, and I made an exception for G, disallow override and appear again for IMAFD, but allowed for Zifence and Zicsr, e.g. rv32gf2 is illegal, but rv32g_zicsr is legal.

For other implied extension, it's ok to override that, just like rv32if2d3 and rv32id3_zicsr2 is legal.

https://gcc.gnu.org/pipermail/gcc-patches/2020-November/558915.html

aswaterman, 2020-11-13:
Cool, thanks.

Nelson1225, 2020-11-25:
Thank you very much, @aswaterman and @kito-cheng.  I think I have got enough information and know how to do for now.  I'm working on this recently, so that binutils should have the same arch string behaviors soon after.

Nelson1225, 2020-11-25:
Oops, reopen the issue since it would be great to mention our consensus spec :P

allenjbaum, 2020-11-25:
I missed this the first time around. The implicit extension that I don't know how to specific is HW misalign HW support.
How do we specify that, r is there some reason we don't need to?

aswaterman, 2025-12-09:
The zeitgeist is that "implies" and "requires" are synonyms in this context, and that it is valid to write e.g. `rv32if`, which is equivalent to `rv32if_zicsr`.  RVIA obviously can't codify what other software projects choose to do, but we will offer this recommendation in a future update to the Naming chapter.

closed 2025-12-09 by aswaterman