riscv/riscv-isa-manual #576: Trapping hypervisor instructions when misa.H=0

pdonahue-ventana, 2020-09-02:
The hypervisor chapter says:
> When misa[7] is clear, the hart behaves as though this extension were not implemented,
and attempts to use hypervisor CSRs or instructions raise an illegal instruction exception.

It makes sense that accesses to hypervisor CSRs must trap since chapter 2 says:
> Attempts to access a non-existent CSR raise an illegal instruction exception.

However, there is no analogous requirement in the privileged architecture that reserved or unimplemented instructions must raise an illegal instruction exception.  The unprivileged architecture is very explicit that the behavior is UNSPECIFIED and, other than specific cases like accesses to reserved CSRs or trapping particular instructions with TVM/TSR/etc., the privileged architecture doesn't add a blanket requirement.

Is it expected that HLV/HLVX/HSV must trap only on implementations that implement H but where H is disabled?  Or is this a broader requirement for all implementations, even those that hard-wire misa.H=0?  Or should this requirement to trap hypervisor instructions be removed?  Or should the privileged architecture be modified to require that illegal instructions must raise illegal instruction exceptions (regardless of support for H)?

/cc @jhauser-us

aswaterman, 2020-09-02:
I'm fine with adding a sentence to the priv ISA that unimplemented instructions must raise illegal-instruction exceptions.  I don't think it addresses all of our woes (e.g., it doesn't forbid non-conforming extensions from reclaiming reserved opcodes), but it does handle this case, right?

pdonahue-ventana, 2020-09-02:
I've had many conversations with people who, biased from experiences with other architectures, incorrectly assumed that such a requirement existed all along.  I would welcome adding such a sentence.

But you raise a good point.  Does it fully cover this case?  What if a non-conforming extension claims these hypervisor instructions?  (The sentence can be added even if this wart remains unresolved.)

gfavor, 2020-09-02:
Aren't "non-conforming" extensions by definition not RISC-V compliant
(whereas "non-standard" extensions are staying within proper boundaries and
only using encodings officially set aside as "custom" encodings)?  In other
words, an implementation with non-conforming extensions is on its own as
far as it compatibility with "RISC-V compliant" software, and has chosen to
"go rougue" wrt the arch specs.  So is there really anything unresolved by
adding the suggested sentence.

>

allenjbaum, 2020-09-02:
Do you want to distinguish unimplemented from disabled? Doesn't this only
happen when an op or extension can be disabled?

aswaterman, 2020-09-03:
"Unimplemented or disabled" is what I meant.

sorear, 2020-09-03:
There's a problem here because we really want to say that if H is emulatable, all encodings used by all past and future versions of H will trap, but we don't have the opcode space structured in that way.

allenjbaum, 2020-09-03:
could you expand on that? As long as those opcodes are in the standard, not
custom, opcode space, a profile  could easily require that specific opcodes
either work or trap.

sorear, 2020-09-03:
A profile "could" do those things, but no profile does do those things, and the requirements that we do have in profiles are being ignored by more hardware vendors than are following them so I'm not sure how much it will help.

allenjbaum, 2020-09-03:
The profiles group is just starting up (and looking for a chair and
co-chair?) , so it's early yet....

allenjbaum, 2020-09-03:
KRsste just posted:
From one direction, we are developing “profiles” that standardize a given
instruction set combination, and there should be only a few of these - the
most important of which is that for Linux ABI.
From the other direction, many RISC-V users want the great customizability,
so I think we might need some innovation in how toolchains are constructed
and delivered.

The profiles are what standard software requires, so no fragmentation
allowed.
The custom toolchain is where the implementer specifically wants to do
something different to tune for a use case.

>
> The profiles group is just starting up (and looking for a chair and
co-chair?) , so it's early yet....
>
>>
>> A profile "could" do those things, but no profile does do those things,
and the requirements that we do have in profiles are being ignored by more
hardware vendors than are following them so I'm not sure how much it will
help.

pdonahue-ventana, 2020-09-15:
@aswaterman: Can you reopen this since #582 was reverted and questions of backward-compatibility and non-conforming extensions still remain?  Based on my quick research, it looks like I am unable to reopen my own issue if you (repo collaborator) closed it.

kasanovic, 2020-09-18:
Reserved instructions are not mandated to trap in the ISA specs - this is for modularity of the specs.  A RISC-V system can be compliant with one or more standard ISA modules but have some non-conforming extensions.  If nothing else, any prototype implementation of a new draft standard extension will fall into this category.  This system can report itself as being RISC-V compliant for those modules, and run all standard software.  A profile could prohibit this by mandating that all instructions trap unless they implement a RISC-V standard extension admitted by the profile (some profiles might forbid some standard extensions or even prohibit custom extensions).

The ISA specs say that attempting to access a non-existent CSR is mandated to raise an illegal instruction exception (Priv, sec 2.1), but non-existent is different than reserved.  A non-conforming extension could implement these CSRs (including trivially as zero registers) and hence not need to raise an exception.

We have, however, used CSR access traps to detect presence of CSRs (e.g., PMP, perf counters).  This effectively prohibits the use of this standard space by a non-conforming extension, even when the standard extension is not present.  We should limit use of this discovery technique in future (e.g., use other smaller fields/encodings to indicate presence), but it has to be grandfathered in for existing ratified uses.

Getting back to this thread - I think the hypervisor extension has to clearly call out what must be trapped, and hopefully can reduce this to the minimum to make the extension maximally useful.

pdonahue-ventana, 2020-10-02:
I think that my concerns are addressed by #595 and #592.  Thanks.

PR #582 (2020-09-15): Require unimplemented instructions to trap
Fixes #576
closed 2020-10-02 by pdonahue-ventana