riscv/riscv-isa-manual #588: Behavior when MXL/SXL/UXL are writable

JamesKenneyImperas, 2020-09-22:
Is there a detailed discussion anywhere on the full implications when MXL/SXL/UXL are writable? I see a few references to this in the Privileged Architecture specification, but mainly with regard to the need to sign-extend GPR values to the full architectural width when operating with restricted XLEN. However, not much is said about the effect on CSRs; for example, I have these questions:

1. In the ARM architecture, system register values in AArch32 and AArch64 modes are generally overlaid so that the same underlying data is accessed in the two modes (often with a pair of AArch32 registers making up a single AArch64 one). Is the same kind of thing expected for RISC-V CSRs?

2. The ARM Architecture forbids operating with lower-privilege levels in AArch64 mode and higher-privilege levels in AArch32 mode. The RISC-V specification says that operating with wider XLEN at lower privilege levels is not prohibited, although this would generally indicate a software bug. What is is the effect when taking an exception from wider XLEN mode to narrower XLEN mode in this case? What happens on return from such an exception, particularly in the case when the original exception address was not in the 32-bit address space?

3. What is the effect on page table and other memory management structures when switching between XLEN widths? For example, suppose that execution has occurred in supervisor mode with SXLEN=64, causing population of TLB structures. If SXLEN is switched to 32, are cached memory translations in TLB structures retained and used, or invalidated, or ignored but reused on a subsequent switch to SXLEN=64?

Thanks.

allenjbaum, 2020-09-22:
These are excellent questions. It is one of the features that the SAIL
model does not (currently) support (and because of its pervasive nature,
will be difficult to support).
Is anyone aware of an implementation that allows XLEN to change?
PRiv spec section 2.4 makes explicit what happens to CSR values and
discusses the implications for some specific cases, but certainly not all
of them.

One oddity that I see is when
     UXLEN=32 and SXLEN=64, user-mode memory accesses reference the lowest
4 GiB of the address space
But, virtual addresses are signed, and this implies that umode address
0xFFFFFFFF maps to smode address 0x00000000_FFFFFFFF rather than
FFFFFFFF_FFFFFFFF; yet an address in a register is sign-extended (I suspect
that "lowest 4GiB" must be interpreted a bit differently than the obvious
one).

My interpretation of  sec 2.4 is that the upper bits of CSRs (that don't
have separate upper halves for 32b mode, e.g. xstatush or hpmcounterNh) are
cleared, though this is ambiguous.
The text describes what happens to the low bits when transitioning to a 32b
XLEN, but not what happens to the upper bits when transitioning to a 32b
XLEN.
Assuming I am correct, though, a trap into a 32b XLEN m-mode cannot return
as the upper address bits are lost. (This is explicit if it ever
transitions to 64b mode).
#1 is explicitly no; some CSRs do, but most do not.
#2 is that it returns to some address in "lower" 4GB, regardless of where
the trap occurred
#3 is complicated. The page table is in memory; that is unchanged*, but
the PTE is effectively moved to the lower 4GB because SATP is now 32b.
TLB structures are not architecturally defined, so that sounds
implementation dependent (if it is a page-fault trap, then it wouldn't be
found in the TLB anyway)
 If SATP points to data that isn't the PTE, you'll get garbage when
accessing anything using virtual memory translation.
 (* complicated by any AD bit changes in a PTE entry)

aswaterman, 2020-09-22:
“lowest 4 GiB” means what it sounds like: 0 - 00000000FFFFFFFF. When UXL=32 and SXL=64, U-mode VAs are zero-extended, as a special case, to increase the address space available to U-mode. Had we sign-extended them instead, the negative addresses would overlap what is conventionally kernel memory, limiting U-mode to 2 GiB of address space in practice.

aswaterman, 2020-09-22:
I’m hoping that near-term work on an operational description of the address-translation system can provide a framework in which we can answer question 3 clearly.

aswaterman, 2020-09-22:
Re: question 2: the exception addresses are truncated. This is implied by the section on CSR width modulation.

aswaterman, 2020-09-22:
Question 1 is explicitly answered by the section on CSR width modulation. Most CSRs are simply narrower in the narrower mode, with no register pair for the high part. The width-modulation section explains the rules for CSR reads and writes across mode changes.

gfavor, 2020-09-22:
Separate from the specific nature of the address-translation system
description, what is the intended answer?  Is it software's
responsibility to do a suitable sfence.vma (or bear the consequences
otherwise), or is hardware required to distinguish translations cached from
under different SXLEN values?

Greg

aswaterman, 2020-09-22:
@gfavor I think an SFENCE.VMA should be required, and to keep address-translation caches opaque, failure to execute an SFENCE.VMA should result in UNPREDICTABLE translation behavior (with the usual caveat that unpredictable behavior can't break the system, violate protection boundaries, etc.).  WDYT?

gfavor, 2020-09-23:
Great.  That's the answer I was hoping for - both wrt requiring an
sfence.vma and wrt what happens otherwise.

Also it would seem that this 1-2 line clarification can be made to the spec
irrespective of any rewriting of the address-translation system description.

gfavor, 2020-09-23:
And by "hoping for" I mean that I think that is a good answer and what the
architecture should go with.

On Tue, Sep 22, 2020 at 5:10 PM Greg Favor <gfavor@ventanamicro.com> wrote:

> Great.  That's the answer I was hoping for - both wrt requiring an
> sfence.vma and wrt what happens otherwise.
>
> Also it would seem that this 1-2 line clarification can be made to the
> spec irrespective of any rewriting of the address-translation system
> description.
>
>

JamesKenneyImperas, 2020-09-23:
I somehow missed the information in the CSR modulation section, so thanks for pointing that out. I'm still not entirely sure what this means though - can you clarify further?

1. Is the expectation in section 2.4 that any XLEN switch requires _immediate_ modification of all CSR values visible in the new mode according to the rules given there? An alternative would be to do this only when the CSR is explicitly _written_ at the new width; it feels to me that the implications of the latter method would be easier to understand, but the effect would be to preserve the high half of unwritten CSRs over an XLEN=64/32/64 switch sequence rather than zeroing them. I'm also not sure if the modification algorithm as described in 2.4 is fully robust when fields are shared by CSRs with different privilege levels, if these have different effective XLEN values, but I'm struggling to think of a concrete example.

2. I think that any CSR which normally has a 'high' register when XLEN=32 needs special mention in section 2.4. For example, if CSR mcycle was modified using the given rules over an XLEN=64/32/64 switch sequence then the effect would be that the high half of mcycle would always be zero on return to XLEN=64 mode, which isn't right - I would expect mcycle to be unmodified by this sequence and the high half of the counter to appear as mcycleh as normal when XLEN=32.

3. Comparing with the ARM architecture again, it seems odd to allow MXL to be writable by code. In that architecture, a switch from AArch64 to AArch32 is only possible on exception return to lower privilege level, and a switch from AArch32 to AArch64 is only possible on a trap to higher privilege level. The initial operating mode (AArch32 or AArch64) at the highest implemented privilege level is set by a signal sampled at reset, if both modes are implemented at that level. This seems to fit the obvious use cases for mixed-width operation well.

aswaterman, 2020-09-23:
1. Yes.

2. I'll let @jhauser-us decide how to clarify this point.  Listing all "high" CSRs in Sec. 2.4 is not the best solution; the text should stand alone.

3. There is obviously at least some value in allowing systems to emulate other systems, even at the machine level.  Whether or not the complexity is justified is a value judgment.

allenjbaum, 2020-09-23:
Point #1 is where I had reservations as to the meaning also - but concluded
that it just didn't make sense to defer clearing the upper half regardless
of how it was interpreted; you can't recover from traps, so why bother.

JamesKenneyImperas, 2020-09-24:
Another point that I think needs clarification, perhaps in section 2.4, is what happens when a CSR for another privilege level is read or written if that privilege level is configured with a different XLEN to the current privilege level. For example, what is the effect of reading and writing the satp CSR in M-mode when MXLEN=64 and SXLEN=32, or when MXLEN=32 and SXLEN=64?

In order for loads and stores from M-mode with MPRV=1 to work correctly, I think the answer must be that the CSR must appear to be in the other-privilege-level format and width, but I can't see that this is explicitly stated.

bharath-sifive, 2022-06-06:
In a 64 bit mstatus CSR, UXL and SXL are allocated two bits [33:32] and [35:34] respectively. When I try to enable paging for Sv48 mode, UXL and SXL becomes [1,0]. What does "10" imply?

JamesKenneyImperas, 2023-11-19:
@aswaterman and @jhauser-us, I've just noticed issue #1028 and the specification text revision there:

> If S-mode is implemented, the set of legal values that the `UXL` field may assume excludes those that would cause `UXLEN` to be greater than `SXLEN`.

This has rekindled my general uneasiness about the way that variable XLEN is defined to work in RISC-V. Consider a case where M-mode control software running with `MXLEN=64` is running an S-mode subprocess with `SXLEN=32`. Now suppose that the M-mode software requires also to directly run a U-mode subprocess with `UXLEN=64`. According to this definition, to do this it needs to set both `UXLEN=64` _and_ `SXLEN=64` before switching to the U-mode process, which will entail all the cost of S-mode XLEN-switch CSR modulation for no reason. This feels very wrong.

I know it is very late in the day, but I'm feeling increasingly strongly that the variable-XLEN behavior specified for RISC-V compares very badly to the ARM architecture, where switching between 32-bit and 64-bit operating modes is a lightweight and intuitive procedure (I've given some details in earlier posts in this thread, for people who may wish to know more). In contrast, the defined RISC-V behavior feels very heavyweight and onerous to implement; for example, every CSR in a variable-XLEN system will have to have logic with it to perform the top-half-zeroing/bit-reordering as required for modulation, and in the future there may be hundreds or even thousands of CSRs to update on such an event. Is this really workable?

I'm sorry to raise such a fundamental issue now, but I feel it's really important to address this. Please read this in the spirit of constructive criticism that I intend.

Thanks.

allenjbaum, 2023-11-20:
First: the encoding 10 indicates 64b mode according to table 3.1.
If you enable SV48 mode, then all XLEN must be 64b,
This really is a somewhat ambiguous case, in my mind.
The combination of SATP.mode=SV48, and SXL=32 shouldn't be legal, and both
are WARL fields:
So -- do you map the illegal SATP.mode field to a legal value, or map the
SXL field to a legal value.
I'm not sure that spec says explicitly, or if it is implementation
choice, and don't have time to dig through the weeds to find it..
If SXL has priority, then you'd have to write SXL first, which would force
satp.MODE to be a legal value (SV39+)

From a SW point of view, Mmode does not control UXL; Smode should be the SW
layer that is doing it.
In this particular case, a 32b Smode OS cannot run a 64b Umode app, period.
If it tries to set UXL to 64b, the write to the UXL field will be
effectively ignored. That sounds like exactly the correct thing to do.

Now if for some reason a 32b Mmode wants to run a 64b app (or a 64b OS) -
note this comment in sec 3.1.6:

Fields SD, SXL, and UXL do not exist in mstatush.

I think that implies (though it perhaps should be explicit) that there is
no choice but to have a 32b SXL & UXL when MXL is 32b.
Mmode would have to promote itself to 64b, and then promote BOTH Smode and
Umode to be 64b in order to do that.
either in that order, or simultaneously.

So following this line of reasoning, the XLen of a priv mode can never be
wider than the XLEN of a higher priv mode.,
and for any satp.MODE>SV32, MXL and SXL must be 64b.
So the legal combinations are (assuming all mode are implemented; if no
Smode, then SXL=0, if no Mmode, then UXL=0
  MXL   SXL    UXL
  32b    32b     32b
  64b    32b     32b
  64b    64b     32b
  64b    64b     64b
and an attempt to set any other combination must be mapped to a legal
combination
(sadly, which one is up to the discretion of implementation, but if you're
building a hart that allows these field to be writable,
I recommend the "leave the fields unchanged option)

I haven't looked at what happens when hypervisors are involved, but I
suspect it is similar )and more complicated ...)

Regarding the 1000's of CSRs that must be modulated: I am a bit confused
about this
From wider to narrower, it clears the upper half, and clears any RO bit
positions that are in the lower half.
From narrower to wider, it's the same: clear any RO bits in the lower half,
then zero extend.
This doesn't make sense to me: if the RO bits in the low half of either
format must be 1, they can't be clear, and
it doesn't address the issue of fields that move from/to a CSR that gets
split into 2 CSRs or combined into 1 CSR, (e.g.the MXL field)
so I think I'm misreading something

allenjbaum, 2023-11-20:
Also: I don't see how a 64b Mmode can run a 64b Umode if SXL=32b. If paging
is enabled, then the page tables are set up for 32b mode, and they'll have
a totally different format, so either the Umode app mose be running in bare
mode (physical memory only) or Mmode has to construct the page table, turn
off all delegation, switch off all delegation, and be prepared to handle
all traps (and all ABI calls) that an OS is supposed to handle. Not a very
lighteweight Mmode., shall we say.

All in all: It hurts when yo do that, don't do that.

JamesKenneyImperas, 2023-11-20:
> Also: I don't see how a 64b Mmode can run a 64b Umode if SXL=32b. If paging is enabled, then the page tables are set up for 32b mode, and they'll have a totally different format, so either the Umode [...]

Yes, that was mostly a Sunday morning brain fade on my part. This isn't a viable scenario. Sorry.

allenjbaum, 2023-11-20:
Well, you noticed that I didn't catch it until after I went the first
email, so not the only brain-fade

allenjbaum, 2023-11-20:
> The CSR Width Modulation section is talking about writable fields, not
> read-only fields, which I think are covered by the CSR Field Modulation
> section instead. But I agree with you that the Privileged Specification
> doesn't describe how this should work for CSRs with high/low halves as far
> as I can see, and there are possibly other individual CSRs that need
> supplemental information too (I haven't checked).
> What concerns me most is the ever increasing cost and complexity of
> modulation as more CSRs are added over time - is this scalable?

Well, spec sec 2.6 CSR Width Modulation.  explicitly talks about read-only
fields:

2. For the read-only bits of the previous-width CSR, the bits at the same
positions in the temporary register are set to zeros.

That doesn't make sense to me. It did once upon a time, I think, but not
now.
(and this gets really weird if a write-once bit is considered read-only
once written; this would make PMPCFG lock bits able to be unlocked)

Among other things, this section doesn't talk about what happens when a CSR
appears or disappears when width is modulated (e.g. any CSR which has a
high half in RV32)

JamesKenneyImperas, 2023-11-21:
Yes, you're right, this section is too superficial to provide definitive information in this context.

I think life would be a lot easier if the definition of `misa.MXL` was changed so that it was _read-only_ from M-mode and instead configurable at reset by an external signal if required - this would remove any dynamic modulation requirement from all M-mode CSRs, and is what the ARM architecture does. I think it would also simplify things for folks in the Debug Specification: unless there's a good reason not to, DXLEN should just be fixed to the reset MXLEN, which would prevent any panic attacks about how to modulate `tdata1` or any other Debug Mode CSR.

If M-mode and Debug Mode CSRs are removed from the modulation set, we're left with a much smaller set of U-mode, S-mode, HS-mode and VS-mode CSRs to consider. I'm going to look through the definitions of all CSRs that I'm currently aware of for those modes to try to understand which ones might be problematic and whether that has implications for the modulation algorithm.

aswaterman, 2023-11-21:
The specification would’ve been simplified by making MXL read-only, but IMO that ship has sailed. With that said, I suspect that most vendors will hardwire MXL to avoid thinking about the kinds of issues you raised.

JamesKenneyImperas, 2023-11-28:
@aswaterman, I don't believe that this can be divided into "bite-sized pieces" as you suggest. I've tried to offer suggestions about this using my experience of the ARM architecture, where the equivalent requirement is handled far better than what is proposed in section 2.6, and does not result in significant state change and information loss as is the case here. I'm sorry that you were unable to understand the alternative I suggested, but I don't have time now to try to decompose this into new issues.

JamesKenneyImperas, 2023-11-29:
@aswaterman, thank you for reopening this. Before creating a number of unwanted extra issues, perhaps it would be better to quickly summarize why I think the current specification is not fit for purpose and what a better one would look like. Then you can decide in discussion with others whether you want to take any of this forward or not, or create new more-specific issues. Does that sound reasonable?

The current width modulation specification is entirely focused on _preservation of value at bit position_ when XLEN changes. This is in fact an unimportant feature: what is important is _preservation of information_, _simplicity_ and _efficiency_.

I think the modulation definition should have the following characteristics:

1. **The cost of the switch must be as low as possible**. Specifically, only fundamental machine state related to XLEN control bits themselves should change - otherwise, the cost and complexity of modulation logic will increase forever as new CSRs are added.
2. **All state visible as writable bits with narrow XLEN should also be visible as writable bits at the widest XLEN** - this greatly simplifies procedures that save and restore machine state (perhaps for switching to another hart) because that process can always be carried out using the widest implemented XLEN. Equivalent bits do _not_ need to appear at the same positions in each width view, however.
3. **Modulation should be reversible without information loss** - this in fact follows from 1 and 2, but should be clearly stated as a requirement so that designers adding new CSRs understand the model.
4. **CSR writes at narrow XLEN should only affect bits visible at that XLEN** - this simplifies understanding of CSRs which are high/low pairs with narrow XLEN, and means that observed wider-XLEN behavior for a narrow CSR is backwards-compatible if a new "upper half" CSR is introduced.
5. **Every CSR definition should have an explicit statement about modulation**. For most CSRs, this will be a very simple definition about mapping to the lower/upper half of a wider CSR. My previous post above identifies all the CSRs I've been able to find in the Privileged Specification and ratified extensions that would need something different.

The specification I proposed above addresses all these issues, but I'm sure other approaches are possible as well.

Thanks.

aswaterman, 2023-11-29:
We discussed this topic in the ARC meeting today, and we've decided it's best to stick with the current definition of CSR width modulation.  We acknowledge that it causes modest inconvenience for some CSRs (`satp` being an obvious case), but we disagree the penalty is severe, and we find some of the specific recommendations (e.g. MSB handling) to be equally objectionable.  But most importantly, it's already a ratified standard, and the proposed changes are backwards incompatible.  We do not agree with your claim that "we can propose something different with equivalent effect", since the "something different" is architecturally visibly different.

JamesKenneyImperas, 2023-11-30:
Given the conclusive nature of the previous post, I've attempted to switch to using the ratified modulation model, but have run into a conceptual problem with modulation of U-mode CSRs when the Hypervisor extension is present. The conceptual scenario is as follows:

1. A hart is executing in 64-bit HS-mode with `sstatus.UXL=1` (i.e. 32-bit HU-mode).
2. It switches to 64-bit VS-mode, with `vsstatus.UXL=2` (i.e. 64-bit VU-mode). Presumably any U-mode CSRs should implicitly undergo widening modulation at this point, otherwise they will have incorrect format when viewed in virtual mode: is that correct?
3. The hart switches to a 64-bit VU-mode process. The hart implements the Zcmt extension, and sets the `jvt` CSR to a 64-bit address.
4. A trap is taken to 64-bit HS-mode to handle a virtualized service. Presumably any U-mode CSRs should implicitly undergo narrowing modulation at this point, otherwise they have incorrect format when viewed in HU-mode: is that correct? If it is, the effect of the ratified modulation rules is to discard the most-significant half of the `jvt` CSR.
5. The trap returns to the 64-bit VU-mode process, which now crashes because the `jvt` CSR has been corrupted.

This is obviously only a conceptual scenario because the Zcmt extension is explicitly described as _not compatible with architecture class processors_ and therefore should not coexist with the Hypervisor, but the problem will exist for _any_ U-mode CSRs where the modulation process destroys information, in either direction. In particular, any future extension that adds U-mode scratch or pointer registers (like CLIC) looks vulnerable.

Can you please clarify section 2.6 of the Privileged Specification to **either** provide explicit information about how width modulation of U-mode CSRs can be robustly handled in this situation, **or** to specify design constraints on CSRs when the Hypervisor extension is implemented that avoid the modulation problem?

Thanks.

(I've reopened this issue to preserve the context of the previous discussions, but if you prefer I can make a new issue entirely.)

JamesKenneyImperas, 2023-12-07:
I will open a more specific issue about interaction of CSR width modulation and the Hypervisor extension

closed 2023-12-07 by JamesKenneyImperas