riscv/riscv-isa-manual #594: The behavior upon decoding a reserved instruction in the "C" Standard Extension

bnossum, 2020-10-02:
Referring to section 2.2 of "Volume I: RISC-V Unprivileged ISA V20191214-draft": 
"The behavior upon decoding a reserved instruction is unspecified".
I presume the same behaviour would be acceptable for (the very few) RVC reserved instructions `16'b100xxxxxxxxxxx00`, `16'b100111xxx1xxxx01`?

If so is the case, may I propose a note to that effect be added to section 16.8?

aswaterman, 2020-10-02:
I’m concerned that adding that note in the C extension chapter will make people wonder about the note’s absence in other contexts. I think it’s better to say it only once, up front.

bnossum, 2020-10-02:
Very good point. But if it is universally such that "The behavior upon decoding a reserved instruction is unspecified", then why is this text in chapter 2, "RV32I Base Integer Instruction Set"?

knightsifive, 2020-10-02:
A lot of "universal" text appears only in Chapter 2. For example, the mis/alignment properties for loads and stores are specified only in Chapter 2 but have implications in Chapters 4,5,6,11,12,13, and perhaps elsewhere.

bnossum, 2020-10-02:
Very well. It may also be argued that the spirit of the text in section 2.2 is encompassed by section 1.7. I rest my case.

closed 2020-10-02 by bnossum

kasanovic, 2020-10-04:
Already closed, but rationale was that RV32I is the base ISA spec, on which extensions are built (and from which other base ISAs are derived), so it was factored into a common place needed to understand any other portion.

David-Horner, 2020-10-04:
What of text to make that explicit?
Such as 
>RV32I is the base ISA spec, on which extensions are built (and from which other base ISAs are derived) It is factored into a common place so specifications needed to understand all other portion are provided here. Unless expressed otherwise the requirements and characteristics expressed in this section apply to all other sections.