riscv/riscv-isa-manual #598: Disabling Non-standard Extensions in Misa

steven-bellock, 2020-10-09:
If an implementation has non-standard extensions, and it supports disabling them via writing misa.x = 0, is the implementation required to disable all of its non-standard extensions, or is the implementation allowed to disable some non-standard extensions and leave some others enabled?

gfavor, 2020-10-09:
Good question.  I wouldn't think that it is acceptable for all X extensions to be either all be disabled or all be enabled.  One thought for a resolution to this issue is:

misa.X=0 disables all 'X' extensions, and misa.X=1 allows custom extensions to be enabled.  Then each extension can also individually be enabled/disabled (e.g. in a custom CSR belonging to that extension).  So, while misa.X=1, extensions can be individually enabled/disabled as desired.  And clearing misa.X forces all of them to be disabled (creating a custom-extension-free hart config).

aswaterman, 2020-10-09:
I think that's the right definition if misa.X is writable.  If misa.X read-only, there still might be custom CSRs that can enable or disable individual extensions, in which case misa.X reads as bitwise OR of their enables.

kasanovic, 2020-10-09:
I think the existing spec text is clear.  misa.X is set if "any" non-standard extensions are present.  So it follows logically that clearing it, if writable, would have to disable all non-standard extensions.  When it is set, the set of non-standard extensions present is determined in some other non-standard way.

gfavor, 2020-10-10:
A small nit:  misa.X in general can't simply be an OR of all non-standard extension enables since "At reset, the Extensions field shall contain the maximal set of supported extensions".  It can be an OR only if reset also enables one or more of those extensions.

Given that the spec says nothing about the relationship between misa.X and individual extension enables, it seems like all that one can infer from the spec is that:
1) It is a global "custom extension enable/disable" that either allows such extensions to be enabled through custom means or forces all of them to effectively be disabled.  It may or may not have a logical relationship to individual extension enable bits.
2) It must start off set after reset if there are any such extensions.  Which implies nothing about which extensions, if any, are individually enabled after reset.

aswaterman, 2020-10-10:
Or we could amend that language to say “maximal set of supported standard
extensions”.

gfavor, 2020-10-10:
Although you and Krste better understand the past motivation behind that reset statement, it seems like it is good for misa to show - after reset - ALL extensions that are supported.  Then software readily knows that there are custom extensions (one or more) to be dealt with as far as enabling and disabling them.

allenjbaum, 2020-10-10:
I'm not sure I need to chime in to support that bit of common sense.
The question remains whether we should require that the maximal number of
custom extensions also be enabled.
But: if some custom extension can't be disabled, then does it follow that
misa.X must be hardwired to 1?

aswaterman, 2020-10-10:
> But: if some custom extension can't be disabled, then does it follow that misa.X must be hardwired to 1?

Yes.

kasanovic, 2020-10-12:
The X bit indicates there are non-standard extensions present.  Beyond that, it is really up to the non-standard extension provider(s) to define operation when X is set, including how they're individually enabled/disabled (i.e., we're not going to standardize non-standard behavior!).  Really, the only standard behavior misa.X can ensure is, if it's writable, to disable all non-standard extensions when misa.X is set to zero.  Remember non-standard includes non-conforming as well as custom -  it would not only disable custom extensions.

steven-bellock, 2020-10-16:
Ah yes I, and maybe others, had been thinking of this only in terms of custom extensions. Including non-conforming extensions complicates things. What if an implementation has a non-conforming extension that completely rewrites the meaning of misa? Upon reset that extension would be enabled and software wouldn't have a familiar, and single-bit, option to put the system into a conforming mode.

closed 2021-10-31 by gfavor