riscv/riscv-isa-manual #605: A question about mip.seip read/write.

tianhu-zeng, 2020-10-15:
Hi, I have a question about mip.seip. In Spec description:
[image]

Sentence marked in blue：Does it means the logic-or of mip.seip and the value of external interrupt controller were writeen 
to destination register by csrrw/csrrs/csrrc accessing In draft1.12 version? 

Sentence marked in orange：the ignoring interrupt value from the external interrupt controller, is it for set/clean opeation of a csrrs/csrrc instruction? And the interrupt from the external interrupt controller shouldn't be written to mip.seip?

BTW, when will a new Spc be released?

Tianhu

aswaterman, 2020-10-15:
The text you quoted was intended to be a clarification, not a semantic change.  In both cases, the value written to rd is the OR of the external signals and the internal flop.

I don't have a release date for v1.12 yet.

tianhu-zeng, 2020-10-15:
Got it. Thanks for you reply.

closed 2020-10-15 by tianhu-zeng

allenjbaum, 2020-10-15:
I had a similar confusion. I think the word "used" in the second sentence
is ambiguous, and should be replaced with "written, to distinguish it from
"read".

aswaterman, 2020-10-16:
There's only one value used in the RMW sequence, and it's both read and written.

allenjbaum, 2020-10-16:
Huh? There is a value that is read (from the CSR to a GPR) and the value
that is written (from a possibly different GPR to the CSR)

aswaterman, 2020-10-16:
That sentence doesn't discuss what's being written back to the x-register.  It's only talking about the RMW on the flop.

tianhu-zeng, 2020-10-16:
@aswaterman Both values will be written to MIP with RMW?

aswaterman, 2020-10-16:
I don't understand the question.

tianhu-zeng, 2020-10-16:
In the 2nd sentence, "ignoring the interrupt value from the external interrupt controller". Does it mean that the interrupt from the  external interrupt controller will not be written to mip?

allenjbaum, 2020-10-16:
The sentence immediately preceding this explicitly discusses the read  part
of the RMW sequence.
The sentence we are talking about is  "*However*, the value used in the
read-modify-write sequence of a CSRRS or CSRRC instruction"
 which I originally interpreted to mean an exception to the previous
sentence about the read. In fact, it means precisely the opposite.
If instead that sentence was worded "*However*, the value written in the
read-modify-write sequence of a CSRRS or CSRRC instruction" it would be
completely obvious.
And clearly, I am far from the only person confused by that sentence. I've
asked it to be clarified- what is obvious to you isn't obvious to others,
and there really isn't a good excuse not to make it obvious - there are no
downsides. IF it can easily be misinterpreted, it isn't a good spec.

aswaterman, 2020-10-16:
The value from the external interrupt controller never gets written to the flop.

When the CSR is read into an x-register, the value read is ((external interrupt) || flop).

When the CSR is written, the RMW action takes place on the flop.

aswaterman, 2020-10-16:
But @allenjbaum your rewording is even more confusing.  It's not just that the value _written_ in the RMW sequence is the software-writable bit.  That value is both _read_ and _written_ in the RMW sequence.  If you just say _written_, it leaves open to interpretation which value was _read_.

kasanovic, 2020-10-17:
I think the text could be made clearer by saying the "However, only the software-writeable SEIP bit participates in the RMW of  ...CSR instructions..., ignoring the interrupt value from the external interrupt controller".

aswaterman, 2020-10-17:
I think that helps slightly.  I also made the dependent clause at the end a bit firmer.

allenjbaum, 2020-10-17:
OK I see the problem here: there are 2 definitions of what RMW means in the
context of a CSR op..
One is an assumption  is that the value used to modify the value written to
the CSR is the CSR bit and GPR source and the CSR bit
The other  assumption is that the value used to modify the value written to
the CSR is the CSR bit and value being returned to GPR destination.
My model is the first, and the only thing that is different from other CSR
ops is that value being returned is modified by ithe nterrupt signal
Yours is the second:  the value returned is different, AND the value
written is different. That's more complicated to explain.
 (unless I'm further confused: does a CSRRW X0, MIP, X0 OR the interrupt
bit with the value being written (0) into SEIP?
  The only way you could tell is if the interrupt were cleared at the
source, and SEIP stayed set.

Using my model, you could delete the second sentence entirely, and it would
just work; you don't need to call out CSRRS or CSRRC as acting any
differently than CSRRW does in this case

Are there other cases where the second assumption is correct and the first
is not?
The other way to make it more clear might be to combine the two sentences
into one:

When mip is target of a CSR instruction, the value of the SEIP bit returned
to the rd destination register is the logical-OR of the software-writable
bit and the interrupt signal from the interrupt controller, but that
interrupt signal is not used to calculate the value written into mip.SEIP

That should remove any ambiguity about what is being read or written.

aswaterman, 2020-10-18:
I've pushed another attempt to clarify this that takes Allen's most recent suggestion into account, but retains the first clause of the second sentence to reinforce the concept.

aswaterman, 2020-10-18:
People can LMK if they don't think this version (48191f86c6e9a2c1bf57731e6f66fd9352591a11) is better than what it replaces.

allenjbaum, 2020-10-19:
OK, I like that a lot better.