riscv/riscv-isa-manual #607: hart and hardware thread

bhwk, 2020-10-25:
In the spec, at the end of section 1.2 there is,

>The important distinction between a hardware thread (hart) and a software thread context
is that the software running inside an execution environment is not responsible for causing
progress of each of its harts; that is the responsibility of the outer execution environment. So
the environment’s harts operate like hardware threads from the perspective of the software inside
the execution environment.

> harts operate like hardware threads

This part seems unclear. What does it actually mean?

aswaterman, 2020-10-26:
Unlike software threads, hardware threads always make forward progress.

knightsifive, 2020-10-26:
I think the confusion is that _hart_ is (arguably) defined here as an abbreviation of _hardware thread_, making the statement

> harts operate like hardware threads

sound rather vacuous. We resolve this vacuity by understanding that harts are actually _abstractions of_ hardware threads, as clarified in the next paragraph of this commentary as well as in the Lithe references.

Actually, it seems that some of this commentary is redundant with earlier (normative) text so probably could be dropped.

aswaterman, 2020-10-26:
Part of the confusion is that, although “hart” is a contraction of
“hardware thread”, harts aren’t necessarily hardware threads—they’re an
abstraction that looks, to the code running within the EE, like a hardware
thread. On a virtualized system, harts within an EE may be time-multiplexed
on physical harts in an outer EE, so long as they independently make
progress (in contrast to e.g. cooperative software threads).

inet56, 2021-04-14:
still unclear to me.

nick-knight, 2021-04-14:
Feel free to ask a question and we will try to answer it.

allenjbaum, 2021-04-15:
The usual answer is that a hart could be a single core, or a single thread
of a multithreaded core.
But it is specifically not a software thread, which requires software to
context switch between threads.
This is as opposed to multithreaded cores that a implement copy of all
architectural state for each thread and does the switching between them in
HW (or even execute them in parallel in a superscalar machine).

TimBandTech, 2021-08-29:
Traditionally multiple software threads operate via context switching. The OS is responsible for this. It seems that the hart is breaking into this a bit. Unlike orderly stack calls, during a context switch the entire processor state of the previous running thread is saved by the OS(registers and flags basically), the last processor state of the new thread is loaded, and then the new thread commences its execution; all thanks to the OS. This does incur some overhead and latency issues arise. Less context switching is more efficient. Possibly the hart will allow for some very low latency performance and raise system efficiency for these routines. For instance an ISR of old is responsible for pushing registers as it uses them and popping them back when done. If an ISR gets its own hart then this would be unnecessary. The hart is blurring quite a few assumptions. This could be trouble if the principles are not accurately documented. Too much historical attention should not be necessary, yet somewhere the philosophy will have to be scrutinized. To what degree will an old OS run well on a RISC-V? It's an exciting shakeup.

closed 2021-10-31 by aswaterman

syemetsgit, 2022-06-23:
Did someone see hart implemented in HW? I've checked several implementations and saw that HARTID is a simple core number. So there is no "HW context", there is multicore CPU. I am hardware engineer and look in Verilog|VHDL source (and into scala/chisel a bit).