riscv/riscv-isa-manual #608: PMP clarification

brucehoult, 2020-10-25:
It seems to me the following questions are not adequately addressed in the specification.

1) Can a TOR entry use a preceding NAPOT/NA4 entry as its base address?

Interestingly, a NAPOT entry interpreted as a TOR falls inside the NAPOT block, and as as first match wins it doesn't actually matter whether the succeeding TOR entry uses as its start address the start of the NAPOT block, the end of the NAPOT bloc, or something in the middle.

2) what happens if a TOR entry specifies a lower address than the preceding TOR or NAPOT/NA4 entry? 

Does this produce a null region?

If not then how do you specify default permissions (necessarily using later entries) with one or more "cut-outs" with different permissions?

3) does an A=00 OFF/Null region use the corresponding pmpaddr address to reset the bottom of range for a subsequent TOR entry, or is it ignored entirely? Does an A=00 entry in fact stop the scanning of PMP entries?

aswaterman, 2020-10-26:
1. Yes.
2. Null region.
3. The corresponding pmpaddr still provides the base address for a subsequent TOR.

I claim all of these follow from a literal reading of the spec.

brucehoult, 2020-10-26:
Thanks for your answer.

While I think the answers make sense and were what I was hoping, I for one couldn't see any way to prove them from the spec, and I could easily imagine some people building hardware making unfortunate incompatible assumptions.

aswaterman, 2020-10-26:
1. follows from the fact that we don't say anything about it, i.e., it's not a special case.
2. follows from the fact that no y can satisfy the relation pmaddr_(i-1) <= y < pmaddr_i unless pmaddr_(i-1) < pmpaddr_i.
3. same as 1.

It's terse, but it's all there.  Some more non-normative text might be called for, but any ambiguity should be resolved by the formal model.

aswaterman, 2020-10-26:
Partially in response to feedback from @allenjbaum, I've made some clarifications to this text.  They aren't normatively necessary, but they might reduce the probability that a reader gets the wrong impression by incorrectly attempting to read between the lines.  https://github.com/riscv/riscv-isa-manual/commit/e1b0e496380710d7705f82a21a497aeeeefd3a7b

closed 2020-10-26 by aswaterman