riscv/riscv-isa-manual #609: PMP is based on physical addresses

pdonahue-ventana, 2020-11-06:
The PMP description says:
> Attempting to execute a load or load-reserved instruction whose **effective** address lies within a PMP region without read permissions raises a load access-fault exception. Attempting to execute a store, store-conditional, or AMO instruction whose **effective** address lies within a PMP region without write permissions raises a store access-fault exception.

This should talk about physical addresses rather than effective addresses.

aswaterman, 2020-11-06:
Good catch, Paul.  Can you make a PR that fixes this?

One thing to consider is that the physical addresses might not be contiguous for accesses that span page boundaries.  So we might need to write something like "whose physical addresses" instead of "whose physical address".

pdonahue-ventana, 2020-11-06:
I tried to contemplate spanning page boundaries in #610.

closed 2020-11-06 by pdonahue-ventana