riscv/riscv-isa-manual #621: should mmu page table walke address be checked by pmp according to the source of access?

leBsky, 2021-01-27:
It is obvious that MMU hardware page table walk address should be checked by PMP. And in my opinion, since  the page table walk is only a kind of load from physical memory, so the PMP should check the read privilege only for the address.  However, someone else holds the opinion that  PMP should check hardware page table walk address according to the source of access. For example, if a store instruction causes the page table walk, then PMP should check the write privilege of the page table walk address. Can anyone clarify this point?

gfavor, 2021-01-27:
PMP is intended to check each and every read/write/ifetch access to memory - irrespective of what caused that access and irrespective of whether it is an explicit or implicit access (tablewalk accesses, for example, are implicit accesses).

If a read access is being checked, then it should be checked for read permission - even if it resulted by a store instruction causing a tablewalk.  Similarly, if a talkwalk for a load instruction does a write access (to set PTE A/D bits), that write access should be checked for write permission.  In general, PMP checking is based on the type of the access itself, independent of the cause of the access.

Btw, also note that the Priv spec says:  "The effective privilege mode for implicit page-table accesses is S."  This is irrespective of the current privilege mode.

allenjbaum, 2021-01-27:
Note that a table walk could result in a write (updating A/D bits) !

aswaterman, 2021-01-27:
There's also an explicit subsection on this topic: https://github.com/riscv/riscv-isa-manual/blob/697ebd1f7562a2492f215a47eb6f2be75f8215a9/src/machine.tex#L3453

closed 2021-01-28 by leBsky