riscv/riscv-isa-manual #622: PTE with W=1,R=0

leBsky, 2021-02-02:
In 4.3.2 Virtual Address Translation Process of riscv-priviledged-v1.10, it is claimed that if pte.r=0 and pte.w =1 then a page fault shoud be raised. Should we consider mstatus.mxr here? For example, when mxr=1, pte.x=1, pte.w=1 and pte.r=0, should we still raise the page fault?

[image]
Besides, from the picuture above, it is said that XWR=110/010 is reserved for future use. Since these combinations of XWR are possibly used in the future, why do they raise page fault now? I think when these combinations have their own use in future then  no page fault will be raised.  As a result,  the design under current spec will be differnt from the design under future spec in this XWR 110/010 page fault mechanism.

allenjbaum, 2021-02-03:
I haven't seen anyone else chime in. The ePMP extension is proposing to use
those encodings for regions that are shared between M and S/U.
But I would guess the reason for faulting is for forward compatibility.
If a future extension assigns a meaning to it, then new SW that attempts to
use it will get the exception and can emulate the new functionality.

gfavor, 2021-02-03:
On Mon, Feb 1, 2021 at 7:46 PM leBsky <notifications@github.com> wrote:

> In 4.3.2 Virtual Address Translation Process of riscv-priviledged-v1.10,
> it is claimed that if pte.r=0 and pte.w =1 then a page fault shoud be
> raised. Should we consider mstatus.mxr here?
>
In general (i.e. for defined permission encodings) it is.  But since the
future definition of a Reserved encoding could be anything, it's pretty
hard to argue that MXR should somehow affect the current behavior for a
Reserved encoding.

> For example, when mxr=1, pte.x=1, pte.w=1 and pte.r=0, should we still
> raise the page fault?
>
> [image: image] [...]
Versus do what?  I think the only alternative would be to interpret it as
an alias of one of the other encodings - which defeats the intent of it
being reserved for future definition.  So the only reasonable behavior is
to trap attempted use of that Reserved encoding.

> I think when these combinations have their own use in future then no page
> fault will be raised. As a result, the design under current spec will be
> differnt from the design under future spec in this XWR 110/010 page fault
> mechanism.
>
But the architectural idea is that this new meaning and behavior for a
reserved encoding is done as part of an architectural extension.  If
software expecting to use that extension runs on hardware not supporting
the extension, it will (properly) get a trap.  (And maybe the functionality
of the extension can be emulated?)  And if software doesn't use that
extension, then it will still run properly (since it couldn't have been
depending on some useful non-trapping existing behavior).

Greg

jscheid-ventana, 2021-08-10:
Is this still pending or considered rejected?

aswaterman, 2021-08-11:
I think this is essentially a duplicate of #631, which I closed on the basis that "reserved" has its usual meaning: software should not rely on any particular behavior; hardware's behavior is left unspecified; future behavior my be defined by future extensions.

closed 2021-08-11 by aswaterman