riscv/riscv-isa-manual #633: Hypervisor interrupt prioritization unspecified

scottj97, 2021-03-16:
See recent discussion on [tech-privileged](https://lists.riscv.org/g/tech-privileged): [Hypervisor interrupt enables](https://lists.riscv.org/g/tech-privileged/topic/hypervisor_interrupt_enables/81068094?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,81068094)

Interrupt enables depend on relative privilege levels. The general rule is that a less-privilege interrupt is always disabled, a same-privilege interrupt depends on an xIE bit (e.g. mstatus.mie or sstatus.sie), and a more-privileged interrupt is always enabled.

But the relative privilege of hypervisor modes (V=1) vs non-hypervisor modes (V=0) is not specified, leading to questions:

1. Are HS-mode interrupts enabled when in VS-mode? Does it depend on sstatus.sie?
2. Are VS-mode interrupts enabled when in U-mode?

This would be resolved if the priorities were specified in the obvious order, though perhaps this order applies only to interrupt enabling: M > HS > U > VS > VU

cc @jhauser-us

fintelia, 2021-03-19:
> Are HS-mode interrupts enabled when in VS-mode? Does it depend on sstatus.sie?

It would be rather unreasonable for HS-mode's timer interrupts to be unconditionally disabled when in VS-mode, since this would make it impossible for a hypervisor to timeshare a hart between multiple VMs.

> Are VS-mode interrupts enabled when in U-mode?

I would have expected the answer to be no, but after a detailed reading of the hideleg section I think the current behavior is that they are:

https://github.com/riscv/riscv-isa-manual/blob/ad5f04dddb58b3708c22bb7aec0456ef1c42afbf/src/hypervisor.tex#L428-L430

In contrast, the conditions for delegating exceptions are analogous but specifically say that V=1 is required: 

https://github.com/riscv/riscv-isa-manual/blob/ad5f04dddb58b3708c22bb7aec0456ef1c42afbf/src/hypervisor.tex#L414-L416

scottj97, 2021-03-19:
I don't believe hideleg answers the question about U-mode. Consider an interrupt that's been delegated to S-mode via mideleg. Such interrupts are globally disabled when in M-mode, since S-mode is lesser privileged. Analogously, if VS-mode is considered lesser privileged than U-mode, interrupts delegated via hideleg would be globally disabled when in U-mode.

Since the hypervisor spec talks about U-mode in the context of hypervisor code running there, it seems to me that the desired behavior is for the guest OS's interrupts (i.e. VS-mode interrupts) should be globally disabled when in U-mode.

Spike (whose hypervisor was developed by @avpatel) disables VS-mode interrupts when in U-mode.

gfavor, 2021-03-22:
To question #1 posed by Scott, didn't John Hauser fairly clearly answer this in his post on 3/7 @11:39am to this thread: [tech-privileged] Hypervisor interrupt enables?  Here's the gist of the question and his response (I believe all in the context of V=1; or does Spike do this even when V=0?):

> I expect, and have observed in Spike, that `sstatus.SIE` will apply
> to interrupts delegated to HS-mode, and `vsstatus.SIE` will apply to
> interrupts delegated to VS-mode.

Your expectation about the roles of sstatus.SIE and vsstatus.SIE
is correct, and you're probably also right to say that this isn't
explained clearly in the current Privileged Architecture document.

scottj97, 2021-03-22:
No, because that doesn't answer whether sstatus.SIE still applies if in VS-mode. That answer depends on the relative privilege level of HS-mode vs VS-mode. As @jhauser-us replied a few minutes later:

> Correction:  Interrupts are globally enabled for HS level if the current privilege mode has less privilege than HS (U, VS, or VU mode) or if the current privilege mode is HS and the HS-level sstatus.SIE = 1.

Which does answer my question 1, but still not 2, and in any case this needs to be explained in the Priv Spec.

scottj97, 2021-03-22:
I think John Hauser did answer my question 2 in a [later post](https://lists.riscv.org/g/tech-privileged/message/524):

> The original intent was that U-level interrupts not be able to interrupt VS mode, and VS-level interrupts not be able to interrupt U mode.

I agree that this is the desired behavior (at least the second half; I haven't thought about U-level interrupts).

But the spec needs to explain this.

gfavor, 2021-03-22:
I think (?) we're on the same page as far as VS-level interrupts being unconditionally globally disabled while in HS-mode.  The sstatus register is factored in for HS-level interrupts, and vsstatus is ignored.

While when in VS-mode both sstatus and vsstatus must be factored in.  The former for HS-level interrupts, and the latter (_masquerading as sstatus wrt the guest_) for VS-level interrupts.

Are we thinking the same things?

I agree that the answer to question #1 should be clarified in the spec, and that question #2 is outstanding?

scottj97, 2021-03-22:
> While when in VS-mode both sstatus and vsstatus must be factored in. The former for HS-level interrupts, and the latter (_masquerading as sstatus wrt the guest_) for VS-level interrupts.

This is how Spike works today, but it's not what John Hauser said. He said sstatus.SIE only applies when in HS-mode, and that seems like the more desirable behavior to me.

gfavor, 2021-03-22:
Before the last post, I was about to correct what I said.  I believe that while in VS-mode HS-level interrupts are unconditionally globally enabled and sstatus is ignored.  This matches what John said.  This also matches John's assertion of this relative priority: M > HS > VS.  So I would side with John in effectively saying that Spike appears to be wrong.

jhauser-us, 2021-03-23:
Assuming I didn't make a mistake, here's how it's supposed to be:

An interrupt is taken in M mode when:  (a) the corresponding bits in `mip` and `mie` are both set, AND (b) the interrupt is not delegated below M level by `mideleg`, AND (c) either the current mode is below M (S/HS, U, VS, or VU) OR the current mode is M and `mstatus`.MIE = 1.

An interrupt is taken in S/HS mode when:  (a) the corresponding bits are set in both HS-level `sip` and `sie` or in both `hip` and `hie`, AND (b) the interrupt is not delegated to a virtual machine by `hideleg`, AND (c) either the current mode is below S/HS (U, VS, or VU) OR the current mode is S/HS and `sstatus`.SIE = 1.

An interrupt is taken in VS mode when:  (a) the corresponding bits in `vsip` and `vsie` are both set, AND (b) either the current mode is below VS in the virtual machine (VU) OR the current mode is VS and `vsstatus`.SIE = 1.

These rules imply that when the current mode is U, an interrupt may be taken to M mode or S/HS mode, but not to VS mode.

allenjbaum, 2021-03-23:
Shouldn't there be an "elseif" in front of the 2nd and 3rd cases?
Otherwise, more than one rule can evaluate to true, can't it?

jhauser-us, 2021-03-23:
And when I wrote "An interrupt is taken in *X* mode", I meant "An interrupt is taken *into* *X* mode", i.e., the interrupt trap causes the current operating mode to become *X* mode if not already.

gfavor, 2021-03-23:
Doesn't the "into S/HS mode" clause also depend on the interrupt being delegated down by mideleg?  And ditto for the "into VS mode" clause wrt mideleg and hideleg?

jhauser-us, 2021-03-23:
Allen Baum wrote:
> Shouldn't there be an "elseif" in front of the 2nd and 3rd cases?  Otherwise, more than one rule can evaluate to true, can't it?

More than one rule can apply only when you have multiple different interrupts occur at the same time.  Currently, I don't know that software has a way to guarantee simultaneous arrival of interrupts targeted to multiple privilege levels at or above the current operating mode, so if the rules trigger in an order you don't expect, we can always just claim that the different interrupts actually arrived in the order in which the rules triggered.

jhauser-us, 2021-03-23:
Greg Favor wrote:
> Doesn't the "into S/HS mode" clause also depend on the interrupt being delegated down by mideleg? And ditto for the "into VS mode" clause wrt mideleg and hideleg?

If an interrupt isn't delegated down to a privilege level, the ip and ie bits are read-only zeros for that privilege level.

gfavor, 2021-03-23:
Ahh yes, that's what I forgot.

allenjbaum, 2021-03-23:
I just reread it, and the (unstated) assumption is that the second clause
can only occur if the interrupt has been delegated.
Reading that clause as-is without stating explicitly, or understanding it,
will lead to any single interrupt  being taken in multiple modes
simultaneously, whatever that means.
A careful reading of some other section of the spec may explain how that
works - but simply adding a single word to this eliminates any possibility
that someone will screw it up.
That fact that even some of us, who are pretty familiar with the RISC-V
priv architecture,  can forget those details (c.f. Greg's question)
validates that point.

jhauser-us, 2021-03-23:
Allen Baum wrote:
> A careful reading of some other section of the spec may explain how that works - but simply adding a single word to this eliminates any possibility that someone will screw it up.

Your screw-up scenario assumes the implementor also mishandled `mideleg` or `hideleg`, allowing a lower-privilege ip bit to be set when the rules require the bit to be read-only zero.  The text on that point should be clear enough, and I feel if we can't assume that simple, clear-cut things are being handled correctly, the spec will quickly go off the rails.

Inserting *elseif*'s would imply to implementors that they need to implement the logic to ensure the priority, when in fact they don't, because it should already be taken care of.

At most, this question would be worthy of a comment in the spec, to remind readers what they should already know about how delegation works.

scottj97, 2021-03-23:
> More than one rule can apply only when you have multiple different interrupts occur at the same time. Currently, I don't know that software has a way to guarantee simultaneous arrival of interrupts [...]

Most of the delegatable interrupts are controlled entirely by M-mode or HS-mode software, e.g. ssip, stip. So it is possible for M-mode software to set up multiple pending delegated interrupts, and even M-mode interrupts with mstatus.mie=0, and then mret to a lesser privilege. At that point it should be deterministic which one of the pending interrupts is taken first.

The spec [already covers this](https://github.com/riscv/riscv-isa-manual/blob/ad5f04dddb58b3708c22bb7aec0456ef1c42afbf/src/machine.tex#L1572-L1576) though it needs the same clarification w.r.t. the relative privilege of HS-level vs VS-level interrupts.

I think we are all in agreement now on the desired behavior. (At least, the four of us.) The remaining question is, where and how to explain this in the spec?

Since the non-hypervisor interrupt prioritization is explained in the section on "Machine Interrupt Registers (mip and mie)", perhaps the hypervisor additions should be explained in "Hypervisor Interrupt Registers (hvip, hip, and hie)"?

fintelia, 2021-03-23:
> An interrupt is taken in S/HS mode when: (a) the corresponding bits are set in both HS-level sip and sie or in both hip and hie, AND (b) the interrupt is not delegated to a virtual machine by [...]

I would suggest replacing (b) with something like the following so that we're absolutely clear:

_(b) the corresponding bit in hideleg is not set OR the current mode is S/HS or U_

allenjbaum, 2021-03-24:
This is a flame. Because.... hot button. I'm going to get in trouble.

I think we all agree that the behavior is what we want.
My bias is enabling spec readers to understand what is going on without
having to reference a chapter somewhere else.
Explanations are much less useful if details are scattered all over.

This is not about the semantics - this is about accessibility and
understandability of documentation.
The audience for the spec is not just us, the RV architects; it's students,
both undergraduates and garduates and post-dpcs, and thier instructors.
It's,verification engineers, programmers, tool vendors, debuggers, and many
more..
A wide audience means the spec should not be any harder to understand than
necessary.
That is the purpose of good documentation. Good documentation doesn't make
someone connect the dots - it draws the lines.
Otherwise, just point them at some Chisel and let them figure it out - why
do you need to write anything else?

Yes, they'll eventually figure out what is going on - but if you can avoid
wasting their time by adding a few words you have few excuses not to.

scottj97, 2021-03-24:
I tend to agree with @allenjbaum on this specific issue; I know when I first learned RISC-V interrupts I found it quite difficult to put all the pieces together. For example, I just reread the section on `sie` and `sip` and nowhere does it say anything about `mideleg` masking anything. (I know that's in the spec somewhere...but I can't find it.)

However I think it's a distraction from the issue at hand. We don't need [this entire explanation](https://github.com/riscv/riscv-isa-manual/issues/633#issuecomment-804667077) added to the spec, because most of that behavior is already explained.

Everything could be resolved with one additional line: "For purposes of interrupt global enables, VS-mode is considered less privileged than HS-mode and U-mode."

jhauser-us, 2021-03-24:
Allen Baum wrote:
> This is a flame. Because.... hot button. I'm going to get in trouble.

If you're referring to me, I don't mind a little debate.  Respectful differences of opinion are expected, if sometimes a bit annoying.

I readily acknowledge that the RISC-V spec isn't totally clear in many details, including about interrupts.  I've said so myself many times, though more often in private conversations.  The document has multiple authors, and I've attempted to improve it in places, and continue to do so.  A while ago, I submitted a pull request that I believed made interrupts a little clearer, but an important detail was rejected.  And because no one has yet offered a plausible replacement, it remains in limbo.

I agree also that the hypervisor chapter, which I bear more personal responsibility for, has a deficiency on the narrow point for which Scott Johnson opened this issue.  I'm not exactly sure when I or someone else will get around to fixing it in the document, but I expect this GitHub issue to remain open until that happens.

fintelia, 2021-03-24:
My proposal would be to make the smallest possible changes to make sure these questions are specified, and then add a big chunk of commentary text with a complete description like @jhauser-us wrote.

gfavor, 2021-03-24:
To resolve this particular case, could I suggest that Scott create a PR with his one-line clarification.  It's concise and to the point.  As to where in the spec, maybe the end of section 5.1 (versus say 5.2.11)?

If desired, then a separate PR could be created that adds non-normative text (that may end up getting more massaging by people to arrive at an acceptable PR to merge).

scottj97, 2021-03-24:
Done: #634

jhauser-us, 2021-03-24:
Scott Johnson wrote:
> Most of the delegatable interrupts are controlled entirely by M-mode or HS-mode software, e.g. ssip, stip. So it is possible for M-mode software to set up multiple pending delegated interrupts, and [...]

You're right; I overlooked that way to do it.

It's of course always possible for multiple different interrupts to occur simultaneously for a single privilege level, which we handle in the spec with separate statements about interrupt priority.  I still believe this is an adequate way to deal with multiple interrupts for different privilege levels.  If I'm not mistaken, the spec already says that interrupts for M mode have higher priority than for lower privilege levels, etc.

closed 2021-03-25 by scottj97