riscv/riscv-isa-manual #635: Information reguarding rv32e lost in newer versions of spec.

EchelonX-Ray, 2021-03-31:
I understand that it is still in draft, but noticed that in v2.2 of the user spec, section 3.2 describes the case of registers 16 through 31 being attempted access in rv32e architecture.  It states that an illegal instruction exception should be raised.  However, there is nothing regarding this in the most recent official release.  The topic is not addressed.

aswaterman, 2021-03-31:
In August 2018, the RV32E spec was changed to indicate that those encodings are designated for custom use.  By implication, raising an illegal-instruction exception continues to be a valid implementation.  The relevant clause is here: https://github.com/riscv/riscv-isa-manual/blob/bf971f91f95e9c9cece7f9a137b3e276a66332f5/src/rv32e.tex#L45

EchelonX-Ray, 2021-04-01:
Thank you for the explanation.

closed 2021-04-01 by EchelonX-Ray