riscv/riscv-isa-manual #636: canonical order for K extension

jim-wilson, 2021-03-31:
Chapter 28.6 "Additional Standard Extension Names" defines the canonical ordering for extensions
    IMAFDQLCBJTPVN
The K extension is missing from this list.  I'm reviewing a binutils path for K, so I need this to be defined.  I don't care where it goes, just that there is a standard place to put it.  The proposed patch puts it between J and T.  So we have
+  return "mafdqlcbjktpvn";
in the patch, but if you want to put it elsewhere that is fine.  We can change binutils to match the docs.

I don't see any other obvious single letter extension missing from the list, so I think only K is missing.

kasanovic, 2021-04-01:
I looked at the latest crypto  specs (0.9), and this seemed to indicate that K was being used to represent a set of scalar crypto extensions.  In this case, I would place it after b rather than after j given the affinity of k with b (i.e., swap j and k in given list).

We had been trying to use single letter names for application processor extensions on basis that they tend to have much more constrained choices (i.e,, a lot of embedded systems will have to have a long string of Z names anyway), which using K for base scalar crypto goes against.  If K was used for vector crypto intended for app processors, then after V would have been more natural. 

Profiles should make this less of an issue, and I understand the experimental toolchain work has to go ahead.

jim-wilson, 2021-04-01:
After B is fine with me.

kito-cheng, 2021-04-01:
+1 for need the canonical order for K extension, either order is fine to me.

neelgala, 2021-04-01:
What about the placement for the "U" extension ?

neelgala, 2021-04-01:
Also, I was under the impression that all "Z" extensions should be placed towards the end of the string. But the table 27.1 seems to put Zicsr, Zifencei, Zam and Ztso before the S and H extensions. Is that how it should be ?

jim-wilson, 2021-04-01:
Apparently you mean table 28.1, but this table does not show the order that extensions go into the architecture name.  It is merely a list of extensions sorted by category.  Zicst etc are standard unpriv extensions, so they go in that category.  But that has nothing to do with the architecture name order.

I haven't heard of U.  It isn't on the specification status spreadsheet.  We probably don't need that solved now.  We do need K solved now because it is on the short list for approval.

neelgala, 2021-04-02:
I am looking at the latest release pdf: https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20210325-bf971f9
and refering to Table 27.1 on page 150. The caption of the table says: 

```
The table also defines the canonical order in which
extension names must appear in the name string, with top-to-bottom in table indicating first-to-last
in the name string, e.g., RV32IMACV is legal, whereas RV32IMAVC is not.
```
Hence the questions on ordering of Z extensions because both K and B have significant number of Z subsets.  Based on what gets decided here, I'll need to make updates to the schema/checker in [riscv-config](https://github.com/riscv/riscv-config)

Also shouldn't the ordering account for the "S" and "U" extensions as well (though they being privileged extensions)?

allenjbaum, 2021-04-02:
Yeah, I had to go back and check which chapter it was in now also. It has
changed as new chapters get added.

"Category" is the character after the "Z" and is in the order
IEMAFDGQLCBJTPVNSH (K is missing in action here...)
Z extensions are ordered by category order, and alphabetically within
category,
This version does not say how Z extensions are ordered within the ISA
string relative to category:
   at the end of the ISA_string? or Interleaved immediately following each
category they belong to?
Krste had indicated that extensions that had Z-dependencies should be next
to each other to (e.g. since Finx modifies F, then F_Zfinx, FD_Zfinx, or
even D_Zfinx since D implies F) as opposed to new subextensions Zfinx,
Zdinx, Zqinx.
That particular example is problematic because Zfinx removes
instructions from F, among other things.

Also changed since earlier version is... version numbers appear to be
allowed

jim-wilson, 2021-04-02:
This is getting badly confused.  Please open new issues for new questions.  This is about the K extension.  I need an answer for the K extension ordering or toolchain work will be delayed.

gfavor, 2021-04-02:
On Thu, Apr 1, 2021 at 11:45 PM Allen Baum ***@***.***> wrote:

> This version does not say how Z extensions are ordered within the ISA
> string relative to category:
> at the end of the ISA_string? or Interleaved immediately following each
> category they belong to? [...]

My impression of the discussions around ZFinx-related naming is that there
is not yet a clear consensus agreement (let alone it being captured in the
Unpriv spec)?  Krste put out his thoughts, others seemed to raise issues
with that and suggest alternatives, ....

If correct, how does this get resolved (versus floundering around for tbd
months)?  If not, can someone please state the definitive resolution and
make a PR to the Unpriv spec?

Greg

allenjbaum, 2021-04-02:
Sorry for the distraction. The newest chapter_whatever_it_is indeed is
completely missing K, which is odd.
But, aren't you also concerned about 8 Z-sub-extensions of K that can be
invoked individually (2 of which are not actually included in K-extension
proper, the Chinese versions) or as groups?
(e.g. Zkne is K without the random number generator)

aswaterman, 2021-04-02:
I've updated the spect to reflect Krste's suggestion for where K belongs in the ISA string.  (The rest of this thread seems somewhat unrelated to the original post; please open a new issue if need be.)

closed 2021-04-02 by aswaterman

jim-wilson, 2021-04-02:
U would be User mode of course.  But we don't have any U* extensions like we do for S, so I don't see a need to define a canonical ordering for U.

commit 12717374 (2021-04-02): Define canonical location of K extension in ISA string

Resolves #636