riscv/riscv-isa-manual #637: canonical order for Z extensions

jim-wilson, 2021-04-02:
The GNU toolchain requires that the single letter extensions come first, followed by S*, H*, Z*, and X*.  However, the table 27.1 lists 4 Z extensions as coming before the S* extensions.  Since the GNU toolchain does not officially support any S* or H* extensions currently, there is no problem yet.  However, as soon as we add our first S* or H* extension the GNU toolchain will not be compatible with the ISA spec.  There are also a number of new Z extensions expected soon, B and K add a number of them, plus Zfh, Zfinx, etc.  What about those?  Will they go after S and H or before?  A consistent rule makes things easier for the toolchain.  I would prefer that all Z extensions go next to each other to make parsing canonical order easier.

rohan:2406$ riscv32-unknown-elf-gcc -march=rv32i_sfoo_zifencei -mabi=ilp32 -S tmp.c
rohan:2407$ riscv32-unknown-elf-gcc -march=rv32i_zifencei_sfoo -mabi=ilp32 -S tmp.c
riscv32-unknown-elf-gcc: error: '-march=rv32i_zifencei_sfoo': unexpected ISA string at end: 'sfoo'
rohan:2408$ 

With binutils modified to accept sfoo as an extension
rohan:2007$ riscv32-unknown-elf-as -march=rv32i_sfoo_zifencei -c tmp.s
rohan:2007$ riscv32-unknown-elf-as -march=rv32i_zifencei_sfoo -c tmp.s
Assembler messages:
Error: -march=rv32i_zifencei_sfoo: unexpected ISA string at end: sfoo
rohan:2008$ 

So both GCC and Binutils have incorrect parsing of architecture strings if zifencei is supposed to come before S*.

Someone should check LLVM.

Nelson1225, 2021-04-06:
I look at the llvm source here,
https://llvm.org/doxygen/RISCVTargetStreamer_8cpp_source.html#l00040

In the emitTargetAttributes, it seems like llvm does not support any S* or H* extensions either.  I think the orders of if statement means the orders of the extensions, so the orders of S*, H* and Z* depend on where do we want to put them in the future patches.

Nelson1225, 2021-04-08:
I have a patch that fixes the prefixed order problem, and also support to parse the multi-letter prefixes like zxm.
https://sourceware.org/pipermail/binutils/2021-April/116054.html

After applying the patch, and modifying the binutils to accept sfoo, hfoo and zxmfoo, I can get the expected result as spec,

$ riscv64-unknown-elf-as -misa-spec=20191213 -march=rv64i_zicsr_sfoo_hfoo_zxmfoo_xriscv1p0 empty.s -o empty.o
$ riscv64-unknown-elf-readelf -A empty.o
Attribute Section: riscv
File Attributes
  Tag_RISCV_arch: "rv64i2p1_zicsr2p0_sfoo1p0_hfoo1p0_zxmfoo1p0_xriscv1p0"

$ riscv64-unknown-elf-as -march=rv64i_sfoo_hfoo_zifencei empty.s -o empty.o
Assembler messages:
Error: -march=rv64i_sfoo_hfoo_zifencei: prefixed ISA extension `zifencei' is not in expected order.  It must come before `hfoo'

We used to get "unexpected ISA string at end: sfoo" error for the above case, but now we can give the more friendly errors to users.  Though it may be better to report "It must come before `sfoo'" rather than the `hfoo'.  I think it can be improved in the future patches.