riscv/riscv-isa-manual #639: Clarify access width requirements for mtime/mtimecmp

jrtc27, 2021-04-17:
Currently, the spec has the following:

> In RV32, memory-mapped writes to `mtimecmp` modify only one 32-bit
> part of the register.  The following code sequence sets a 64-bit
> `mtimecmp` value without spuriously generating a timer interrupt due
> to the intermediate value of the comparand:
> ...
> For RV64, naturally aligned 64-bit memory accesses to the `mtime` and
> `mtimecmp` registers are atomic.

To me this says the following:

1. Software must use 32-bit accesses on RV32 and <32-bit accesses are not required to be supported (though a suitably general bus-register interface might permit them, even if using it is a completely foolish thing to do).
2. Software can use 64-bit accesses on RV64.
3. I believe there is also the implication that, given it doesn't say otherwise, anything that applies to RV32 also applies to RV64, i.e. if you really want to you can do 32-bit accesses, but <32-bit accesses are still not required to be supported.

Is my interpretation of the spec correct? Either way, I feel this could be stated more explicitly and clearly. T-Head have an RV64 implementation which does not permit 2, requiring 32-bit accesses on RV64[1], which I believe to be in violation of the spec as it currently stands. Is the intent of the spec that such an implementation be legal and software be unable to assume it can actually make use of native 64-bit accesses on RV64?

[1] And want OpenSBI to support this: http://lists.infradead.org/pipermail/opensbi/2021-April/000865.html

pdonahue-ventana, 2021-04-20:
See also https://lists.riscv.org/g/tech-privileged/topic/73100152

There was a lot of discussion including:
>> So the only constraint is that when a 64b naturally-aligned access is made to mtime/mtimecmp, the access must be completed atomically if the platform allows 64b naturally-aligned accesses to those registers? A platform is still allowed to signal an error on such accesses and to force an RV64 processor to access those registers with 32b loads and stores, right?
> 
> I think your interpretation of that sentence is accurate. FWIW, the insufficiently described Linux platform does assume such accesses are legal (more precisely, the various SBI implementations make that assumption).

But I don't think there was a spec change so it's still unclear.

jscheid-ventana, 2021-08-10:
Resolution to this remains of value.

aswaterman, 2021-08-11:
I'll speak on behalf of the priv spec editors in saying that it's pretty obvious that our intent was that these registers would support XLEN-bit memory accesses at the ISA level.

Otherwise, why would we have called out that baroque sequence as "setting the 64-bit time comparand in RV32", while providing no code sequences for RV64?  Had we expected that some RV64 systems might do something similar, we would've instead written something like, "[...] on systems that don't permit 64-bit accesses to this register".

Since you brought it up, I'll clarify our intent.

commit 50694a2c (2021-08-11): Clarify that RV64 accesses to mtime[cmp] are atomic

As stated clearly by the preceding text, the baroque sequence for updating
mtimecmp is for RV32.  RV64 simply uses aligned loads and stores.

Closes #639