riscv/riscv-isa-manual #640: Timer interruptions in Supervisor Mode

matheuss-leonel, 2021-04-20:
Hello guys!

I'm an undergrate student at UFSC (Universidade Federal de Santa Catarina - Brasil).
We are studying kernel development, and RISC-V was chosen as the CPU (**32 bits**). Initially, we had just the Machine mode working, but now we have an OS with supervisor mode support (without user mode, for now), in a **singlecore** scenario.

When implementing the supervisor mode, we've faced some trouble with the Timer interruptions. We just couldn't delegate the timer interruptions from Machine mode to Supervisor mode directly. Our workaround:

- Set the **MTIE** bit of the **MIE** register always set to 1, for the entire life time of the OS
- Set **MTVEC** to the address of a function that only handle timer interruptions and forward it to Supervisor mode (it was very hard to do, since in supervisor mode we have pagination enabled, and Machine mode uses physical addresses). More of it below
- Set the **STIP** bit of the **MIP** register set to 1, leaving the Supervisor mode with the timer interruption always enabled, for the entire life time of the OS.
- Set the **STIE** bit of the **SIE** register set to zero. It's set to 1 by the Machine mode timer interruption handler, when timer interruptions arrive. Note that since the **STIP** bit (**MIP** register) is set to 1, when the Machine interruption handler executes "**mret**", the supervisor mode should call it's own Supervisor interruption handler

As you can see, the information flows something like this (our implementation in high level pseudo-code):

1. CPU executing in Supervisor mode, with pagination enabled
2. Timer interruption arrives
3. Machine mode interruption handler is called, and since pagination is ignored in Machine mode, we have to be very careful calculating the correct addresses involved, avoiding function calls, etc.
4. The M mode interruption handler saves the context of the registers
5. The M mode interruption handler checks to see if the interruption is a Timer interruption. If not, go to 8
6. If it's a Timer interruption, checks to see if the Supervisor has interruptions enabled (**SIE** bit of the **SSTATUS** register). Note that if the supervisor mode was handling some other interruption and the **SIE** bit set to zero, thus the timer interruption would be lost. If the **SIE** bit (**SSTATUS**) is set to 0, go to 8
7. Set the **STIE** bit of the **SIE** register to 1
8. Reset **MTIMECMP**
9. Restore context
10. call "**mret**", returning to Supervisor mode
11. If the **STIE** bit (**SIE**) is 1, there's a timer interruption pending to the Supervisor mode (set by the Machine mode interruption handler). Call the S mode interruptions handler
12. Saves register context (again)
13. Handle the interruption
14. If interruption is a Timer interruption, set **STIE** (**SIE**) bit to zero
15. Restore context

As you can see, there's room for a variety of bugs, race conditions, etc. When we tried to implement a **multicore** scenario, it was a debugging nightmare.

**Some questions:**

1. How the community handles this? 
2. It is impossible to completely delegate all the interruptions from Machine mode to Supervisor mode?
3. There's a better approach to solve this issue?

---

We are using:

- QEMU to simulate a machine with the RISC-V CPU. Output of "qemu-system-riscv32 --help":
> QEMU emulator version 5.0.0 (v5.0.0-dirty)
> Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers
> usage: qemu-system-riscv32 [options] [disk_image]
- Flag "-cpu", of the QEMU, set to "rv32gcsu-v1.10.0"

I tried to find some template for github issues, but couldn't, so I tried to do my best describing our scenario.

About my english: I'm working on it, tried my best.

Thanks!

scottj97, 2021-06-11:
@matheuss-leonel it sounds like you're using STIP and STIE backwards.

Also the M-mode code should not need to look at sstatus.SIE. The hardware will block the pending STIP interrupt until sstatus.SIE goes back to 1, and thereby avoid losing any timer interrupts. (Unless a second interrupt occurs before the first is processed.)

Gallium70, 2021-07-07:
The delegation mechanism allows the supervisor trap handler handles **supervisor** interrupt, while the hardware timer raises **machine** timer interrupt, which won't be handled in supervisor mode anyway. 
However, if you set mideleg.STI (mideleg[5]) to 1, and set mip.STIP in machine (timer) interrupt handler, then after `mret` , the trap will be routed to supervisor trap handler , otherwise it will still be handled in machine trap handler.

closed 2021-10-31 by gfavor