riscv/riscv-isa-manual #646: Erratum: Hypervisor CSRs still present in Vol II, Table2.1

acuster, 2021-05-08:
Hypervisor mode has been, to my understanding, deprecated in newer versions of the standard.

Unfortunately, in the ISA manual, vol II, section 2.2, Table 2.1 still includes a section called "Hypervisor CSRs". This appears to be an omission. If so, lines 77 to 89 should be deleted in
https://github.com/riscv/riscv-isa-manual/blob/master/src/priv-csrs.tex

cheers, and thanks for all the work on RISC-V, ~adrian

aswaterman, 2021-05-08:
Yeah, the table's still accurate.  True as it is that we deprecated an older hypervisor proposal that added an H-mode, there is now a new hypervisor proposal in Chapter 5 that uses this address range for new CSRs accessible to HS-mode but not VS-mode.

gfavor, 2021-10-31:
H extension is now being ratified, so this is no longer a problem.

closed 2021-10-31 by gfavor