riscv/riscv-isa-manual #650: Are single-byte modifications of the instruction guaranteed to be atomic?

palmer-dabbelt, 2021-05-22:
We jump through some hoops in Linux to avoid concurrently modifying the instruction stream while executing from it on another hart because that's not generally safe.  The scheme we currently use for ftrace is very unsafe, but as part of a tangentially related fix Steven pointed out that x86 only relies on the atomicity of single-byte modifications to the instruction stream in order to concurrently patch kernel code.  I laid out a scheme that should work for ftrace on RISC-V on the kernel mailing lists <https://lore.kernel.org/linux-riscv/mhng-edd9e8bd-e585-4b6b-8e40-797215bfdf75@palmerdabbelt-glaptop/T/#m5baafedd4ce8706a64a08b0e2ca3913ebccff39e>, but it does rely on single-byte modifications if the instruction stream being atomic WRT instruction fetch on other harts (we don't need any ordering guarantees, just that we either get the old byte or the new byte until we can enforce ordering via an IPI).

IMO it's kind of pedantic to even worry about this, but instruction fetch is super complicated so I generally try to avoid assuming anything about it.  I figured it'd be best to just ask here, in case I'm missing something.

For some reason I thought we'd written something down about this, but I couldn't find it in the ISA manual.  Not sure if I just missed it, but I remember having a more complicated scheme than this one -- something along the lines of aligned accesses of the minimum supported instruction size being atomic.  We could probably avoid the traps there by swapping between a jump and a nop, but since we have to IPI over anyway that doesn't buy us a whole lot when compared to the single-byte scheme.

David-Horner, 2021-05-23:
Are single-byte modifications of the instruction guaranteed to be atomic?

The safe answer is "No, not in the general case."

But I will go out on a limb and say, "Yes, but it depends".

In that you are working in Linux, the RVWMO model can be assumed,

and that model stipulates some Memory Model Primitives, specifically 
relevant is

"All memory operations are single-copy
atomic: they can never be observed in a partially-complete state.".

Of course there are many other Memory Models and memory implementations 
that may not stipulate this,

[consider data paths in microcontroller memory accesses to 4bit wide 
devices].

But, you do ask the relevant question "atomic with respect to ...".

That is one of the important matters that needs to be confirmed for the 
scheme to work.

It is almost assured that for your use case this condition will be met 
for "Application Class" processors on the same hart, an errata should be 
provided if in any specific cirecumstances it is not.

And you are most correct that instruction fetch, caching, translation 
and buffering can all complicate this property.

Considering this property across multiple harts is even more complicated 
given that the harts may not share the same microarchitecture and 
synchronization mechanisms.

Hence my "depends" qualifier.

And you tangentially ask an even more subtle questions:

"What if each hart has differing state WRT the instruction flow? What if 
thread control transfers between harts?

How do we ensure the integrity of this "one-byte-write" redirection [I 
have not read your proposed code, but this is a typical mechanism]. [or 
state transition to be more general and more obtuse],

throughout the necessary code path to resolve the code patch?

I expect there are numerous doctorate candidates considering these 
questions as thesis material.

> We jump through some hoops in Linux to avoid concurrently modifying
> the instruction stream while executing from it on another hart because
> that's not generally safe. The scheme we currently use [...]
very good.
>
> if the instruction stream being atomic WRT instruction fetch on other 
> harts
Ah. yes, here is the concern.
>
> (we don't need any ordering guarantees, just that we either get the 
> old byte or the new byte until we can enforce ordering via an IPI).
I suspect on a macro level you do need ordering guarantees, but that is 
another discussion.
>
> IMO it's kind of pedantic to even worry about this,
Pedantic is apparently my thing.
>
> but instruction fetch is super complicated so I generally try to avoid 
> assuming anything about it. I figured it'd be best to just ask here, 
> in case I'm missing something.
To avoid missing anything, that is to have very many questions and 
concerns raised,

I suggest you raise the issue with the memory model group which to date 
have tactically avoided advising on this issue [and other issues that 
are less complicated .]

> For some reason I thought we'd written something down about this, but 
> I couldn't find it in the ISA manual.
To the best of my recollection, some tentative responses were removed.
>
> Not sure if I just missed it, but I remember having a more complicated 
> scheme than this one -- something along the lines of aligned accesses 
> of the minimum supported instruction size being atomic.
And that would align with the Memory Model Primitives.
>
> We could probably avoid the traps there by swapping between a jump and 
> a nop, but since we have to IPI over anyway that doesn't buy us a 
> whole lot when compared to the single-byte scheme.
I will have to read this single-byte scheme in detail, but as it 
apparently also modifies actual executed code, the main issue is 
effectively the same; aligned half-word write is as atomic as byte in 
most memory subsystems.

gfavor, 2021-05-24:
On Sat, May 22, 2021 at 12:44 PM Palmer Dabbelt ***@***.***>
wrote:

> We jump through some hoops in Linux to avoid concurrently modifying the
> instruction stream while executing from it on another hart because that's
> not generally safe. The scheme we currently use for [...]
Palmer,

I haven't read the description of your scheme yet (and will), but regarding
atomicity I'll say the following:

- Single-byte writes are single-copy atomic in RVWMO (and RVTSO).  They
also are multi-copy atomic, i.e. "any store value that is visible to a hart
other than the one that originally issued it must also be conceptually
visible to all other harts in the system".

- By default, a series of single-byte writes to different addresses are
weakly ordered in RVWMO.  RISC-V store fences (e.g. fence w,w) can be used
to enforce the order in which they become visible to harts.

- There is also the uncertainty, within each hart, of coherent or
non-coherent instruction caching.  And possible microarchitectural things
like out-of-order fetch from the Icache into instruction buffers, and
having post-fetch/decode "op caches" like x86 and ARM.  Although,
practically speaking, the latter are probably not found in current RISC-V
implementations.

- If you need to ensure that all harts have seen a set of instruction
modifications all the way through their instruction cache, instruction
prefetch buffers, and execution pipelines, this will require use of the new
I/D Consistency extension being developed by the J TG (maybe to be ratified
this year - tbd?).  OR use of the existing IPI mechanism and use of fence.i
or equivalent.  (I'm guessing you are not requiring coherent instruction
caches.)

Past all that, I don't think there are any other complications.

Greg

> I laid out a scheme that should work for ftrace on RISC-V on the kernel
> mailing lists
> ***@***.***/T/#m5baafedd4ce8706a64a08b0e2ca3913ebccff39e,
> but it does rely on single-byte modifications if the instruction stream
> being atomic WRT instruction fetch on other harts (we don't need any
> ordering guarantees, just that we either get the old byte or the new byte
> until we can enforce ordering via an IPI).
>

palmer-dabbelt, 2021-06-02:
Sorry if I wasn't clear here, but I'm specifically talking about atomicity WRT the instruction stream.  The ISA is pretty clear about the data side of things, but IIUC the RVWMO is just for the data side of things and doesn't apply to instruction fetch.  We're trying to do is update the contents of a memory location concurrently with execution from that memory location.  IIUC that's not safe in general, but I'm wondering if it is guaranteed to be safe for byte accesses.

Specifically, we're trying to change the contents of the instruction stream from 0x00000013 to 0x00000000 (and back again, but I don't think that matters) by writing just the single byte that needs to change.  In order for our scheme to work we need a guarantee that the machine executes either one of those two instructions, and not some other instruction (which may, for example, modify machine state).  We're handling all the ordering ourselves, which is why we're using the trapping instruction, which we're doing via regular-looking fence.i/IPI/RVWMO stuff.

I don't see any direct language in the specification that provides any guarantee here, but I also don't see any discussion of sub-byte accesses.  When something's not mentioned in the ISA directly I'm never quite sure if that means we can rely on it not existing or if it can be anything, so that's why I'm asking.

aswaterman, 2021-06-02:
The ISA spec does not provide such an atomicity guarantee.  Fortunately, this is not a problem, because the Unix platform spec, which Linux depends on, provides a much stronger guarantee: https://github.com/riscv/riscv-platform-specs/commit/2511000cd54f3c62403bf48e07f4877a0416d506#diff-1d10d1d1a3a3b42a023da13bbb14b680bafcc95aaf22def56fa5c452125f39d6R16-R22

So, with @palmer-dabbelt's permission, I think we can close this issue.

Our approach does invite the question: why didn't we impose such a requirement in the base ISA itself?  Some microcontrollers connected to narrow instruction buses will perform instruction fetch serially.  We didn't want to unduly burden this design point, especially since software running on such processors isn't likely to benefit much from this property.

palmer-dabbelt, 2021-06-02:
Thanks, the platform spec is good enough for this as it's a Linux problem -- I thought we'd put it somewhere, that explains why I couldn't find it.

closed 2021-06-02 by palmer-dabbelt