riscv/riscv-isa-manual #651: Behaviour on non-leaf PTE's D/A/U bits

quxfred, 2021-05-27:
Hi,

In the section "Addressing and Memory Protection" of Privileged Architecture spec, I found this sentence below requiring the D/A/U bits to be cleared on non-leaf PTEs.

For non-leaf PTEs, the D, A, and U bits are reserved for future standard use and must be cleared by software for forward compatibility.

However in the follow section "Virtual Address Translation Process", the algorithm description ignored these bits. This leaves an apparent gap in how the implementation should do with regard to these non-leaf PTE bits. I notice the above sentence specifically mentioned software. Is this supposed to be a constraint to software only therefore not a hardware requirement? Am I right to assume that this is intended to be an implementation defined point?

Regards,
Freddie

aswaterman, 2021-05-27:
"Reserved" is the key word here.  In general, using such an encoding results in UNSPECIFIED behavior.  The implementation might trap, or ignore the bits, or do any other thing that doesn't violate protection boundaries.

cc @gfavor @daniellustig: although the intent here is clear, I can't recall whether we've furnished the necessary priv-spec machinery to make my claim true.  Essentially, PTEs should behave like instruction opcodes w.r.t. reserved encodings, but we need to actually write that if we haven't.

daniellustig, 2021-06-02:
I don't know if there's wording specifically defining "reserved" for priv.  The unpriv spec is clear that decoding a reserved instruction is UNSPECIFIED behavior, but I don't see similar language in the priv spec for cases like this.  (Entirely possible I'm just missing it.)  We should make sure this is clear both now and as RISC-V goes through its documentation overhaul.

FWIW the virtual-memory branch currently has this proposed change:

> If $pte.v=0$, or if $pte.r=0$ and $pte.w=1$, **or if any bits or encodings
>   that are reserved for future standard use are set within $pte$**, stop and
>   raise a page-fault exception corresponding to the original access type.

https://github.com/riscv/riscv-isa-manual/pull/611/files#diff-1673d75098c5b339d06598ff6bb126f7a429d6bed61593e9b41092ee0587c9b9R1541

That's actually stricter than just calling it UNSPECIFIED.  We wrote that text with setting of the PTE reserved bits in mind.  Is it indeed true that seeing a PTE with reserved bits set should specifically trigger a page fault while setting D/A/U on a non-leaf PTE is UNSPECIFIED?

gfavor, 2021-06-08:
I don't see anything in the Priv spec defining what "reserved" means.

The DAU bits are described as "reserved for future standard use and must be cleared by software for forward compatibility".  The high Reserved bits probably would have said the same thing if the 1.11 spec had said anything, which it doesn't.

Since nothing was said for the Reserved bits, we added the explicit "trap on non-zero" statement to v1.12 to fill in that hole in the spec.  But it seems like that statement would be equally suitable for the DAU bits in v1.12.

One could say that v1.11 left all these bits to trap, be ignored, or ....  And v1.12 tightens this up to better ensure that v1.12 implementations will be forward compatible with future standard uses of these bits.  That seems like a desirable and worthwhile motivation for doing this tightening in v1.12.

So I propose also requiring a page fault on non-zero DAU bits in no-leaf PTEs.  Thoughts for or against?

aswaterman, 2021-06-08:
Makes sense to me—but this should be listed as a technically incompatible change in the preface, since I agree with your interpretation that “reserved” by itself is a much weaker statement.

gfavor, 2021-06-08:
Dan, if you agree, can you add verbiage to the v1.12 tablewalk pseudocode for non-zero non-leaf DAU bits?  And add the "technically incompatible change" comment in the preface.

daniellustig, 2021-06-08:
Works for me too.  I think the added (bolded) text in my previous comment covers this already then, but I'll add the note about this being a technically incompatible change to the preface.

gfavor, 2021-10-31:
#759 will address this.

aswaterman, 2021-11-02:
Resolved via #759

closed 2021-11-02 by aswaterman